project-chip/connectedhomeip #34705: [CERT-TEST-FAILURE] TC-CCTRL-2.2 - Failed due to multiple AttributeErrors during the setup_class, on_fail, and teardown_class methods

Rajashreekalmane, 2024-08-01:
### Feature Area

Other

### Test Case

TC_CCTRL_2_2

### Reproduction steps

**TH** - python controller 
 commands used to build - 

- source scripts/activate.sh && scripts/build_python.sh -m platform -d true -i no
- source no/bin/activate

**TH_SERVER** - fabric admin
Command used to build -

-  scripts/examples/gn_build_example.sh examples/fabric-admin out/fabric-admin 'chip_mdns="platform" chip_inet_config_enable_ipv4=false'

**DUT** - fabric-bridge-app

- ./scripts/examples/gn_build_example.sh examples/fabric-bridge-app/linux out/debug/standalone chip_config_network_layer_ble=false 'import("//with_pw_rpc.gni")'

**Command used to launch the DUT** :`  ./fabric-bridge-app`
**Command used to run the python script** : ` python3 TC_CCTRL_2_2.py --commissioning-method on-network --discriminator 3840 --passcode 20202021 --paa-trust-store-path ../../credentials/development/paa-root-certs/ --storage-path admin_storage.json --string-arg th_server_app_path:/home/ubuntu/Jul_Dut/connectedhomeip/out/fabric-admin/fabric-admin`
The TC_CCTRL_2_2 test case encounters multiple AttributeErrors during the setup_class, on_fail, and teardown_class methods. These issues prevent the test case from executing as expected.
> [MatterTest] 08-01 09:29:53.614 INFO ==========> TC_CCTRL_2_2 <==========
> [MatterTest] 08-01 09:29:53.618 ERROR Error in TC_CCTRL_2_2#setup_class.
> Traceback (most recent call last):
>   File "/home/ubuntu/Jul_Dut/connectedhomeip/no/lib/python3.10/site-packages/mobly/base_test.py", line 418, in _setup_class
>     self.setup_class()
>   File "/home/ubuntu/Jul_Dut/connectedhomeip/src/python_testing/matter_testing_support.py", line 1671, in async_runner
>     return _async_runner(body, self, *args, **kwargs)
>   File "/home/ubuntu/Jul_Dut/connectedhomeip/src/python_testing/matter_testing_support.py", line 1659, in _async_runner
>     return asyncio.run(runner_with_timeout)
>   File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
>     return loop.run_until_complete(main)
>   File "/usr/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
>     return future.result()
>   File "/usr/lib/python3.10/asyncio/tasks.py", line 445, in wait_for
>     return fut.result()
>   File "/home/ubuntu/Jul_Dut/connectedhomeip/src/python_testing/TC_CCTRL_2_2.py", line 48, in setup_class
>     app = self.matter_test_config.user_params.get("th_server_app_path", None)
> AttributeError: 'MatterTestConfig' object has no attribute 'user_params'
> [MatterTest] 08-01 09:29:53.625 ERROR Exception happened when executing on_fail for setup_class.
> Traceback (most recent call last):
>   File "/home/ubuntu/Jul_Dut/connectedhomeip/no/lib/python3.10/site-packages/mobly/base_test.py", line 418, in _setup_class
>     self.setup_class()
>   File "/home/ubuntu/Jul_Dut/connectedhomeip/src/python_testing/matter_testing_support.py", line 1671, in async_runner
>     return _async_runner(body, self, *args, **kwargs)
>   File "/home/ubuntu/Jul_Dut/connectedhomeip/src/python_testing/matter_testing_support.py", line 1659, in _async_runner
>     return asyncio.run(runner_with_timeout)
>   File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
>     return loop.run_until_complete(main)
>   File "/usr/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
>     return future.result()
>   File "/usr/lib/python3.10/asyncio/tasks.py", line 445, in wait_for
>     return fut.result()
>   File "/home/ubuntu/Jul_Dut/connectedhomeip/src/python_testing/TC_CCTRL_2_2.py", line 48, in setup_class
>     app = self.matter_test_config.user_params.get("th_server_app_path", None)
> AttributeError: 'MatterTestConfig' object has no attribute 'user_params'
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "/home/ubuntu/Jul_Dut/connectedhomeip/no/lib/python3.10/site-packages/mobly/base_test.py", line 646, in _exec_procedure_func
>     func(copy.deepcopy(tr_record))
>   File "/home/ubuntu/Jul_Dut/connectedhomeip/no/lib/python3.10/site-packages/mobly/base_test.py", line 562, in _on_fail
>     self.on_fail(record)
>   File "/home/ubuntu/Jul_Dut/connectedhomeip/src/python_testing/matter_testing_support.py", line 1054, in on_fail
>     test_duration = (datetime.now(timezone.utc) - self.test_start_time) / timedelta(microseconds=1)
> AttributeError: 'TC_CCTRL_2_2' object has no attribute 'test_start_time'
> [MatterTest] 08-01 09:29:53.648 WARNING Stopping app with SIGTERM
> [MatterTest] 08-01 09:29:53.648 ERROR Error encountered in teardown_class.
> Traceback (most recent call last):
>   File "/home/ubuntu/Jul_Dut/connectedhomeip/no/lib/python3.10/site-packages/mobly/base_test.py", line 467, in _teardown_class
>     self.teardown_class()
>   File "/home/ubuntu/Jul_Dut/connectedhomeip/src/python_testing/TC_CCTRL_2_2.py", line 77, in teardown_class
>     self.app_process.send_signal(signal.SIGTERM.value)
> AttributeError: 'TC_CCTRL_2_2' object has no attribute 'app_process'
> [MatterTest] 08-01 09:29:53.654 INFO Summary for test class TC_CCTRL_2_2: Error 2, Executed 0, Failed 0, Passed 0, Requested 1, Skipped 1
> [MatterTest] 08-01 09:29:53.656 INFO Summary for test run MatterTest@08-01-2024_09-29-51-877:
> Total time elapsed 1.7762039420013025s
> Artifacts are saved in "/tmp/matter_testing/logs/MatterTest/08-01-2024_09-29-51-877"
> Test summary saved in "/tmp/matter_testing/logs/MatterTest/08-01-2024_09-29-51-877/test_summary.yaml"

### Bug prevalence

Everytime

### GitHub hash of the SDK that was being used

5d280dedd0bcea407f575aef11e67c8bb7f0a7a1

### Platform

raspi

### Anything else?

**PFA log below :** 
[TC-CCTRL-2.2.txt](https://github.com/user-attachments/files/16455765/TC-CCTRL-2.2.txt)

**Python script reference -** 
https://github.com/project-chip/connectedhomeip/blob/master/src/python_testing/TC_CCTRL_2_2.py

Rajashreekalmane, 2024-08-01:
[CERT-TEST-FAILURE] TC-MCORE-FS-1.1 - Failed due to multiple AttributeErrors during the setup_class, on_fail, and teardown_class methods. Specifically, the MatterTestConfig object lacks the user_params attribute, and the TC_MCORE_FS_1_1 object lacks the test_start_time and app_process attributes. These issues prevent the test case from executing as expected.

> [MatterTest] 08-01 12:04:44.559 INFO Commissioning complete
> [MatterTest] 08-01 12:04:44.561 INFO <<< [E:33796i S:32726 M:256992684 (Ack:232273237)] (S) Msg TX to 1:0000000012344321 [743A] [UDP:[fe80::a70c:61dc:df51:6945%eth0]:5540] --- Type 0000:10 (SecureChannel:StandaloneAck) (B:34)
> [MatterTest] 08-01 12:04:44.562 INFO [Test] test_run_commissioning PASS
> [MatterTest] 08-01 12:04:44.567 INFO Summary for test class CommissionDeviceTest: Error 0, Executed 1, Failed 0, Passed 1, Requested 1, Skipped 0
> [MatterTest] 08-01 12:04:44.572 INFO ==========> TC_MCORE_FS_1_1 <==========
> [MatterTest] 08-01 12:04:44.576 ERROR Error in TC_MCORE_FS_1_1#setup_class.
> Traceback (most recent call last):
>   File "/home/ubuntu/Jul_Dut/connectedhomeip/no/lib/python3.10/site-packages/mobly/base_test.py", line 418, in _setup_class
>     self.setup_class()
>   File "/home/ubuntu/Jul_Dut/connectedhomeip/src/python_testing/matter_testing_support.py", line 1671, in async_runner
>     return _async_runner(body, self, *args, **kwargs)
>   File "/home/ubuntu/Jul_Dut/connectedhomeip/src/python_testing/matter_testing_support.py", line 1659, in _async_runner
>     return asyncio.run(runner_with_timeout)
>   File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
>     return loop.run_until_complete(main)
>   File "/usr/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
>     return future.result()
>   File "/usr/lib/python3.10/asyncio/tasks.py", line 445, in wait_for
>     return fut.result()
>   File "/home/ubuntu/Jul_Dut/connectedhomeip/src/python_testing/TC_MCORE_FS_1_1.py", line 40, in setup_class
>     app = self.matter_test_config.user_params.get("th_server_app_path", None)
> AttributeError: 'MatterTestConfig' object has no attribute 'user_params'
> [MatterTest] 08-01 12:04:44.583 ERROR Exception happened when executing on_fail for setup_class.
> Traceback (most recent call last):
>   File "/home/ubuntu/Jul_Dut/connectedhomeip/no/lib/python3.10/site-packages/mobly/base_test.py", line 418, in _setup_class
>     self.setup_class()
>   File "/home/ubuntu/Jul_Dut/connectedhomeip/src/python_testing/matter_testing_support.py", line 1671, in async_runner
>     return _async_runner(body, self, *args, **kwargs)
>   File "/home/ubuntu/Jul_Dut/connectedhomeip/src/python_testing/matter_testing_support.py", line 1659, in _async_runner
>     return asyncio.run(runner_with_timeout)
>   File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
>     return loop.run_until_complete(main)
>   File "/usr/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
>     return future.result()
>   File "/usr/lib/python3.10/asyncio/tasks.py", line 445, in wait_for
>     return fut.result()
>   File "/home/ubuntu/Jul_Dut/connectedhomeip/src/python_testing/TC_MCORE_FS_1_1.py", line 40, in setup_class
>     app = self.matter_test_config.user_params.get("th_server_app_path", None)
> AttributeError: 'MatterTestConfig' object has no attribute 'user_params'
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "/home/ubuntu/Jul_Dut/connectedhomeip/no/lib/python3.10/site-packages/mobly/base_test.py", line 646, in _exec_procedure_func
>     func(copy.deepcopy(tr_record))
>   File "/home/ubuntu/Jul_Dut/connectedhomeip/no/lib/python3.10/site-packages/mobly/base_test.py", line 562, in _on_fail
>     self.on_fail(record)
>   File "/home/ubuntu/Jul_Dut/connectedhomeip/src/python_testing/matter_testing_support.py", line 1054, in on_fail
>     test_duration = (datetime.now(timezone.utc) - self.test_start_time) / timedelta(microseconds=1)
> AttributeError: 'TC_MCORE_FS_1_1' object has no attribute 'test_start_time'
> [MatterTest] 08-01 12:04:44.606 WARNING Stopping app with SIGTERM
> [MatterTest] 08-01 12:04:44.606 ERROR Error encountered in teardown_class.
> Traceback (most recent call last):
>   File "/home/ubuntu/Jul_Dut/connectedhomeip/no/lib/python3.10/site-packages/mobly/base_test.py", line 467, in _teardown_class
>     self.teardown_class()
>   File "/home/ubuntu/Jul_Dut/connectedhomeip/src/python_testing/TC_MCORE_FS_1_1.py", line 68, in teardown_class
>     self.app_process.send_signal(signal.SIGTERM.value)
> AttributeError: 'TC_MCORE_FS_1_1' object has no attribute 'app_process'
> [MatterTest] 08-01 12:04:44.612 INFO Summary for test class TC_MCORE_FS_1_1: Error 2, Executed 0, Failed 0, Passed 0, Requested 1, Skipped 1
> [MatterTest] 08-01 12:04:44.614 INFO Summary for test run MatterTest@08-01-2024_12-04-42-831:
> Total time elapsed 1.7809680819991627s
> Artifacts are saved in "/tmp/matter_testing/logs/MatterTest/08-01-2024_12-04-42-831"
> Test summary saved in "/tmp/matter_testing/logs/MatterTest/08-01-2024_12-04-42-831/test_summary.yaml"

**PFA log below :** 
[TC_MCORE_FS_1_1.txt](https://github.com/user-attachments/files/16455987/TC_MCORE_FS_1_1.txt)

TH - python controller
commands used to build -

- source scripts/activate.sh && scripts/build_python.sh -m platform -d true -i no
- source no/bin/activate

DUT - fabric-bridge-app

- ./scripts/examples/gn_build_example.sh examples/fabric-bridge-app/linux out/debug/standalone chip_config_network_layer_ble=false 'import("//with_pw_rpc.gni")'

joyoungjun, 2024-08-14:
@Rajashreekalmane HI, is there any update or solution on this?

cecille, 2024-08-22:
This was fixed by https://github.com/project-chip/connectedhomeip/pull/34674

closed 2024-08-22 by cecille