project-chip/connectedhomeip #34719: Error in building RVC app example through gn-build.sh

w8floosh, 2024-08-01:
**Feature area**
Building

**Platform & package versions**
EndeavourOS (linux 6.10.2.arch1-1)
gcc 14.1.1+r309+gbb34b7eda1f-1
ninja 1.12.1-1

**Version used (commit number)**
[c7f5f65](https://github.com/project-chip/connectedhomeip/commit/c7f5f65cb21e11af81626012e2be2dcb4c77ccf2)

**Error description**
Trying to build the whole source code from scratch via gn-build.sh, I get the following output from ninja:

```[10909/11742] c++ standalone/obj/examples/rvc-app/rvc-common/src/chip-rvc-app.rvc-service-area-delegate.cpp.o
FAILED: standalone/obj/examples/rvc-app/rvc-common/src/chip-rvc-app.rvc-service-area-delegate.cpp.o 
g++ -MMD -MF standalone/obj/examples/rvc-app/rvc-common/src/chip-rvc-app.rvc-service-area-delegate.cpp.o.d -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -Wall -Werror -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wundef -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -Wno-cast-function-type -Wno-psabi -Wno-maybe-uninitialized -fdiagnostics-color -fno-strict-aliasing -fmacro-prefix-map=../../= -I/usr/include/gio-unix-2.0 -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/sysprof-6 -std=gnu++17 -fno-rtti -Wnon-virtual-dtor -DCHIP_HAVE_CONFIG_H=1 -DCHIP_MINMDNS_USE_EPHEMERAL_UNICAST_PORT=1 -DCHIP_MINMDNS_HIGH_VERBOSITY=0 -DCHIP_MINMDNS_DEFAULT_POLICY=1 -I../../examples/rvc-app/linux/include -I../../examples/rvc-app/rvc-common/include -I../../examples/platform/linux -I../../src/include -I../../src -Istandalone/gen/include -I../../config/standalone -I../../zzz_generated/app-common -I../../third_party/nlassert/repo/include -I../../third_party/nlio/repo/include -I../../third_party/nlfaultinjection/include -I../../third_party/inipp/repo/inipp -Istandalone/gen/examples/rvc-app/rvc-common -Istandalone/gen/examples/rvc-app/rvc-common/zapgen -I../../third_party/jsoncpp/repo/include -c ../../examples/rvc-app/rvc-common/src/rvc-service-area-delegate.cpp -o standalone/obj/examples/rvc-app/rvc-common/src/chip-rvc-app.rvc-service-area-delegate.cpp.o
In file included from ../../examples/rvc-app/rvc-common/src/rvc-service-area-delegate.cpp:19:
../../examples/rvc-app/rvc-common/include/rvc-service-area-delegate.h:38:10: error: ‘vector’ in namespace ‘std’ does not name a template type
   38 |     std::vector<ServiceArea::LocationStructureWrapper> mSupportedLocations;
      |          ^~~~~~
../../examples/rvc-app/rvc-common/include/rvc-service-area-delegate.h:24:1: note: ‘std::vector’ is defined in header ‘<vector>’; this is probably fixable by adding ‘#include <vector>’
   23 | #include <cstring>
  +++ |+#include <vector>
   24 | #include <utility>
../../examples/rvc-app/rvc-common/include/rvc-service-area-delegate.h:39:10: error: ‘vector’ in namespace ‘std’ does not name a template type
   39 |     std::vector<ServiceArea::MapStructureWrapper> mSupportedMaps;
      |          ^~~~~~
../../examples/rvc-app/rvc-common/include/rvc-service-area-delegate.h:39:5: note: ‘std::vector’ is defined in header ‘<vector>’; this is probably fixable by adding ‘#include <vector>’
   39 |     std::vector<ServiceArea::MapStructureWrapper> mSupportedMaps;
      |     ^~~
../../examples/rvc-app/rvc-common/include/rvc-service-area-delegate.h:40:10: error: ‘vector’ in namespace ‘std’ does not name a template type
   40 |     std::vector<uint32_t> mSelectedLocations;
      |          ^~~~~~
../../examples/rvc-app/rvc-common/include/rvc-service-area-delegate.h:40:5: note: ‘std::vector’ is defined in header ‘<vector>’; this is probably fixable by adding ‘#include <vector>’
   40 |     std::vector<uint32_t> mSelectedLocations;
      |     ^~~
../../examples/rvc-app/rvc-common/include/rvc-service-area-delegate.h:41:10: error: ‘vector’ in namespace ‘std’ does not name a template type
   41 |     std::vector<ServiceArea::Structs::ProgressStruct::Type> mProgressList;
      |          ^~~~~~
../../examples/rvc-app/rvc-common/include/rvc-service-area-delegate.h:41:5: note: ‘std::vector’ is defined in header ‘<vector>’; this is probably fixable by adding ‘#include <vector>’
   41 |     std::vector<ServiceArea::Structs::ProgressStruct::Type> mProgressList;
      |     ^~~
../../examples/rvc-app/rvc-common/src/rvc-service-area-delegate.cpp: In member function ‘virtual uint32_t chip::app::Clusters::ServiceArea::RvcServiceAreaDelegate::GetNumberOfSupportedLocations()’:
../../examples/rvc-app/rvc-common/src/rvc-service-area-delegate.cpp:108:34: error: ‘mSupportedLocations’ was not declared in this scope; did you mean ‘AddSupportedLocation’?
  108 |     return static_cast<uint32_t>(mSupportedLocations.size());
      |                                  ^~~~~~~~~~~~~~~~~~~
      |                                  AddSupportedLocation
../../examples/rvc-app/rvc-common/src/rvc-service-area-delegate.cpp:108:34: note: maximum limit of 1000 namespaces searched for ‘mSupportedLocations’
../../examples/rvc-app/rvc-common/src/rvc-service-area-delegate.cpp: In member function ‘virtual bool chip::app::Clusters::ServiceArea::RvcServiceAreaDelegate::GetSupportedLocationByIndex(uint32_t, chip::app::Clusters::ServiceArea::LocationStructureWrapper&)’:
../../examples/rvc-app/rvc-common/src/rvc-service-area-delegate.cpp:113:21: error: ‘mSupportedLocations’ was not declared in this scope; did you mean ‘aSupportedLocation’?
  113 |     if (listIndex < mSupportedLocations.size())
      |                     ^~~~~~~~~~~~~~~~~~~
      |                     aSupportedLocation
../../examples/rvc-app/rvc-common/src/rvc-service-area-delegate.cpp:113:21: note: maximum limit of 1000 namespaces searched for ‘mSupportedLocations’
../../examples/rvc-app/rvc-common/src/rvc-service-area-delegate.cpp: In member function ‘virtual bool chip::app::Clusters::ServiceArea::RvcServiceAreaDelegate::GetSupportedLocationById(uint32_t, uint32_t&, chip::app::Clusters::ServiceArea::LocationStructureWrapper&)’:
../../examples/rvc-app/rvc-common/src/rvc-service-area-delegate.cpp:130:24: error: ‘mSupportedLocations’ was not declared in this scope; did you mean ‘aSupportedLocation’?
  130 |     while (listIndex < mSupportedLocations.size())
      |                        ^~~~~~~~~~~~~~~~~~~
      |                        aSupportedLocation
../../examples/rvc-app/rvc-common/src/rvc-service-area-delegate.cpp:130:24: note: maximum limit of 1000 namespaces searched for ‘mSupportedLocations’
../../examples/rvc-app/rvc-common/src/rvc-service-area-delegate.cpp: In member function ‘virtual bool chip::app::Clusters::ServiceArea::RvcServiceAreaDelegate::AddSupportedLocation(const chip::app::Clusters::ServiceArea::LocationStructureWrapper&, uint32_t&)’:
../../examples/rvc-app/rvc-common/src/rvc-service-area-delegate.cpp:150:9: error: ‘mSupportedLocations’ was not declared in this scope; did you mean ‘AddSupportedLocation’?
  150 |     if (mSupportedLocations.size() < kMaxNumSupportedLocations)
      |         ^~~~~~~~~~~~~~~~~~~
      |         AddSupportedLocation
../../examples/rvc-app/rvc-common/src/rvc-service-area-delegate.cpp:150:9: note: maximum limit of 1000 namespaces searched for ‘mSupportedLocations’
../../examples/rvc-app/rvc-common/src/rvc-service-area-delegate.cpp:154:43: error: ‘mSupportedMaps’ was not declared in this scope; did you mean ‘AddSupportedMap’?
  154 |         listIndex = static_cast<uint32_t>(mSupportedMaps.size()) - 1; // new element is last in list
      |                                           ^~~~~~~~~~~~~~
[... 103 lines cut ...]
../../examples/rvc-app/rvc-common/src/rvc-service-area-delegate.cpp:383:9: note: maximum limit of 1000 namespaces searched for ‘mProgressList’
../../examples/rvc-app/rvc-common/src/rvc-service-area-delegate.cpp: In member function ‘virtual bool chip::app::Clusters::ServiceArea::RvcServiceAreaDelegate::ClearProgress()’:
../../examples/rvc-app/rvc-common/src/rvc-service-area-delegate.cpp:405:10: error: ‘mProgressList’ was not declared in this scope
  405 |     if (!mProgressList.empty())
      |          ^~~~~~~~~~~~~
../../examples/rvc-app/rvc-common/src/rvc-service-area-delegate.cpp:405:10: note: maximum limit of 1000 namespaces searched for ‘mProgressList’
At global scope:
cc1plus: note: unrecognized command-line option ‘-Wno-unknown-warning-option’ may have been intended to silence earlier diagnostics
[... cut ...] ssors.cpp.o
ninja: build stopped: subcommand failed.

real    49m22,965s
user    334m56,918s
sys     23m0,151s
```

As per compiler's tip, I tried to add

```#include <vector>``` to `rvc-service-area-delegate.h`

and the build completes flawlessly with no errors.

I'm currently a newcomer in Matter code understanding, but I think the build shall succeed without this addition. May this be an oversight in the example code.

Thank you in advance for any help. Feel free to ask more details if needed

hicklin, 2024-08-02:
We did think that this needed `#include <vector>`, but it was working without it, probably due to some convoluted include magic, so we opted to leave this out. I'll add it in as I'm still submitting PRs to update this example to match the latest spec.

hicklin, 2024-08-02:
I have added this as part of [this PR](https://github.com/project-chip/connectedhomeip/pull/34738).

andy31415, 2024-10-21:
Closing as a PR was merged for this.

closed 2024-10-21 by andy31415