project-chip/connectedhomeip #34725: [BUG] Chip-Cert-Bins docker build failing on TOT

raju-apple, 2024-08-02:
### Reproduction steps

Chip-Cert-bins docker container failing getting built, this is the SDK version used : 

SDK SHA used : d0d91272068f267cf880f9d56787ca28da885673

Build command used : 
`docker buildx build --build-arg COMMITHASH=d0d91272068f267cf880f9d56787ca28da885673 --platform linux/arm64 --tag connectedhomeip/chip-cert-bins:d0d91272068f267cf880f9d56787ca28da885673 --push .
`

Log snippet of failure : 

```
#0 977.8 2024-08-02 00:09:23 INFO    [545/735] c++ obj/commands/storage/chip-tool-utils.StorageManagementCommand.cpp.o
#0 989.2 2024-08-02 00:09:35 INFO    [546/735] c++ obj/chip-tool.main.cpp.o
#0 989.2 2024-08-02 00:09:35 INFO    FAILED: obj/chip-tool.main.cpp.o 
#0 989.2 2024-08-02 00:09:35 INFO    aarch64-linux-gnu-g++ -MMD -MF obj/chip-tool.main.cpp.o.d -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -Wall -Werror -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wundef -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -Wno-cast-function-type -Wno-psabi -Wno-maybe-uninitialized -fdiagnostics-color -fno-strict-aliasing -Wconversion -Wno-format-nonliteral -Wno-format-nonliteral -fmacro-prefix-map=../../examples/chip-tool/third_party/connectedhomeip/= -D_REENTRANT -I/usr/include/glib-2.0 -I/usr/lib/aarch64-linux-gnu/glib-2.0/include -I/usr/include/gio-unix-2.0 -I/usr/include/libmount -I/usr/include/blkid -std=gnu++17 -fno-rtti -Wnon-virtual-dtor -DCONFIG_USE_SEPARATE_EVENTLOOP=true -DCONFIG_USE_INTERACTIVE_MODE=true -DCONFIG_USE_LOCAL_STORAGE -DCHIP_HAVE_CONFIG_H=1 -I../../examples/chip-tool -I../../examples/chip-tool/third_party/connectedhomeip/examples/common -I../../examples/chip-tool/third_party/connectedhomeip/zzz_generated/app-common/app-common -I../../examples/chip-tool/third_party/connectedhomeip/zzz_generated/chip-tool -I../../examples/chip-tool/third_party/connectedhomeip/src/lib -I../../examples/chip-tool/third_party/connectedhomeip/examples/common/tracing -I../../examples/chip-tool/third_party/connectedhomeip/src/tracing/perfetto/include -I../../examples/chip-tool/third_party/connectedhomeip/third_party/perfetto/repo/sdk -I../../examples/chip-tool/third_party/connectedhomeip/src/include -I../../examples/chip-tool/third_party/connectedhomeip/src -Igen/include -I../../examples/chip-tool/third_party/connectedhomeip/examples/chip-tool/include -I../../examples/chip-tool/third_party/connectedhomeip/config/standalone -I../../examples/chip-tool/third_party/connectedhomeip/zzz_generated/app-common -I../../examples/chip-tool/third_party/connectedhomeip/third_party/nlassert/repo/include -I../../examples/chip-tool/third_party/connectedhomeip/third_party/nlio/repo/include -I../../examples/chip-tool/third_party/connectedhomeip/third_party/nlfaultinjection/include -I../../examples/chip-tool/third_party/connectedhomeip/third_party/inipp/repo/inipp -Igen/third_party/connectedhomeip/src/controller/data_model -Igen/third_party/connectedhomeip/src/controller/data_model/zapgen -I../../examples/chip-tool/third_party/connectedhomeip/third_party/jsoncpp/repo/include -c ../../examples/chip-tool/main.cpp -o obj/chip-tool.main.cpp.o
#0 989.2 2024-08-02 00:09:35 INFO    aarch64-linux-gnu-g++: fatal error: Killed signal terminated program cc1plus
#0 989.2 2024-08-02 00:09:35 INFO    compilation terminated.

```

Full log : 
[Docker_failure_chip-cert-bins.txt](https://github.com/user-attachments/files/16463683/Docker_failure_chip-cert-bins.txt)

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

d0d91272068f267cf880f9d56787ca28da885673

### Platform

raspi

### Anything else?

Point of reference is , this build worked fine on the SDK SHA : [ccd8da9764597a1f48f712c8eb0098f1d3c51b6e] just a couple of days ago

raju-apple, 2024-08-02:
Tried re-running it and went past this spot but failed at this line : 

```
#0 153.4 Err:130 http://ports.ubuntu.com/ubuntu-ports noble-updates/universe arm64 openjdk-8-jre-headless arm64 8u412-ga-1~24.04.2
#0 153.4   404  Not Found [IP: 185.125.190.39 80]
#0 153.6 Err:131 http://ports.ubuntu.com/ubuntu-ports noble-updates/universe arm64 openjdk-8-jre arm64 8u412-ga-1~24.04.2
#0 153.6   404  Not Found [IP: 185.125.190.39 80]
#0 153.6 Err:132 http://ports.ubuntu.com/ubuntu-ports noble-updates/universe arm64 openjdk-8-jdk-headless arm64 8u412-ga-1~24.04.2
#0 153.6   404  Not Found [IP: 185.125.190.39 80]
#0 153.7 Err:133 http://ports.ubuntu.com/ubuntu-ports noble-updates/universe arm64 openjdk-8-jdk arm64 8u412-ga-1~24.04.2
#0 153.7   404  Not Found [IP: 185.125.190.39 80]
#0 153.8 Fetched 65.5 MB in 2min 33s (429 kB/s)
#0 153.8 E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/universe/o/openjdk-8/openjdk-8-jre-headless_8u412-ga-1%7e24.04.2_arm64.deb  404  Not Found [IP: 185.125.190.39 80]
#0 153.8 E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/universe/o/openjdk-8/openjdk-8-jre_8u412-ga-1%7e24.04.2_arm64.deb  404  Not Found [IP: 185.125.190.39 80]
#0 153.8 E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/universe/o/openjdk-8/openjdk-8-jdk-headless_8u412-ga-1%7e24.04.2_arm64.deb  404  Not Found [IP: 185.125.190.39 80]
#0 153.8 E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/universe/o/openjdk-8/openjdk-8-jdk_8u412-ga-1%7e24.04.2_arm64.deb  404  Not Found [IP: 185.125.190.39 80]
#0 153.8 E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
------
Dockerfile:301
--------------------

```

Full file here : 
[docker_failure_chip_cert_jdk.txt](https://github.com/user-attachments/files/16464107/docker_failure_chip_cert_jdk.txt)

Alami-Amine, 2024-10-16:
Issue is outdated --> More recent Chip-Cert-bins docker builds have been passing.