project-chip/connectedhomeip #34793: [1.3] Issues about Thermostat cluster

agatah2333, 2024-08-06:
### Reproduction steps

1. [image]
https://github.com/project-chip/connectedhomeip/blob/v1.3.0.0/src/app/zap-templates/zcl/data-model/chip/thermostat-cluster.xml
[image]
Attribute Occupancy Should be OccupancyBitmap: The attribute Occupancy should be renamed to OccupancyBitmap, but the definition of this type is missing.

2. HVACSystemTypeBitmap Defined but Unused: The HVACSystemTypeBitmap is defined but never used. It should be associated with constraint 0x0009.

3. Attribute Types for IDs 0x0011-0x0018 Should be Temperature: The types of attributes with IDs from 0x0011 to 0x0018 should be temperature.
[image]

### Bug prevalence

each time

### GitHub hash of the SDK that was being used

561d23d0db215a99705ff0696e73853c8edf11b2 

### Platform

core

### Platform Version(s)

1.3

### Type

Spec Compliance Issue

hasty, 2024-08-06:
I believe these were all fixed in #34570

agatah2333, 2024-08-06:
> I believe these were all fixed in #34570

Will the fixed MR be picked up in branch 1.3?

bzbarsky-apple, 2024-08-06:
No, this sort of work will not get backported to stable branches.

closed 2024-08-06 by bzbarsky-apple