project-chip/connectedhomeip #34794: [1.3] Out-of-constrain error in Door Lock cluster

agatah2333, 2024-08-06:
### Reproduction steps

1. In the SetUser command, the constraint on field 1 is 1 to NumberOfTotalUsersSupported, and the constraint on field 5 is 0-9. However, if you try the command with 0 in field 1 and 10 in field 5, the implementation code will not reject it.

[image]

2. In the GetUser command, the constraint on field 0 is 1 to NumberOfTotalUsersSupported. However, if you try the command with 0 in field 0, the implementation code will not reject it.

3. In the ClearUser command, the constraint on field 0 is 1 to NumberOfTotalUsersSupported. However, if you try the command with 0 in field 0, the implementation code will not reject it.

### Bug prevalence

each time

### GitHub hash of the SDK that was being used

561d23d0db215a99705ff0696e73853c8edf11b2 

### Platform

core

### Platform Version(s)

1.3

### Type

Spec Compliance Issue, Security Issue

bzbarsky-apple, 2024-08-06:
> In the SetUser command, the constraint on field 1 is 1 to NumberOfTotalUsersSupported, and the constraint on field 5 is 0-9. However, if you try the command with 0 in field 1 and 10 in field 5, the implementation code will not reject it.

0 in field 1 is absolutely rejected on tip.  You get INVALID_COMMAND if you try to do that.  See `DoorLockServer::userIndexValid` and the call to it in `DoorLockServer::setUserCommandHandler`.

10 in field 5 is likewise rejected; `DoorLockServer::setUserCommandHandler` has:

```
    if (userType == UserTypeEnum::kUnknownEnumValue)
    {
        ChipLogProgress(Zcl, "[SetUser] Unable to set the user: user type is unknown [endpointId=%d,userIndex=%d,userType=%u]",
                        commandPath.mEndpointId, userIndex, to_underlying(userType.Value()));

        sendClusterResponse(commandObj, commandPath, ClusterStatusCode(Status::InvalidCommand));
        return;
    }
```

> In the GetUser command, the constraint on field 0 is 1 to NumberOfTotalUsersSupported. However, if you try the command with 0 in field 0, the implementation code will not reject it.

`DoorLockServer::getUserCommandHandler` also has a `userIndexValid` check.  And the command is rejected.

> In the ClearUser command, the constraint on field 0 is 1 to NumberOfTotalUsersSupported. However, if you try the command with 0 in field 0, the implementation code will not reject it.

Again, this is rejected, both by looking at the code and by jus trying it.

I have no idea where the claims in this issue come from.

closed 2024-08-06 by bzbarsky-apple

agatah2333, 2024-08-13:
@bzbarsky-apple 
please try the following command with the virtual door lock app(0x654324):

chip-tool any command-by-id 0x0101 0x1a '{"0":3,"1":11}' 0x654324 1 --timedInteractionTimeoutMs 3