project-chip/connectedhomeip #34799: [CERT-TEST-FAILURE] [TC-ICDB-1.3] Test failed

Rajashreekalmane, 2024-08-06:
### Feature Area

Area:ICDPhase2

### Test Case

TC-ICDB-1.3

### Reproduction steps

 At step 2b. According to the test plan, when the DUT receives a Check-In message with a Check-In Counter ICDCounter2 indicating that 2^31 counter values have been used, the DUT should refresh its entry with a new key Key2 via the RegisterClient command to the TH with the following details:
CheckInNodeID: CheckInNodeID
MonitoredSubject: MonitoredSubID
Key: Key2
However, in the current SDK, TH receiving the status 0x7f (UNSUPPORTED_ENDPOINT).

Command used to launch the DUT :./lit-icd-app --enable-key 00112233445566778899AABBCCDDEEFF
Command used to launch the TH(chip-tool) : ./chip-tool pairing onnetwork 1 20202021
After the successful pairing entered to  ./chip-tool interactive start
step 1 : TH sends {C_REGISTER_CLIENT} command.
- CheckInNodeID: CheckInNodeID
- MonitoredSubject: MonitoredSubID
- Key: Key1
  icdmanagement register-client 112233 1 hex:abcdef1234567890abcdef1234567890 0 1 0
> [1722922531.490] [11122:11124] [DMG] Received Command Response Data, Endpoint=0 Cluster=0x0000_0046 Command=0x0000_0001
> [1722922531.490] [11122:11124] [TOO] Endpoint: 0 Cluster: 0x0000_0046 Command 0x0000_0001
> [1722922531.490] [11122:11124] [TOO]   RegisterClientResponse: {
> [1722922531.490] [11122:11124] [TOO]     ICDCounter: 579832728
> [1722922531.490] [11122:11124] [TOO]    }
> [1722922531.491] [11122:11124] [DMG] ICR moving to [AwaitingDe]

step 2a : TH sends TestEventTrigger command to General Diagnostics Cluster on Endpoint 0 with EnableKey field set to PIXIT.ICDB.S.TEST_EVENT_TRIGGER_KEY and EventTrigger field set to PIXIT.ICDB.S.TEST_EVENT_TRIGGER for Invalidate ICD half counter values event.
generaldiagnostics test-event-trigger hex:00112233445566778899AABBCCDDEEFF 0x0046000000000003 1 0
> [1722922538.436] [11122:11124] [DMG] Received Command Response Status for Endpoint=0 Cluster=0x0000_0033 Command=0x0000_0000 Status=0x0
> [1722922538.436] [11122:11124] [DMG] ICR moving to [AwaitingDe]

**waited DUT refresh its entry with a new key Key2 via {C_REGISTER_CLIENT} command to TH: - CheckInNodeID: CheckInNodeID
- MonitoredSubject: MonitoredSubID
- Key: Key2**
**However, in the current SDK, TH receiving the status 0x7f (UNSUPPORTED_ENDPOINT).** 

> [1722922559.028] [11122:11124] [DMG] },
> [1722922559.028] [11122:11124] [DMG] No command 0x0000_0000 in Cluster 0x0000_0046 on Endpoint 0x0
> [1722922559.028] [11122:11124] [DMG] Endpoint=0 Cluster=0x0000_0046 Command=0x0000_0000 status 0x7f (UNSUPPORTED_ENDPOINT) (no additional context)
> [1722922559.028] [11122:11124] [DMG] Command handler moving to [NewRespons]

### Bug prevalence

Everytime

### GitHub hash of the SDK that was being used

b9c192b7cf8e7ffc697fa9b25eaeb6c5faa09365

### Platform

raspi

### Anything else?

**PFA Log below** 
[TC-ICDB-1.3.txt](https://github.com/user-attachments/files/16505657/TC-ICDB-1.3.txt)
[TC-ICDB-1.3 DUT.txt](https://github.com/user-attachments/files/16505658/TC-ICDB-1.3.DUT.txt)

**Test plan reference** 
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/icdbehavior.adoc#tc-icdb-1-3-icd-check-in-protocol-client-response-dut_client

Rajashreekalmane, 2024-08-06:
Raised Test plan issue for inconsistencies in the labels, PICS defined in the Purpose, Required Devices, and Test Steps: https://github.com/CHIP-Specifications/chip-test-plans/issues/4420

yinyihu-silabs, 2024-08-21:
Test event trigger itself works.
for step 2b. there is no commands to issue. after the TestEventTrigger command, client should receives a ICD counter of 2^31; as in the chip-tool log
```
[1724091971.716] [71367:628002] [ICD] Check In Message processing complete: start_counter=193333958 offset=2147483648 nodeid=<000000000001B669, 1>
```

in case of step 3, client should  discard the check in message, as a chip-tool example
```
[1724092461.073] [75279:642457] [EM] Flushed pending ack for MessageCounter:79848502 on exchange 11856i
[1724092468.802] [75279:642607] [EM] >>> [E:62109r S:0 M:23635683] (U) Msg RX from 0:A8ED2DD45B2A71A7 [0000] --- Type 0000:50 (SecureChannel:ICD_CheckInMessage) (B:57)
[1724092468.802] [75279:642607] [EM] Handling via exchange: 62109r, Delegate: 0x103b013c8
[1724092468.803] [75279:642607] [ICD] A duplicate check-in message was received and discarded
```

The unsupported endpoint log seems to be related to an register client response.
```
[1722921929.969] [10536:10538] [DMG] InvokeRequestMessage =
[1722921929.969] [10536:10538] [DMG] {
[1722921929.969] [10536:10538] [DMG] 	suppressResponse = false, 
[1722921929.969] [10536:10538] [DMG] 	timedRequest = false, 
[1722921929.969] [10536:10538] [DMG] 	InvokeRequests =
[1722921929.969] [10536:10538] [DMG] 	[
[1722921929.969] [10536:10538] [DMG] 		CommandDataIB =
[1722921929.969] [10536:10538] [DMG] 		{
[1722921929.969] [10536:10538] [DMG] 			CommandPathIB =
[1722921929.969] [10536:10538] [DMG] 			{
[1722921929.969] [10536:10538] [DMG] 				EndpointId = 0x0,
[1722921929.969] [10536:10538] [DMG] 				ClusterId = 0x46,
[1722921929.969] [10536:10538] [DMG] 				CommandId = 0x0,
[1722921929.969] [10536:10538] [DMG] 			},
[1722921929.969] [10536:10538] [DMG] 			
[1722921929.969] [10536:10538] [DMG] 			CommandFields = 
[1722921929.969] [10536:10538] [DMG] 			{
[1722921929.969] [10536:10538] [DMG] 				0x0 = 112233 (unsigned), 
[1722921929.969] [10536:10538] [DMG] 				0x1 = 1 (unsigned), 
[1722921929.969] [10536:10538] [DMG] 				0x2 = [
[1722921929.969] [10536:10538] [DMG] 						
[1722921929.969] [10536:10538] [DMG] 				] (0 bytes)
[1722921929.969] [10536:10538] [DMG] 				0x4 = 0 (unsigned), 
[1722921929.969] [10536:10538] [DMG] 			},
[1722921929.969] [10536:10538] [DMG] 		},
[1722921929.969] [10536:10538] [DMG] 		
[1722921929.969] [10536:10538] [DMG] 	],
[1722921929.969] [10536:10538] [DMG] 	
[1722921929.969] [10536:10538] [DMG] 	InteractionModelRevision = 11
[1722921929.969] [10536:10538] [DMG] },
[1722921929.969] [10536:10538] [DMG] No command 0x0000_0000 in Cluster 0x0000_0046 on Endpoint 0x0
[1722921929.969] [10536:10538] [DMG] Endpoint=0 Cluster=0x0000_0046 Command=0x0000_0000 status 0x7f (UNSUPPORTED_ENDPOINT) (no additional context)
```

closed 2024-08-30 by yunhanw-google