project-chip/connectedhomeip #34800: [CERT-TEST-FAILURE] [TC-ICDB-1.1] Test Failed - DUT Fails to Send Correct Check-In Counter in Step 4

Rajashreekalmane, 2024-08-06:
### Feature Area

Area:ICDPhase2

### Test Case

[TC-ICDB-1.1] 

### Reproduction steps

In Step 4, after waiting for one or more cycles of IdleModeDuration, the DUT should send a check-in message to the TH after each IdleModeDuration with:
Check-In Counter greater than the ICDCounter in Step 2.
Application Data matching ActiveModeThreshold in Step 1.

**Current SDK Behavior:**
The Check-In Counter is not greater than the ICDCounter in Step 2.

**Logs:**
From the logs, we can see the offset increase which indicates we received a valid OFFSET increasing but There seems to be an issue with the chip-tool in terms of logging.

**Command used to launch the DUT** :./lit-icd-app --enable-key 00112233445566778899AABBCCDDEEFF
**Command used to launch the TH** : ./chip-tool pairing onnetwork 1 20202021
after successful provision validated TC using mentioned commands 
step 1a - TH reads from the DUT the RegisteredClients attribute and make sure RegisteredClients is empty. If not empty, TH sends command UnregisterClient to clear all clients in RegisteredClients.
./chip-tool icdmanagement read registered-clients 1 0

>  [1721442129.426] [2869:2871] [TOO] Endpoint: 0 Cluster: 0x0000_0046 Attribute 0x0000_0003 DataVersion: 1771700755
>  [1721442129.426] [2869:2871] [TOO] RegisteredClients: 0 entries

step 1b - on TH(chip-tool) by sending the below mentioned commands, read the attribute values from DUT and Store the values of IdleModeDuration, ActiveModeDuration, and ActiveModeThreshold.

./chip-tool icdmanagement read idle-mode-duration 1 0

>  [1721442226.931] [2879:2881] [TOO] Endpoint: 0 Cluster: 0x0000_0046 Attribute 0x0000_0000 DataVersion: 1771700755
>  [1721442226.932] [2879:2881] [TOO] IdleModeDuration: 3600

 
 ./chip-tool icdmanagement read active-mode-duration 1 0

>  [1721442264.328] [2883:2885] [TOO] Endpoint: 0 Cluster: 0x0000_0046 Attribute 0x0000_0001 DataVersion: 1771700755
>  [1721442264.328] [2883:2885] [TOO] ActiveModeDuration: 0

 
 ./chip-tool icdmanagement read active-mode-threshold 1 0

>  [1721442301.727] [2886:2888] [TOO] Endpoint: 0 Cluster: 0x0000_0046 Attribute 0x0000_0002 DataVersion: 1771700755
>  [1721442301.727] [2886:2888] [TOO] ActiveModeThreshold: 5000

Step 2 - interactive mode 
./chip-tool interactive start
 >>> icdmanagement register-client 112233 1 hex:abcdef1234567890abcdef1234567890 0 1 0

>  [1721511831.055] [6101:6103] [DMG] Received Command Response Data, Endpoint=0 Cluster=0x0000_0046 Command=0x0000_0001
>  [1721511831.055] [6101:6103] [TOO] Endpoint: 0 Cluster: 0x0000_0046 Command 0x0000_0001 
>  [1721511831.055] [6101:6103] [TOO] RegisterClientResponse: { 
>  [1721511831.055] [6101:6103] [TOO] ICDCounter: 3296975699 
>  [1721511831.055] [6101:6103] [TOO] }

after waiting  for 1 or more cycle of IdleModeDuration, DUT should send check-in message to the TH after each IdleModeDuration with Payload
- Check-In Counter: greater than ICDCounter in Step 2.
- Application Data: matches ActiveModeThreshold in Step 1.

> [1721511896.849] [6101:6103] [EM] >>> [E:45939r S:0 M:34978434] (U) Msg RX from 0:8537A57771E71CF2 [0000] --- Type 0000:50 (SecureChannel:ICD_CheckInMessage) (B:57)
>  [1721511896.849] [6101:6103] [EM] Handling via exchange: 45939r, Delegate: 0xaaaab8a6a370
>  [1721511896.850] [6101:6103] [ICD] Check In Message processing complete: start_counter=3296975699 offset=1 nodeid=<000000000001B669, 1>
>  [1721511896.850] [6101:6103] [TOO] The node <000000000001B669, 1> is not the one we are interested in.
>  [1721511962.942] [6101:6103] [EM] >>> [E:45940r S:0 M:34978435] (U) Msg RX from 0:8D397F5863F36ECE [0000] --- Type 0000:50 (SecureChannel:ICD_CheckInMessage) (B:57)
>  [1721511962.942] [6101:6103] [EM] Handling via exchange: 45940r, Delegate: 0xaaaab8a6a370
>  [1721511962.943] [6101:6103] [ICD] Check In Message processing complete: start_counter=3296975699 offset=2 nodeid=<000000000001B669, 1>

### Bug prevalence

Everytime

### GitHub hash of the SDK that was being used

b9c192b7cf8e7ffc697fa9b25eaeb6c5faa09365

### Platform

raspi

### Anything else?

**Test Plan Reference :** 
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/icdbehavior.adoc#tc-icdb-1-1-icd-check-in-protocol-register-client-idle-mode-duration-dut_server

**PFA Log for your Reference :** 
[TC-ICDB-1.1 DUT.txt](https://github.com/user-attachments/files/16506371/TC-ICDB-1.1.DUT.txt)
[TC-ICDB-1.1 TH.txt](https://github.com/user-attachments/files/16506372/TC-ICDB-1.1.TH.txt)

yinyihu-silabs, 2024-08-15:
check-in message shows `start_counter=3296975699 offset=2` which is incrementing.

Rajashreekalmane, 2024-08-20:
@yinyihu-silabs I noticed that in the logs, the offset is being incremented while the start_counter remains the same:

> [1722926159.113] [12781:12783] [DMG] Received Command Response Data, Endpoint=0 Cluster=0x0000_0046 Command=0x0000_0001
> [1722926159.113] [12781:12783] [TOO] Endpoint: 0 Cluster: 0x0000_0046 Command 0x0000_0001
> [1722926159.113] [12781:12783] [TOO]   RegisterClientResponse: {
> [1722926159.113] [12781:12783] [TOO]     ICDCounter: 3604653456
> [1722926159.113] [12781:12783] [TOO]    }
> [1722926159.114] [12781:12783] [DMG] ICR moving to [AwaitingDe]
> [1722926159.114] [12781:12783] [EM] <<< [E:57814i S:48085 M:124487333 (Ack:3546885)] (S) Msg TX to 1:0000000000000001 [A101] [UDP:[fe80::7d4b:a2b1:f206:c28%enxd03745ce8f62]:5540] --- Type 0000:10 (SecureChannel:StandaloneAck) (B:34)
> [1722926159.114] [12781:12783] [EM] Flushed pending ack for MessageCounter:3546885 on exchange 57814i
> [1722926159.735] [12781:12783] [DL] Avahi group established
> [1722926159.738] [12781:12783] [DL] Avahi group established
> [1722926172.521] [12781:12783] [EM] >>> [E:17407r S:0 M:99088820] (U) Msg RX from 0:915D01878ED108E5 [0000] --- Type 0000:50 (SecureChannel:ICD_CheckInMessage) (B:57)
> [1722926172.521] [12781:12783] [EM] Handling via exchange: 17407r, Delegate: 0x5624fc02fb20
> [1722926172.521] [12781:12783] [ICD] Check In Message processing complete: start_counter=3604653456 offset=1 nodeid=<000000000001B669, 1>
> [1722926172.521] [12781:12783] [TOO] The node <000000000001B669, 1> is not the one we are interested in.
> [1722926182.523] [12781:12783] [EM] >>> [E:17408r S:0 M:99088821] (U) Msg RX from 0:19352498F160C012 [0000] --- Type 0000:50 (SecureChannel:ICD_CheckInMessage) (B:57)
> [1722926182.523] [12781:12783] [EM] Handling via exchange: 17408r, Delegate: 0x5624fc02fb20
> [1722926182.523] [12781:12783] [ICD] Check In Message processing complete: start_counter=3604653456 offset=2 nodeid=<000000000001B669, 1>
> [1722926182.523] [12781:12783] [TOO] The node <000000000001B669, 1> is not the one we are interested in.
> [1722926192.523] [12781:12783] [EM] >>> [E:17409r S:0 M:99088822] (U) Msg RX from 0:3267FD26A44DE27A [0000] --- Type 0000:50

Is it expected behavior for the offset to increase without a corresponding increment in the start_counter?

yinyihu-silabs, 2024-08-20:
yes. client is expected to store starting icd counter and offset.

Rajashreekalmane, 2024-08-21:
Thank you for the clarification. As per the above conversation, this issue has been resolved and can be closed

closed 2024-08-21 by Rajashreekalmane