project-chip/connectedhomeip #34801: [CERT-TEST-FAILURE] TC-ICDB-2.3 :

Rajashreekalmane, 2024-08-06:
### Feature Area

Area:ICDPhase2

### Test Case

TC-ICDB-2.3

### Reproduction steps

**Test Case Step:**
Step 3 expects the DUT to send periodic check-in messages to both TH1 and TH2 with updated ICDCounters. The test plan specifies that both TH1 and TH2 should receive the same ICDCounter.

**Current SDK Behavior:**
The current SDK provides different ICDCounters for TH1 and TH2.

**Logs:**
Test PLan expects same ICDCounter for both, but the actual output indicates different ICDCounters for TH1 and TH2.

Use the below commands to provision the DUT with 2 TH's
execute the below mentioned command to put DUT into a commissionable state, Pls use equivalent command on the respective DUT
`./lit-icd-app`

Once DUT reach the  commissionable state pls send below mentioned command on TH's respectively. Pls use equivalent command on the respective DUT
Provision the device using chip tool on first controller(TH1)
` ./chip-tool pairing onnetwork 1 20202021`
`./chip-tool pairing open-commissioning-window 1 1 400 2000 3840 `
Provision the device using chip tool on 2nd  controller(TH2)
`  ./chip-tool pairing code 2 36453420312(mannualcode) --commissioner-name beta`

On TH1 - 
**Step 2a -** Verify that after TH1 sends the RegisterClient command with the following parameters:
CheckInNodeID: CheckInNodeID1 = NodeId1
MonitoredSubject: MonitoredSubID1
Key: Key1
The DUT should respond with a status of SUCCESS (0x00). Additionally, confirm that the DUT’s response includes the ICDCounter.
`./chip-tool interactive start`\ icdmanagement register-client 112233 1 hex:abcdef1234567890abcdef1234567890 0 1 0
`
> [1721997295.344] [41279:41281] [DMG] },
> [1721997295.344] [41279:41281] [DMG] Received Command Response Data, Endpoint=0 Cluster=0x0000_0046 Command=0x0000_0001
> [1721997295.344] [41279:41281] [TOO] Endpoint: 0 Cluster: 0x0000_0046 Command 0x0000_0001
> [1721997295.344] [41279:41281] [TOO]   RegisterClientResponse: {
> [1721997295.344] [41279:41281] [TOO]     ICDCounter: 1790469174
> [1721997295.344] [41279:41281] [TOO]    }
> [1721997295.345] [41279:41281] [DMG] ICR moving to [AwaitingDe]

**step 2b**
On TH2 -  Verify that after TH2 sends the RegisterClient command with the following parameters:
CheckInNodeID: CheckInNodeID2 = NodeId2
MonitoredSubject: MonitoredSubID2
Key: Key2
The DUT should respond with a status of SUCCESS (0x00). Additionally, confirm that the DUT’s response includes the ICDCounter.
``./chip-tool interactive start --commissioner-name beta
`
 icdmanagement register-client 112233 2 hex:abcdef1234567890abcdef1234567890 0 2 0 --commissioner-name beta
> [1721997307.443] [51916:51918] [DMG] },
> [1721997307.443] [51916:51918] [DMG] Received Command Response Data, Endpoint=0 Cluster=0x0000_0046 Command=0x0000_0001
> [1721997307.443] [51916:51918] [TOO] Endpoint: 0 Cluster: 0x0000_0046 Command 0x0000_0001
> [1721997307.443] [51916:51918] [TOO]   RegisterClientResponse: {
> [1721997307.443] [51916:51918] [TOO]     ICDCounter: 1790469175
> [1721997307.443] [51916:51918] [TOO]    }

**step 3 - After waiting for one or more IdleModeDuration cycles, verify that the DUT sends periodic check-in messages to both TH1 and TH2 with an updated ICDCounter. Ensure that both TH1 and TH2 receive the same ICDCounter value. The current SDK provides different ICDCounters for TH1 and TH2.**

 On TH1 and

> [1721997301.877] [41279:41281] [EM] >>> [E:26760r S:0 M:5967021] (U) Msg RX from 0:DE30777B3909BA06 [0000] --- Type 0000:50 (SecureChannel:ICD_CheckInMessage) (B:57)
> [1721997301.878] [41279:41281] [EM] Handling via exchange: 26760r, Delegate: 0x557b687e7ac0
> [1721997301.878] [41279:41281] [ICD] Check In Message processing complete: start_counter=1790469174 offset=1 nodeid=<000000000001B669, 1>
> [1721997301.878] [41279:41281] [TOO] The node <000000000001B669, 1> is not the one we are interested in.
> [1721997321.888] [41279:41281] [EM] >>> [E:26761r S:0 M:5967024] (U) Msg RX from 0:3835CD38BEABA128 [0000] --- Type 0000:50 (SecureChannel:ICD_CheckInMessage) (B:57)
> [1721997321.888] [41279:41281] [EM] Handling via exchange: 26761r, Delegate: 0x557b687e7ac0
> [1721997321.888] [41279:41281] [ICD] Check In Message processing complete: start_counter=1790469174 offset=2 nodeid=<000000000001B669, 1>
> [1721997321.888] [41279:41281] [TOO] The node <000000000001B669, 1> is not the one we are interested in.
> [1721997331.898] [41279:41281] [EM] >>> [E:26763r S:0 M:5967026] (U) Msg RX from 0:5C924742068ECCD5 [0000] --- Type 0000:50 (SecureChannel:ICD_CheckInMessage) (B:57)
> [1721997331.898] [41279:41281] [EM] Handling via exchange: 26763r, Delegate: 0x557b687e7ac0
> [1721997331.898] [41279:41281] [ICD] Check In Message processing complete: start_counter=1790469174 offset=3 nodeid=<000000000001B669, 1>

 TH2 

> [1721997321.885] [51916:51918] [EM] >>> [E:26762r S:0 M:5967025] (U) Msg RX from 0:DD22F3AA5C6819AB [0000] --- Type 0000:50 (SecureChannel:ICD_CheckInMessage) (B:57)
> [1721997321.885] [51916:51918] [EM] Handling via exchange: 26762r, Delegate: 0xaaaac09b73f8
> [1721997321.886] [51916:51918] [ICD] Check In Message processing complete: start_counter=1790469175 offset=1 nodeid=<000000000001B669, 1>
> [1721997321.886] [51916:51918] [TOO] The node <000000000001B669, 1> is not the one we are interested in.
> [1721997331.894] [51916:51918] [EM] >>> [E:26764r S:0 M:5967027] (U) Msg RX from 0:D13A96B82506C61D [0000] --- Type 0000:50 (SecureChannel:ICD_CheckInMessage) (B:57)
> [1721997331.895] [51916:51918] [EM] Handling via exchange: 26764r, Delegate: 0xaaaac09b73f8

### Bug prevalence

Everytime

### GitHub hash of the SDK that was being used

 a5124a9e5b8fdaed6d7304d406c0edb2cf4a49dd

### Platform

raspi

### Anything else?

**PFA log below :** 
[TC-ICDB-2.3 DUT.txt](https://github.com/user-attachments/files/16506564/TC-ICDB-2.3.DUT.txt)
[TC-ICDB-2.3 TH2.txt](https://github.com/user-attachments/files/16506568/TC-ICDB-2.3.TH2.txt)
[TC-ICDB-2.3 TH1.txt](https://github.com/user-attachments/files/16506638/TC-ICDB-2.3.TH1.txt)

**Test Plan REference :** 
[image]
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/icdbehavior.adoc#tc-icdb-2-3-icd-state-machine-with-both-client-registrations-and-no-active-subscription-multiple-fabrics-dut_server

yinyihu-silabs, 2024-08-15:
The result is correct.
from your log TH1: -> 1790469174 offset 2 = 1790469176
`[1721997321.888] [41279:41281] [ICD] Check In Message processing complete: start_counter=1790469174 offset=2 nodeid=<000000000001B669, 1>`
At the same time stamp in TH2: 1790469175 offset 1 = 1790469176
`[1721997321.886] [51916:51918] [ICD] Check In Message processing complete: start_counter=1790469175 offset=1 nodeid=<000000000001B669, 1>`

yinyihu-silabs, 2024-08-20:
it was bit odd that you're getting different start counter. i ran a few times and register TH2 after 1 idle duration, and i didn't get he same start_counter.
but overall, if the the end counter value is the same, it's correct.

Rajashreekalmane, 2024-08-29:
Closing this issue as per conversation withy @yinyihu-silabs.

closed 2024-08-29 by Rajashreekalmane