project-chip/connectedhomeip #34803: [CERT-TEST-FAILURE] [TC-ICDB-2.5] -

Rajashreekalmane, 2024-08-06:
### Feature Area

Area:ICDPhase2

### Test Case

TC-ICDB-2.5

### Reproduction steps

Use below commands to provision the DUT with 2 TH's in RPI platform
execute the below mentioned command to put DUT into a commissionable state, Pls use equivalent command on the respective DUT
`./lit-icd-app`

Once DUT reach the commissionable state pls send below mentioned command on TH's respectively. Pls use equivalent command on the respective DUT
Provision the device using chip tool on first controller(TH1)
`./chip-tool pairing onnetwork 1 20202021`
./chip-tool pairing open-commissioning-window 1 1 400 2000 3840
Provision the device using chip tool on 2nd controller(TH2)
` ./chip-tool pairing code 2 36453420312(mannualcode) --commissioner-name beta
`
**at step 3a - Activate the subscription to subscribe to all by sending below mentioned command**

**Activate Subscription from TH1:
>>> any subscribe-by-id 0xFFFFFFFF 0xFFFFFFFF 10 100 1 0
> Ensure the DUT response contains the ICDCounter attribute.
> Verify the MaxInterval values returned for both TH1 and TH2 are within the specified range

**Activate Subscription from TH2:**
 >>> any subscribe-by-id 0xFFFFFFFF 0xFFFFFFFF 30 300 2 0 --commissioner-name beta
Ensure the DUT response contains the ICDCounter attribute.
Verify the MaxInterval values returned for both TH1 and TH2 are within the specified range

**at step 3b - Verify that DUT sends subscription reports to TH1 and TH2 periodically within the MaxInterval1 time and Ensure that the DUT does not send any check-in messages to the client during this period. but currently in SDK,  TH2 is not receiving check-in message but TH1 is receiving  check-in message**
On TH1 

> [1721994226.986] [37494:37496] [EM] >>> [E:10149r S:20730 M:73088039 (Ack:246255576)] (S) Msg RX from 1:0000000000000001 [BE3E] --- Type 0000:10 (SecureChannel:StandaloneAck) (B:18)
> [1721994226.986] [37494:37496] [EM] Found matching exchange: 10149r, Delegate: (nil)
> [1721994226.986] [37494:37496] [EM] Rxd Ack; Removing MessageCounter:246255576 from Retrans Table on exchange 10149r
> [1721994227.182] [37494:37496] [EM] >>> [E:10150r S:0 M:215735937] (U) Msg RX from 0:02A69721601AAB7A [0000] --- Type 0000:50 (SecureChannel:ICD_CheckInMessage) (B:57)
> [1721994227.182] [37494:37496] [EM] Handling via exchange: 10150r, Delegate: 0x55f840420ac0
> [1721994227.183] [37494:37496] [ICD] Check In Message processing complete: start_counter=2449466178 offset=13 nodeid=<000000000001B669, 1>
> [1721994227.183] [37494:37496] [TOO] The node <000000000001B669, 1> is not the one we are interested in.
> [1721994236.989] [37494:37496] [EM] >>> [E:10151r S:20730 M:73088040] (S) Msg RX from 1:0000000000000001 [BE3E] --- Type 0001:05 (IM:ReportData) (B:25)
> [1721994236.989] [37494:37496] [EM] Handling via exchange: 10151r, Delegate: 0x55f840424868
> [1721994236.990] [37494:37496] [DMG] ReportDataMessage =
> [1721994236.990] [37494:37496] [DMG] {
> [1721994236.990] [37494:37496] [DMG] 	SubscriptionId = 0x45ad1aec,
> [1721994236.990] [37494:37496] [DMG] 	InteractionModelRevision = 11
> [1721994236.990] [37494:37496] [DMG] }
> [1721994236.990] [37494:37496] [DMG] Refresh LivenessCheckTime for 76082 milliseconds with SubscriptionId = 0x45ad1aec Peer = 01:0000000000000001
> [1721994236.990] [37494:37496] [EM] <<< [E:10151r S:20730 M:246255577 (Ack:73088040)] (S) Msg TX to 1:0000000000000001 [BE3E] [UDP:[fe80::e65f:1ff:fe49:ae1a%enxd03745ce8f62]:5540] --- Type 0001:01 (IM:StatusResponse) (B:42)

### Bug prevalence

Everytime

### GitHub hash of the SDK that was being used

a5124a9e5b8fdaed6d7304d406c0edb2cf4a49dd

### Platform

raspi

### Anything else?

**Test Plan Reference :** 
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/icdbehavior.adoc#tc-icdb-2-5-icd-state-machine-with-1-client-registration-with-subscription-and-1-unregistered-client-with-subscription-multiple-fabrics-dut_server

yunhanw-google, 2024-08-30:
@Rajashreekalmane  please rebase your code to latest master, and retest it, thanks

Rajashreekalmane, 2024-09-02:
@yunhanw-google Still facing the same issue in latest TOT :  708558881bf6e5b0dac3c09489124907927b6cd2 (HEAD -> master, origin/master, origin/HEAD)

TC-ICDB-2.4 - At Step 3b, the DUT is expected not to send Check-In Messages to TH1 and TH2 after MaxInterval cycles. Currently, the DUT is sending Check-In Messages to both TH1 and TH2 after these MaxInterval cycles

PFA log below : 
[acl.zip](https://github.com/user-attachments/files/16837431/acl.zip)
[TC-ICDB-2.4 Dut log.txt](https://github.com/user-attachments/files/16837433/TC-ICDB-2.4.Dut.log.txt)
[TC-ICDB-2.4 TH1.txt](https://github.com/user-attachments/files/16837434/TC-ICDB-2.4.TH1.txt)
[TC-ICDB-2.4 TH2.txt](https://github.com/user-attachments/files/16837435/TC-ICDB-2.4.TH2.txt)

TC-ICDB-2.5 - As per Test plan, at Step 3b, the DUT is expected not to send Check-In Messages to TH1 and TH2 after MaxInterval cycles. Currently, the DUT is sending Check-In Messages to TH1.

**PFA log below :** 
[TC-ICDB-2.5 TH.txt](https://github.com/user-attachments/files/16837448/TC-ICDB-2.5.TH.txt)
[TC-ICDB-2.5 TH2.txt](https://github.com/user-attachments/files/16837449/TC-ICDB-2.5.TH2.txt)
[TC-ICDB-2_5 DUT log.txt](https://github.com/user-attachments/files/16837450/TC-ICDB-2_5.DUT.log.txt)

closed 2024-09-04 by jmartinez-silabs