project-chip/connectedhomeip #34825: [Feature] Adaptive Lighting

rgreco87, 2024-08-06:
### Feature description

Please, add Adaptive Lighting feature to Matter.

This feature has been added by Apple to HomeKit in 2020 with iOS/iPadOS 14 release.

### Platform

all

### Platform Version(s)

All

### Anything else?

https://support.apple.com/guide/iphone/control-accessories-iph0a717a8fd/ios

https://www.howtogeek.com/712520/how-to-use-adaptive-lighting-with-apple-homekit-lights/

https://www.theverge.com/2020/11/12/21562177/philips-hue-adaptive-lighting-homekit-ios-14-white-and-color-ambiance

bzbarsky-apple, 2024-08-07:
Not an SDK issue, and "adaptive lighting" can already be implemented on top of the functionality Matter provides, as far as I can tell.

FezVrasta, 2024-08-08:
It looks like Apple just added support for Adaptive Lighting for Matter lights on the last iOS Beta 3 update

rgreco87, 2024-08-15:
👍

wotip123, 2024-10-20:
So what does it mean if i have a new matter ceiling lights and they are not showing adaptive lighting on HomeKit ? While my other devices are showing adaptive lighting ?

FezVrasta, 2024-10-20:
I don't think Apple has released this to the general public yet? Or if so, just on selected partners.