project-chip/connectedhomeip #34835: Compiling Android v1.3.0.0 version: 'CHIP_DEVICE_CONFIG_DYNAMIC_SERVER' is not defined

kuyu132, 2024-08-07:
### Build issue(s)

In the v1.3.0.0 branch, an error occurs when compiling Android products. The problem seems to be the definition of the macro writing problem

********
../../src/controller/java/AndroidDeviceControllerWrapper.h:49:5: error: 'CHIP_DEVICE_CONFIG_DYNAMIC_SERVER' is not defined, evaluates to 0 [-Werror,-Wundef]
#if CHIP_DEVICE_CONFIG_DYNAMIC_SERVER
    ^
../../src/controller/java/AndroidDeviceControllerWrapper.h:264:5: error: 'CHIP_DEVICE_CONFIG_DYNAMIC_SERVER' is not defined, evaluates to 0 [-Werror,-Wundef]
#if CHIP_DEVICE_CONFIG_DYNAMIC_SERVER
    ^
../../src/controller/java/AndroidDeviceControllerWrapper.cpp:647:5: error: 'CHIP_DEVICE_CONFIG_DYNAMIC_SERVER' is not defined, evaluates to 0 [-Werror,-Wundef]
#if CHIP_DEVICE_CONFIG_DYNAMIC_SERVER
    ^
../../src/controller/java/AndroidDeviceControllerWrapper.cpp:677:5: error: 'CHIP_DEVICE_CONFIG_DYNAMIC_SERVER' is not defined, evaluates to 0 [-Werror,-Wundef]
#if CHIP_DEVICE_CONFIG_DYNAMIC_SERVER
    ^
4 errors generated.

### Platform

android

bzbarsky-apple, 2024-08-07:
Sounds like someone is forgetting to include the right config headers...

yunhanw-google, 2024-10-08:
this should have already be fixed in master branch via https://github.com/project-chip/connectedhomeip/pull/32704/commits/bbe758b3a2e4af14b13223cc554f5bdf415f9289

I am closing this one, feel free to reopen if the issue is still outstanding, thanks

closed 2024-10-08 by yunhanw-google