project-chip/connectedhomeip #34871: Apple swift registry help request

CasoDisco, 2024-08-08:
Hi I accidentally upload this file instead of the write file could the owner help with this? [image]https://swiftpackageregistry.com/project-chip/connectedhomeip-doc

CasoDisco, 2024-08-08:
If the owner can, could you add support for the Apple swift registry package. I think it be a good place to upload especially making it easy for Apple use that use swift.

CasoDisco, 2024-08-08:
Here is the link for the file: https://swiftpackageregistry.com/project-chip/connectedhomeip-doc

bzbarsky-apple, 2024-08-20:
@CasoDisco Is this about Matter.framework?  Or something else?

You should not need Swift Package Manager to use Matter.framework; it ships with the OS.