project-chip/connectedhomeip #34913: Follow-up comments for ARL PR 34537

tleacmcsa, 2024-08-09:
These comments need resolution post 1.4:

kSubscribeEventRequest: why is this needed given subscribe privileges are the same as read?  https://github.com/project-chip/connectedhomeip/pull/34537files/8dbf1ac6abd62079c7acf6478859fdc79d361707#r1697116544

Clarification on request type and how to identify action being performed.  I believe it is addressed, but @bzbarsky-apple should verify.  https://github.com/project-chip/connectedhomeip/pull/34537/files/38db8432439c8e2253b3189c5df72c1ca01e8cea#r1693433448

Similar to above, I believe I have addressed this comment but would like @andy31415 to confirm. https://github.com/project-chip/connectedhomeip/pull/34537/files/38db8432439c8e2253b3189c5df72c1ca01e8cea#r1693431527

bzbarsky-apple, 2024-08-20:
The request type bits are not addressed at least in the following:

1. There is some sort of asymmetry between "event" and "attribute" in the RequestType values that does not actually exist: you can read attributes and events, and you can subscribe to attributes and events.
2. The meaning of "wildcard" entityId is not defined.  There is nothing like that in the spec in terms of the ARL bits, and it's really not clear what it means for that field to be "wildcard".
3. There is no clear definition of what entityId means for different request types.

tleacmcsa, 2024-08-23:
@tcarmelveilleux The first piece of this issue came from your comment. Are we good with where it landed now?

@bzbarsky-apple and @andy31415, this should line up with the spec work (https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/10211). Is it clear in the spec?  The intent is to be able to support restrictions on events separately from attributes.

bzbarsky-apple, 2024-08-23:
The spec bits I looked at this morning looked reasonable, yes.

tcarmelveilleux, 2024-08-29:
This is resolved.

closed 2024-08-29 by tcarmelveilleux