project-chip/connectedhomeip #34937: [BUG] Buffer Overrun Potential in External Attribute Reads

witowsg-amazon, 2024-08-12:
### Reproduction steps

Reads of external attributes [here](https://github.com/project-chip/connectedhomeip/blob/master/src/app/util/attribute-storage.cpp#L644-L645) can potentially overrun the [attribute data buffer](https://github.com/project-chip/connectedhomeip/blob/354464e4444e712df559ee3cece4872fe00f84b6/src/app/util/ember-io-storage.cpp#L32), leading to corruption and SDK crashes when attempting to access other global data objects.

Steps to reproduce:

1. Use a build with [kAttributeReadBufferSize](https://github.com/project-chip/connectedhomeip/blob/10531e1cd18be7a288798a8b5e4e7a985586e311/src/app/util/ember-io-storage.cpp#L31) set to some arbitrary value. In my reproductions, I used the default of 8. Additionally, have an Endpoint 0 with a standard descriptor cluster.
2. Publish a dynamic endpoint with an external attribute with a size greater than kAttributeReadBufferSize. In my example, I used a string attribute of 32 bytes.
3. Attempt to read that external attribute from another device (the bug occurs here)
4. Attempt to read the descriptor cluster of Endpoint 0, observe a crash on the server-side SDK

### Bug prevalence

Always when the steps are followed

### GitHub hash of the SDK that was being used

5bb5c9e23d532cea40476fc0bd1d3008522792ba

### Platform

other

### Platform Version(s)

v1.3.0.0

bzbarsky-apple, 2024-08-20:
Fundamentally, if you plan to publish a dynamic endpoint with some cluster definition on it, you must have a known-to-ZAP instance of that cluster on some endpoint (e.g. see how the bridge example apps do this).

That will make sure that ATTRIBUTE_LARGEST is set correctly for the XML definition of the relevant attribute, make sure that the ACL machinery is correct (if ZAP does not know about this cluster, ACL enforcement for the dynamic endpoint will not be correct), and will make sure that command dispatch for the cluster is generated properly.

witowsg-amazon, 2024-08-27:
@bzbarsky-apple just to summarize our conversation from today, we agreed to make the following changes:

1. I will add a check at the call-site to ensure that the attribute data buffer size is sufficiently large enough to hold the attribute data, and return an appropriate error code if not
2. I will also add logic to `emberAfSetDynamicEndpoint` to validate that each `EmberAfAttributeMetadata::size` is no larger than the attribute buffer size, and return an appropriate error code if not