project-chip/connectedhomeip #34939: [BUG] Matter OTA Abnormal in Apple Home

emesamwong, 2024-08-13:
### Reproduction steps

This same issue is in Apple feedback: https://feedbackassistant.apple.com/feedback/14775639

HomePod Mini version: 17.6
iPhone8 Plus with 16.7.5

I am using Apple Home TestNet to test Matter Device OTA but found unexpected behavior after OTA successes.

Before the OTA, the Matter device version is 251689472(Dec) = 0x0F007A00(Hex), the software version string should be "1.5.122.0".

[image]
shows the New SW version and the update available for the Matter device, the new version is 268470784(Dec) = 0x10008A00(Hex) and the software version string should be "1.6.138.0".

[image]
and
[image]
 show the Matter device is OTA succeeded to the OLD software version and a pop-up message for the Matter Device upgrade again a few minutes later. However, I access the device information in Apple Home, the Firmware version is already the 268470784(Dec) = 0x10008A00(Hex)  [image]

The Firmware version is the newest (268470784) when I repair the Matter Device.
When I power down and power up the HomePod Mini, the update available area in the Matter Devie disappears.

Was any similar issue reported?
I would confirm what is the problem right now, esp-matter SDK/connectedhomeip SDK/HomePod Mini/iPhone?

### Bug prevalence

always

### GitHub hash of the SDK that was being used

83f4163

### Platform

esp32

### Platform Version(s)

release/v1.1 #548ef58

### Anything else?

Please let me know if other information is needed.

bzbarsky-apple, 2024-08-20:
This does not sound like a core SDK issue.  The Apple feedback assistant ticket linked above is the right venue for this (and I prodded some folks to take a look).

aajain-com, 2024-08-23:
@emesamwong Can you please test with iOS 18 and tvOS 18? There have been many fixes in this area since. If you still see this issue, please file another FB ticket with new set of logs.