project-chip/connectedhomeip #34954: [Test Failed] DEM_TC_2_6 Device crashes.

jadhavrohit924, 2024-08-13:
### Test issue(s)

Device crashes at TH step for Test DEM_TC_2_6
`Executing Test Step: TH sends ModifyForecastRequest with ForecastID=Forecast.ForecastID+1, SlotAdjustments[0].{SlotIndex=0, Duration=Forecast.Slots[0].MaxDurationAdjustment}, Cause=GridOptimization.`

### Anything else?

[DEM2-6_2024_08_13_13_45_15.log](https://github.com/user-attachments/files/16600571/DEM2-6_2024_08_13_13_45_15.log)
[device.txt](https://github.com/user-attachments/files/16600572/device.txt)

jamesharrow, 2024-08-13:
There is a bounds check failure on the slots (although it checks the length it should use >= not just >).

A second crash then occurs which is because there is no check on the optional .HasValue() before using the value.

In fixing both of these however the test script bombs out with a check on the error being Failure but it returns a Constraint error. So needs a proper look at this code since the test case is not including a nominal power in the adjustmentSlot structure.

This was the patch that stops it from crashing.

```
+++ b/src/app/clusters/device-energy-management-server/device-energy-management-server.cpp
@@ -676,7 +676,7 @@ void Instance::HandleModifyForecastRequest(HandlerContext & ctx, const Commands:
         const Structs::SlotAdjustmentStruct::Type & slotAdjustment = iterator.GetValue();
 
         // Check for an invalid slotIndex
-        if (slotAdjustment.slotIndex > forecast.Value().slots.size())
+        if (slotAdjustment.slotIndex >= forecast.Value().slots.size())
         {
             ChipLogError(Zcl, "DEM: Bad slot index %d", slotAdjustment.slotIndex);
             ctx.mCommandHandler.AddStatus(ctx.mRequestPath, Status::Failure);
@@ -697,6 +697,7 @@ void Instance::HandleModifyForecastRequest(HandlerContext & ctx, const Commands:
         if (HasFeature(Feature::kPowerForecastReporting))
         {
             if (!slot.minPowerAdjustment.HasValue() || !slot.maxPowerAdjustment.HasValue() ||
+                !slotAdjustment.nominalPower.HasValue() || 
                 slotAdjustment.nominalPower.Value() < slot.minPowerAdjustment.Value() ||
                 slotAdjustment.nominalPower.Value() > slot.maxPowerAdjustment.Value())
             {

```

PeterC1965, 2024-08-14:
Test 2.6 might being run on the ESP32 with both kPowerForecastReporting and kStateForecastReporting set when it should only be run with kStateForecastReporting (they are mutually exclusive flags). When I run the Linux test with both of these set, I see the same fail in test step 4. 

I'll update the test plan + scripts to check for this and fail straightaway if they see this.

PR #35067 (2024-08-29): Address various DEM testing issues
Addresses various DEM testing issues:

- Fixes #34954
- Fixes #34960 
- Fixes #34962
- Fixes #34297

- https://github.com/CHIP-Specifications/chip-test-plans/issues/4415
- https://github.com/CHIP-Specifications/chip-test-plans/issues/4482

- Adds a new TC_DEM_2.10.py script for checking Q Quality and that MatterReporting callback is called when expected (found and fixed some issues here!)

This should be cross-reviewed against the DEM Test Plan change PR: 
https://github.com/CHIP-Specifications/chip-test-plans/pull/4543
closed 2024-08-29 by mergify[bot]