project-chip/connectedhomeip #34973: [Build] Unable to build chiptool

Kartikgupta1502, 2024-08-14:
### Build issue(s)

As per the documentation, I have followed all the steps, but I'm encountering some issues while building the CHIPTool from the terminal. One of the issues is detailed below: Please help here.

```
ERROR at //third_party/jsoncpp/BUILD.gn:26:1: Source file not found.
2024-08-14 08:55:47 INFO    source_set("jsoncpp") {
```

kartikgupta.int@MacBook-Pro connectedhomeip % ./scripts/build/build_examples.py --target android-arm64-chip-tool build
2024-08-14 08:55:45 INFO    Building targets: android-arm64-chip-tool
2024-08-14 08:55:45 INFO    Preparing builder 'android-arm64-chip-tool'
2024-08-14 08:55:45 INFO    Generating /Users/kartikgupta.int/Desktop/ucrepos/connectedhomeip/out/android-arm64-chip-tool
2024-08-14 08:55:45 INFO    Setting up Android deps through Gradle
2024-08-14 08:55:46 INFO    > Task :buildSrc:compileJava NO-SOURCE
2024-08-14 08:55:46 INFO    > Task :buildSrc:compileGroovy UP-TO-DATE
2024-08-14 08:55:46 INFO    > Task :buildSrc:processResources NO-SOURCE
2024-08-14 08:55:46 INFO    > Task :buildSrc:classes UP-TO-DATE
2024-08-14 08:55:46 INFO    > Task :buildSrc:jar UP-TO-DATE
2024-08-14 08:55:46 INFO    > Task :buildSrc:assemble UP-TO-DATE
2024-08-14 08:55:46 INFO    > Task :buildSrc:compileTestJava NO-SOURCE
2024-08-14 08:55:46 INFO    > Task :buildSrc:compileTestGroovy NO-SOURCE
2024-08-14 08:55:46 INFO    > Task :buildSrc:processTestResources NO-SOURCE
2024-08-14 08:55:46 INFO    > Task :buildSrc:testClasses UP-TO-DATE
2024-08-14 08:55:46 INFO    > Task :buildSrc:test NO-SOURCE
2024-08-14 08:55:46 INFO    > Task :buildSrc:check UP-TO-DATE
2024-08-14 08:55:46 INFO    > Task :buildSrc:build UP-TO-DATE
2024-08-14 08:55:46 INFO    > Task :copyArtifacts UP-TO-DATE
2024-08-14 08:55:46 INFO    > Task :setUpAndroidDeps
2024-08-14 08:55:46 INFO    
2024-08-14 08:55:46 INFO    BUILD SUCCESSFUL in 524ms
2024-08-14 08:55:46 INFO    4 actionable tasks: 1 executed, 3 up-to-date
2024-08-14 08:55:46 INFO    Command ['python3', 'third_party/android_deps/set_up_android_deps.py'] completed
2024-08-14 08:55:46 INFO    Setting up Java deps
2024-08-14 08:55:46 INFO    Command ['third_party/java_deps/set_up_java_deps.sh'] completed
2024-08-14 08:55:46 INFO    Generating android-arm64-chip-tool
2024-08-14 08:55:47 INFO    ERROR at //third_party/jsoncpp/BUILD.gn:26:1: Source file not found.
2024-08-14 08:55:47 INFO    source_set("jsoncpp") {
2024-08-14 08:55:47 INFO    ^----------------------
2024-08-14 08:55:47 INFO    The target:
2024-08-14 08:55:47 INFO      //third_party/jsoncpp:jsoncpp
2024-08-14 08:55:47 INFO    has a source file:
2024-08-14 08:55:47 INFO      //third_party/jsoncpp/repo/include/json/allocator.h
2024-08-14 08:55:47 INFO    which was not found.
2024-08-14 08:55:47 INFO    ___________________
2024-08-14 08:55:47 INFO    ERROR at //third_party/jsoncpp/BUILD.gn:26:1: Source file not found.
2024-08-14 08:55:47 INFO    source_set("jsoncpp") {
2024-08-14 08:55:47 INFO    ^----------------------
2024-08-14 08:55:47 INFO    The target:
2024-08-14 08:55:47 INFO      //third_party/jsoncpp:jsoncpp
2024-08-14 08:55:47 INFO    has a source file:
2024-08-14 08:55:47 INFO      //third_party/jsoncpp/repo/include/json/assertions.h
2024-08-14 08:55:47 INFO    which was not found.
2024-08-14 08:55:47 INFO    ___________________
2024-08-14 08:55:47 INFO    ERROR at //third_party/jsoncpp/BUILD.gn:26:1: Source file not found.
2024-08-14 08:55:47 INFO    source_set("jsoncpp") {
2024-08-14 08:55:47 INFO    ^----------------------
2024-08-14 08:55:47 INFO    The target:
2024-08-14 08:55:47 INFO      //third_party/jsoncpp:jsoncpp
2024-08-14 08:55:47 INFO    has a source file:
2024-08-14 08:55:47 INFO      //third_party/jsoncpp/repo/include/json/config.h
2024-08-14 08:55:47 INFO    which was not found.
2024-08-14 08:55:47 INFO    ___________________
2024-08-14 08:55:47 INFO    ERROR at //third_party/jsoncpp/BUILD.gn:26:1: Source file not found.
2024-08-14 08:55:47 INFO    source_set("jsoncpp") {
2024-08-14 08:55:47 INFO    ^----------------------
2024-08-14 08:55:47 INFO    The target:
2024-08-14 08:55:47 INFO      //third_party/jsoncpp:jsoncpp
2024-08-14 08:55:47 INFO    has a source file:
2024-08-14 08:55:47 INFO      //third_party/jsoncpp/repo/include/json/json.h
[... 94 lines cut ...]
  File "/Users/kartikgupta.int/Desktop/ucrepos/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/Users/kartikgupta.int/Desktop/ucrepos/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/Users/kartikgupta.int/Desktop/ucrepos/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/click/core.py", line 1688, in invoke
    rv.append(sub_ctx.command.invoke(sub_ctx))
  File "/Users/kartikgupta.int/Desktop/ucrepos/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/kartikgupta.int/Desktop/ucrepos/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/Users/kartikgupta.int/Desktop/ucrepos/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Users/kartikgupta.int/Desktop/ucrepos/connectedhomeip/./scripts/build/build_examples.py", line 220, in cmd_build
    context.obj.Build()
  File "/Users/kartikgupta.int/Desktop/ucrepos/connectedhomeip/scripts/build/build/__init__.py", line 67, in Build
    self.Generate()
  File "/Users/kartikgupta.int/Desktop/ucrepos/connectedhomeip/scripts/build/build/__init__.py", line 62, in Generate
    builder.generate()
  File "/Users/kartikgupta.int/Desktop/ucrepos/connectedhomeip/scripts/build/builders/android.py", line 408, in generate
    self._Execute(gn_gen, title="Generating " + self.identifier)
  File "/Users/kartikgupta.int/Desktop/ucrepos/connectedhomeip/scripts/build/builders/builder.py", line 104, in _Execute
    self._runner.Run(cmdarray, title=title)
  File "/Users/kartikgupta.int/Desktop/ucrepos/connectedhomeip/scripts/build/runner/shell.py", line 73, in Run
    raise Exception('Command %r failed: %d' % (cmd, code))
Exception: Command ['gn', 'gen', '--check', '--fail-on-unused-args', '/Users/kartikgupta.int/Desktop/ucrepos/connectedhomeip/out/android-arm64-chip-tool', '--args=target_os="android" target_cpu="arm64" android_ndk_root="/Users/kartikgupta.int/Library/Android/sdk/ndk/23.2.8568313" android_sdk_root="/Users/kartikgupta.int/Library/Android/sdk" chip_build_controller_dynamic_server=true '] failed: 1

huytungst, 2024-08-14:
`git submodule update --init` may help.

Kartikgupta1502, 2024-08-16:
Still facing issues getting this error I have verified my jdk versiopn is 11

kartikgupta.int@MacBook-Pro connectedhomeip % ./scripts/build/build_examples.py --target android-arm64-chip-tool build
2024-08-16 10:51:14 INFO    Building targets: android-arm64-chip-tool
2024-08-16 10:51:14 INFO    Preparing builder 'android-arm64-chip-tool'
2024-08-16 10:51:14 INFO    Generating /Users/kartikgupta.int/Desktop/duplicateRepos/connectedhomeip/out/android-arm64-chip-tool
2024-08-16 10:51:14 INFO    Setting up Android deps through Gradle
2024-08-16 10:51:18 INFO    > Task :buildSrc:compileJava NO-SOURCE
2024-08-16 10:51:18 INFO    > Task :buildSrc:compileGroovy UP-TO-DATE
2024-08-16 10:51:18 INFO    > Task :buildSrc:processResources NO-SOURCE
2024-08-16 10:51:18 INFO    > Task :buildSrc:classes UP-TO-DATE
2024-08-16 10:51:18 INFO    > Task :buildSrc:jar UP-TO-DATE
2024-08-16 10:51:18 INFO    > Task :buildSrc:assemble UP-TO-DATE
2024-08-16 10:51:18 INFO    > Task :buildSrc:compileTestJava NO-SOURCE
2024-08-16 10:51:18 INFO    > Task :buildSrc:compileTestGroovy NO-SOURCE
2024-08-16 10:51:18 INFO    > Task :buildSrc:processTestResources NO-SOURCE
2024-08-16 10:51:18 INFO    > Task :buildSrc:testClasses UP-TO-DATE
2024-08-16 10:51:18 INFO    > Task :buildSrc:test NO-SOURCE
2024-08-16 10:51:18 INFO    > Task :buildSrc:check UP-TO-DATE
2024-08-16 10:51:18 INFO    > Task :buildSrc:build UP-TO-DATE
2024-08-16 10:51:18 INFO    > Task :copyArtifacts UP-TO-DATE
2024-08-16 10:51:18 INFO    > Task :setUpAndroidDeps
2024-08-16 10:51:18 INFO    
2024-08-16 10:51:18 INFO    BUILD SUCCESSFUL in 4s
2024-08-16 10:51:18 INFO    4 actionable tasks: 1 executed, 3 up-to-date
2024-08-16 10:51:19 INFO    Command ['python3', 'third_party/android_deps/set_up_android_deps.py'] completed
2024-08-16 10:51:19 INFO    Setting up Java deps
2024-08-16 10:51:19 INFO    Command ['third_party/java_deps/set_up_java_deps.sh'] completed
2024-08-16 10:51:19 INFO    Building /Users/kartikgupta.int/Desktop/duplicateRepos/connectedhomeip/out/android-arm64-chip-tool
2024-08-16 10:51:19 INFO    Building JNI android-arm64-chip-tool
2024-08-16 10:51:19 INFO    ninja: Entering directory `/Users/kartikgupta.int/Desktop/duplicateRepos/connectedhomeip/out/android-arm64-chip-tool'
2024-08-16 10:51:19 INFO    [0/1] Regenerating ninja files
2024-08-16 10:51:20 INFO    [1/1415] stamp obj/src/app/icd/client/handler.inputdeps.stamp
2024-08-16 10:51:20 INFO    [2/1415] stamp obj/src/app/data-model-provider/data-model-provider.inputdeps.stamp
2024-08-16 10:51:20 INFO    [3/1415] stamp obj/src/app/data-model-provider/string-builder-adapters.inputdeps.stamp
2024-08-16 10:51:20 INFO    [4/1415] stamp obj/src/app/data-model-provider/tests/tests.lib.inputdeps.stamp
2024-08-16 10:51:20 INFO    [5/1415] stamp obj/src/app/icd/client/manager.inputdeps.stamp
2024-08-16 10:51:20 INFO    [6/1415] stamp obj/src/app/data-model/data-model.inputdeps.stamp
2024-08-16 10:51:20 INFO    [7/1415] stamp obj/src/app/icd/server/configuration-data.inputdeps.stamp
2024-08-16 10:51:20 INFO    [8/1415] stamp obj/src/app/icd/server/manager.inputdeps.stamp
2024-08-16 10:51:21 INFO    [9/1415] c++ obj/src/app/data-model/data-model.FabricScopedPreEncodedValue.cpp.o
2024-08-16 10:51:21 INFO    [10/1415] c++ obj/src/app/data-model-provider/data-model-provider.MetadataTypes.cpp.o
2024-08-16 10:51:21 INFO    [11/1415] c++ obj/src/app/data-model/data-model.PreEncodedValue.cpp.o
2024-08-16 10:51:21 INFO    [12/1415] c++ obj/src/app/data-model-provider/data-model-provider.ActionReturnStatus.cpp.o
2024-08-16 10:51:21 INFO    [13/1415] c++ obj/src/app/data-model-provider/string-builder-adapters.StringBuilderAdapters.cpp.o
2024-08-16 10:51:21 INFO    [14/1415] c++ obj/src/app/data-model-provider/tests/libIMInterfaceTests.TestActionReturnStatus.cpp.o
2024-08-16 10:51:21 INFO    [15/1415] stamp obj/src/app/icd/server/monitoring-table.inputdeps.stamp
2024-08-16 10:51:22 INFO    [16/1415] c++ obj/src/app/data-model/tests/libAppDataModelTests.TestNullable.cpp.o
2024-08-16 10:51:22 INFO    [17/1415] c++ obj/src/app/icd/server/configuration-data.ICDConfigurationData.cpp.o
2024-08-16 10:51:22 INFO    [18/1415] stamp obj/src/app/icd/server/notifier.inputdeps.stamp
2024-08-16 10:51:22 INFO    [19/1415] stamp obj/src/app/icd/server/tests/tests.lib.inputdeps.stamp
2024-08-16 10:51:22 INFO    [20/1415] c++ obj/src/app/icd/server/notifier.ICDNotifier.cpp.o
2024-08-16 10:51:22 INFO    [21/1415] c++ obj/src/app/icd/server/monitoring-table.ICDMonitoringTable.cpp.o
2024-08-16 10:51:23 INFO    [22/1415] c++ obj/src/app/icd/client/manager.DefaultICDClientStorage.cpp.o
2024-08-16 10:51:23 INFO    [23/1415] stamp obj/src/app/server/server.inputdeps.stamp
2024-08-16 10:51:23 INFO    [24/1415] c++ obj/src/app/icd/server/tests/libICDServerTests.TestDefaultICDCheckInBackOffStrategy.cpp.o
2024-08-16 10:51:25 INFO    [25/1415] ACTION //build/chip/java/tests:child_library_2__javac(//build/toolchain/android:android_arm64)
2024-08-16 10:51:25 INFO    FAILED: gen/build/chip/java/tests/child_library_2.classlist 
2024-08-16 10:51:25 INFO    python3 ../../build/chip/java/javac_runner.py --classdir obj/build/chip/java/tests/child_library_2/classes --outfile gen/build/chip/java/tests/child_library_2.classlist --build-config gen/build/chip/java/tests/child_library_2.json -- -d obj/build/chip/java/tests/child_library_2/classes @gen/build/chip/java/tests/child_library_2.sources -Werror -Xlint:all
[... cut ...]
    self._Execute(
  File "/Users/kartikgupta.int/Desktop/duplicateRepos/connectedhomeip/scripts/build/builders/builder.py", line 112, in _Execute
    self._runner.Run(cmdarray, title=title)
  File "/Users/kartikgupta.int/Desktop/duplicateRepos/connectedhomeip/scripts/build/runner/shell.py", line 73, in Run
    raise Exception('Command %r failed: %d' % (cmd, code))
Exception: Command ['ninja', '-C', '/Users/kartikgupta.int/Desktop/duplicateRepos/connectedhomeip/out/android-arm64-chip-tool'] failed: 1

closed 2024-08-16 by Kartikgupta1502