project-chip/connectedhomeip #34974: [BUG] MatterSupport will not call commissionDevice method when device is removed from ThirdParty ecosystem and then multiadmin again.

MrLiuYunPing, 2024-08-14:
### Reproduction steps

1, Using Google Home  app to commission a matter device.
2, Using GoogleHome app open commission window.
3, Using MatterSupport to multiadmin the device to our fabric.
4, Unpair the device from our fabric.
5, Using GoogleHome app open commission window.
6,  Using MatterSupport to multiadmin the device to our fabric again.
6, The MatterAddDeviceExtensionRequestHandler extention will not call commissionDevice(in home: MatterAddDeviceRequest.Home?, onboardingPayload: String, commissioningID: UUID). When we checked the log of device, it shows an error 0x7e, which means fabric exist. I think, it is because the device's fabrics has the ios keychain's fabric already.

### Bug prevalence

every time

### GitHub hash of the SDK that was being used

branch: master; commit id is:a40963d7d571c18f4da18205b918e0c1b4e788f5

### Platform

darwin

### Anything else?

Because when commissioning device on ios, we must use MatterSupport framework, so i think, the is BUG should be solved.

bzbarsky-apple, 2024-08-20:
This sounds like an issue with MatterSupport, not the Matter SDK or Matter.framework.  Please report this issue to Apple via Feedback Assistant, as follows:

Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.

closed 2024-08-20 by bzbarsky-apple

aajain-com, 2024-08-23:
@MrLiuYunPing once you have filed a feedback assistant request by following the instructions above, please provide the FB ticket number here and I will follow up on it on the ticket.

JoeyYang2010, 2024-08-29:
@MrLiuYunPing Did you add Matter device to your Fabric, I got similar issue  when first add sensor to my custom fabric, can you help me a little bit?