project-chip/connectedhomeip #34977: [Test Failed] EWATERHRT1.1: Failure return on reading Feature-Map and Cluster Revision attribute.

jadhavrohit924, 2024-08-14:
### Test issue(s)

DUT returned failure when a read command was sent from the controller for feature-map and cluster-revision attributes.

### Anything else?

[chip_tool.txt](https://github.com/user-attachments/files/16611372/chip_tool.txt)
[device.txt](https://github.com/user-attachments/files/16611373/device.txt)

jamesharrow, 2024-08-14:
This was on ESP32 all-clusters-app - so maybe the cluster isn't correctly configure in ZAP to say where these two attributes come from?

bzbarsky-apple, 2024-08-20:
@jamesharrow The ZAP part is fine.  It's just that `Clusters::WaterHeaterManagement::WhmApplicationInit` happens only in `all-clusters-app/linux/main-common.cpp`, but this is _esp32_, not _linux_.

As in, the other all-clusters apps have to be fixed to call this bit too.

jamesharrow, 2024-08-20:
@PeterC1965 is reworking this to move WaterHeater from all-clusters into Energy Management App and then to have a ESP32 and Silabs version which is compatible with Linux.

PR in progress: https://github.com/project-chip/connectedhomeip/pull/34886

jamesharrow, 2024-08-23:
Note that https://github.com/CHIP-Specifications/chip-test-plans/issues/4509 has proposed removing the TC-EWATERHTR-1.1 test from the test plan because it is covered by the TC-IDM-10.2 Conformance Testing now.

To confirm - the https://github.com/project-chip/connectedhomeip/pull/34886 has been manually tested on ESP32 to check that this now works with energy management app with water heater logic enabled.

PR #34886 (2024-08-28): Add water heater to energy management app
This PR aims to add a Water Heater example into the energy-management example app so that EP1 has EVSE and EP2 has WaterHeater.

Both would then share the Electrical Sensor, DEM and Power Topology clusters.

Summary of changes made in this PR
 - Disabled OnMode / StartupMode for EVSEM, DEMM and WHM Energy Management ZAP file (to align with spec)

 - Moved several WaterHeater related files from All-clusters-app to energy-management-app (updates all-clusters-build path and re-used in energy-management-app)
 - Fixed bugs around Forecast attribute not bein [... cut]
closed 2024-08-28 by mergify[bot]