project-chip/connectedhomeip #34985: [CERT-TEST-FAILURE] ICDM-3.4 python script doesn't work in practice

fessehaeve, 2024-08-14:
### Feature Area

Area:ICDPhase2

### Test Case

TC-ICDM-3.4

### Reproduction steps

Run tests until step 2b and observe that after DUT reboots, previous secure session is not usable anymore. Therefore, test fails.

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

350b13f76c0ad630081c4e50d608f6d62477ac1a

### Platform

python

### Anything else?

There needs to be a way for DUTs or controllers to fallback to establishing a new secure session in such scenarios.

mkardous-silabs, 2024-08-14:
@fessehaeve  how do other reboot test cases succeed if the a new secure session cannot established?

fessehaeve, 2024-08-20:
> @fessehaeve how do other reboot test cases succeed if the a new secure session cannot established?

@mkardous-silabs this is an issue since chip-tool interactive mode and the python chip reuse the session. I am not sure if there is a python test that involves rebooting the DUT. May be @cecille or @ReneJosefsen could know about such a test?

PR #35092 (2024-08-20): In TC-ICDM-3.4 make sure that a new session will be used after DUT reboots
Close session to force a new CASE session to be established after DUT reboots

fixes #34985
commit 88b66443 (2024-08-20): In TC-ICDM-3.4 make sure that a new session will be used after DUT reboots (#35092)

* after DUT reboots make sure that a new session will be used
fixes #34985

* updated comment to reflect closing the session than expiring it

* re-establish session only when not in CI
closed 2024-08-20 by mergify[bot]