project-chip/connectedhomeip #35008: [TC-S-2.2] Fail due to unsupported operand type(s) for -: 'NoneType' and 'int'

ashishkumar4infineon, 2024-08-15:
### Test issue(s)

1.
2.
3.
...

### Anything else?

Error occurred during execution of test case TC-S-2.2. unsupported operand type(s) for -: 'NoneType' and 'int'

[TC-S-2-3_iswjY2Le_TH_log_2024_08_15_15_49_51.log](https://github.com/user-attachments/files/16624263/TC-S-2-3_iswjY2Le_TH_log_2024_08_15_15_49_51.log)

ashishkumar4infineon, 2024-08-15:
TC-S-2.3 also failing due to same reason.

[TC-S-2.3_iswjY2Le_TH_Log.log](https://github.com/user-attachments/files/16624302/TC-S-2.3_iswjY2Le_TH_Log.log)

bzbarsky-apple, 2024-08-20:
@lpbeliveau-silabs

lpbeliveau-silabs, 2024-08-20:
In the logs:

> INFO       | 2024-08-15 15:38:20.938357 | --------------------------------------------------------------------------------
INFO       | 2024-08-15 15:38:20.939708 | Executing Test Step: TH reads from the DUT the (0x0001) SceneTableSize attribute
WARNING    | 2024-08-15 15:38:21.193636 | Test Step Not Applicable: TH reads from the DUT the (0x0001) SceneTableSize attribute - Test step skipped: TH reads from the DUT the (0x0001) SceneTableSize attribute. S.S.A0001 == False
INFO       | 2024-08-15 15:38:21.199170 | --------------------------------------------------------------------------------
INFO       | 2024-08-15 15:38:21.200509 | Executing Test Step: Arithmetic operation to get the maxScenes - 1
ERROR      | 2024-08-15 15:38:21.204190 | Test Step Error: Error occurred during execution of test case TC-S-2.3. unsupported operand type(s) for -: 'NoneType' and 'int'
INFO       | 2024-08-15 15:38:21.206583 | Test Step Completed [ERROR]: Arithmetic operation to get the maxScenes - 1
INFO       | 2024-08-15 15:38:21.207896 | --------------------------------------------------------------------------------
INFO       | 2024-08-15 15:38:21.209705 | Test Case Completed [ERROR]: TC-S-2.3 (Semi-automated)
INFO       | 2024-08-15 15:38:21.210931 | ================================================================================

We can see this is failing because the previous step was skipped, so maxScenes is NoneType.

SceneTableSize is a mandatory attribute in the spec, so I think the device would be meant the tests here.

lpbeliveau-silabs, 2024-08-30:
Closing as the issue is coming from the device not enabling a mandatory attribute (SceneTableSize) this is not a test issue.

closed 2024-08-30 by lpbeliveau-silabs