project-chip/connectedhomeip #35010: [TC-S-2.5] Failed due to no shared trusted root.

ashishkumar4infineon, 2024-08-15:
### Test issue(s)

1.
2.
3.
...

### Anything else?

[TC-S-2-5_iswjY2Le_TH_Log_2024_08_15_16_45_25.log](https://github.com/user-attachments/files/16624741/TC-S-2-5_iswjY2Le_TH_Log_2024_08_15_16_45_25.log)

Snippet:

[1723740444.539] [3826:3828] [EM] CHIP MessageCounter:225455209 not in RetransTable on exchange 51297i
[1723740444.539] [3826:3828] [SC] Received error (protocol code 1) during pairing process: src/protocols/secure_channel/CASESession.cpp:2140: CHIP Error 0x000000C9: **No shared trusted root**
[1723740444.539] [3826:3828] [IN] SecureSession[0xffff8800bc40]: Released - Type:2 LSID:58200
[1723740444.540] [3826:3828] [-] src/protocols/secure_channel/CASESession.cpp:2140: CHIP Error 0x000000C9: No shared trusted root at ../../examples/chip-tool/commands/clusters/ModelCommand.cpp:67
[1723740444.540] [3826:3828] [EM] <<< [E:51297i S:0 M:225455210 (Ack:121767223)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[fe80::2a0:50ff:fe04:6d9c%eth0]:5540] --- Type 0000:10 (SecureChannel:StandaloneAck) (B:26)

bzbarsky-apple, 2024-08-20:
@lpbeliveau-silabs

lpbeliveau-silabs, 2024-08-20:
This is a secure channel error when trying to execute a group-add command, it seems more related to the configuration of the device than this specific test. Was the device able to clear the group clusters tests?

lpbeliveau-silabs, 2024-08-30:
Removing Scenes label as the failure is not happening on a scenes command.