project-chip/connectedhomeip #35015: [BUG] [TC-ICDM-2.1] test execution calls 2 hook methods for the same step - Test Harness

rquidute, 2024-08-15:
### Reproduction steps

Using Test Harness Run `TC-ICDM-2.1` test case
At step `TH reads from the DUT the UserActiveModeTriggerHint attribute`, TH is getting notified by 2 hook methods (step_skipped and then step_failure)

TH Screenshot: 
[image]

TH Execution log: [UI_Test_Run_2024_08_15_13_06_27.log](https://github.com/user-attachments/files/16626113/UI_Test_Run_2024_08_15_13_06_27.log)

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

d0d91272068f267cf880f9d56787ca28da885673

### Platform

python

rquidute, 2024-08-15:
Hi @cecille, just tagging you here in case you have any thoughts about this.
Thanks in advance.

tcarmelveilleux, 2024-08-21:
Wrong sequences of hooks/matter_testing_support call that are detectable as completely wrong should cause the test to end in ERROR not FAIL or PASS or hang.

mkardous-silabs, 2025-09-05:
Is this relevant? I remember modyfing the test script around those tests because wrong indents / order.