project-chip/connectedhomeip #35043: Occupancy Sensing: all timing attributes should be marked non-volatile in all examples

tcarmelveilleux, 2024-08-16:
The spec implied in "7.12.1. Persistence":

> Cluster attributes that represent configuration data SHALL be persistent data unless otherwise specified.
> 
> For example: a writable attribute that persistently changes the behavior (or mode) of the cluster.
> 
> Examples of non-configuration data: device state data, data that is calculated or comes from an external source, such as a sensor value, a time value, etc.
> 
> Many clusters define persistent data that is not surfaced as attributes, but is managed by commands. Commissioning or configuration data that is created to allow the cluster to perform its function is persistent data. A group table entry and binding entries are both persistent data across a restart.
> 

All the occupancy settings like XxxOccupiedToUnoccupiedDelay and XxxUnoccupiedToOccupiedDelay are obvious configuration data that need to be stored across reset, but this was not reflected in the samples.

Everyone working on occupancy sensors should be reminded to store these values persistently if the attributes are supported.

See also https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/10222

leorozendaal, 2024-08-16:
spec issue to clarify the spec text on persistence vs. `N` plus do a walkhthrough of all clusters: https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/10221

tcarmelveilleux, 2024-08-30:
Not a SVE blocker