project-chip/connectedhomeip #35071: [TC-ICDB-2.3] Check-In Messages uses the same node ID for both TH1 and TH2

yinyihu-silabs, 2024-08-19:
### Reproduction steps

After TH1 and TH2 sent RegisterClient command to DUT (without subscription), DUT sent periodic check-in message to TH1 and TH2. However, the node ID are the same in chip-tool log where `CheckInNodeID` are 1 for TH1 and 2 for TH2.
```
[1723734265.546] [2151:2153] [ICD] Check In Message processing complete: start_counter=3542478618 offset=1 **nodeid=<000000000001B669, 1>**
[1723734265.546] [2151:2153] [TOO] The node <000000000001B669, 1> is not the one we are interested in.
[1723734265.640] [2151:2153] [EM] >>> [E:2573r S:0 M:111352178] (U) Msg RX from 0:A23DE6DD41B951FE [0000] --- Type 0000:50 (SecureChannel:ICD_CheckInMessage) (B:57)
[1723734265.641] [2151:2153] [EM] Handling via exchange: 2573r, Delegate: 0xaaaac12884c8
[1723734265.642] [2151:2153] [ICD] Check In Message processing complete: start_counter=3542478618 offset=1 **nodeid=<000000000001B669, 1>**
[1723734265.642] [2151:2153] [TOO] The node <000000000001B669, 1> is not the one we are interested in.
[1723734331.832] [2151:2153] [EM] >>> [E:2574r S:0 M:111352179] (U) Msg RX from 0:D08E2CA3B132D171 [0000] --- Type 0000:50 (SecureChannel:ICD_CheckInMessage) (B:57)
[1723734331.833] [2151:2153] [EM] Handling via exchange: 2574r, Delegate: 0xaaaac12884c8
``` 

### Bug prevalence

always

### GitHub hash of the SDK that was being used

51dbf0ace5

### Platform

darwin, efr32

yunhanw-google, 2024-08-27:
may I know how do you run this command for register client for this test case?

yunhanw-google, 2024-08-28:
000000000001B669 means 112233, 1 is the fabric id

after calling icdmanagement register-client 112233 1 
you also need call command      icd wait-for-device 112233 30

closed 2024-08-28 by yunhanw-google