project-chip/connectedhomeip #35121: [BUG] Clang-Tidy error

jlatusek, 2024-08-21:
### Reproduction steps

New version of clang-tidy avaliable with latest pigweed version, found new types of errors:
- `bugprone-pointer-arithmetic-on-polymorphic-object` - error:

```
2024-08-19 08:28:29 INFO TIDY ../../src/app/EventManagement.cpp: /mnt/dane/projects/matter/connectedhomeip-second/out/sanitizers/../../src/app/EventManagement.cpp:107:51: error: pointer arithmetic on polymorphic object of type 'CircularEventBuffer' can result in undefined behavior if the dynamic type differs from the pointer type [bugprone-pointer-arithmetic-on-polymorphic-object,-warnings-as-errors]`  
    107 | next = (bufferIndex < aNumBuffers - 1) ? &apCircularEventBuffer[bufferIndex + 1] : nullptr;
            | ^~~~~~~~~~~~~~~~~~~~~
/mnt/dane/projects/matter/connectedhomeip-second/out/sanitizers/../../src/app/EventManagement.cpp:109:20: error: pointer arithmetic on polymorphic object of type 'CircularEventBuffer' can result in undefined behavior if the dynamic type differs from the pointer type [bugprone-pointer-arithmetic-on-polymorphic-object,-warnings-as-errors]
    109 | current = &apCircularEventBuffer[bufferIndex];
```
- `bugprone-suspicious-stringview-data-usage` - error: [clang-tidy.log](https://github.com/user-attachments/files/16688414/clang-tidy.log)

### Bug prevalence

always

### GitHub hash of the SDK that was being used

ff6863ed127fb1a8a6f134ed15b2ed98ed75e418

### Platform

core

bzbarsky-apple, 2024-08-21:
@jlatusek Does marking CircularEventBuffer as `final` silence the warning?

jlatusek, 2024-08-21:
@bzbarsky-apple you are right, it silence the warning

bzbarsky-apple, 2024-08-21:
Great, a warning that makes sense and goes away when it's not relevant.  :)  @jlatusek do you want to create that PR, or should I?

jlatusek, 2024-08-21:
I will fix that in PR in which I found it #34194

closed 2025-02-04 by andy31415