project-chip/connectedhomeip #35125: [BUG] Commission Fail with error "Inet Layer shutdown","BLE shutdown","System Layer shutdown","Discovery timed out","Mdns: Browse failure (src/platform/Darwin/DnssdImpl.cpp:412: CHIP Error 0x00000074: The operation has been cancelled)","XPC connection invalid"

JoeyYang2010, 2024-08-21:
### Reproduction steps

Hi there:
Below is my sample code, but I can not commission a sensor, the error is as below:
Inet Layer shutdown
BLE shutdown
System Layer shutdown
Can't start on new fabric that matches existing fabric
Creating NSError from /Library/Caches/com.apple.xbs/Sources/CHIPFramework/connectedhomeip/src/darwin/Framework/CHIP/MTRDeviceControllerFactory.mm:834: CHIP Error 0x00000003: Incorrect state
Discovery timed out
Mdns: Browse failure (src/platform/Darwin/DnssdImpl.cpp:412: CHIP Error 0x00000074: The operation has been cancelled)
status MTRCommissioningStatus
XPC connection invalid
===============================
Code:
//
//  ViewController.swift
//  SampleHybridApp
//
//  Created by Ömer Faruk Öztürk on 18.07.2020.
//  Copyright © 2020 Omer Faruk Ozturk. All rights reserved.
//

import UIKit
import WebKit
import MatterSupport
import Matter
class ViewController: UIViewController {
    public var sController: MTRDeviceController?
    private var webView: WKWebView!
    private let callbackName = "callback"
    let matterQueue = DispatchQueue(label: "com.zigbee.chip.qrcodevc.callback", qos: .userInteractive)
    override func viewDidLoad() {
        super.viewDidLoad()
        Task {
            await self.setup()
        }
        
    }
    func setup() async {
        test()
    }
    func test() {
        let request = MatterAddDeviceRequest(
            topology: .init(ecosystemName: "RRRRRR", homes: [
                .init(displayName: "R11111"), .init(displayName: "R222222")
            ])
        )
        let topology = MatterAddDeviceRequest.Topology(ecosystemName: "RRRRRR", homes: [MatterAddDeviceRequest.Home(displayName: "R11111")])
        Task
        {
            do
            {
                print(1)
//                try await request.perform()
                print(2)
                let grpId = "123"
                let testKeys = MTRCSRKeys(groupId: grpId)
                let params = MTRDeviceControllerStartupParams(ipk: testKeys.ipk, fabricID: 1, nocSigner: testKeys)
                params.vendorID = NSNumber(value: 123)
//                params.operationalCertificateIssuerQueue = self.matterQueue
                let setupPayload : MTRSetupPayload
                do {
                    setupPayload = try MTRSetupPayload(onboardingPayload: "3228-130-8191" )
                } catch {
                    return
                }
                let factory = MTRDeviceControllerFactory.sharedInstance()
                let storage = ESPMatterStorage.shared
                let factoryParams = MTRDeviceControllerFactoryParams(storage: storage)
[... 205 lines cut ...]
    /// - Parameters:
    ///   - controller: controller
    ///   - opaqueDeviceHandle: opaque device handle
    ///   - error: error
    func deviceAttestationFailed(for controller: MTRDeviceController, opaqueDeviceHandle: UnsafeMutableRawPointer, error: Error) {
        print("abcd")
        do {
            try controller.continueCommissioningDevice(opaqueDeviceHandle, ignoreAttestationFailure: true)
        } catch {
            print(error.localizedDescription)
        }
    }
}

### Bug prevalence

Each time

### GitHub hash of the SDK that was being used

2024/4/19 master version

### Platform

darwin

bzbarsky-apple, 2024-08-21:
>                     setupPayload = try MTRSetupPayload(onboardingPayload: "3228-130-8191" )

Where is that coming from?

What you should be doing is using the payload passed to this:

```
    override func commissionDevice(in home: MatterAddDeviceRequest.Home?, onboardingPayload: String, commissioningID: UUID) async throws
```

to do your commissioning attempt.

closed 2024-08-21 by bzbarsky-apple

JoeyYang2010, 2024-08-22:
@bzbarsky-apple   try await request.perform(), when I run this code so that I can get the onboardingPayload, the code will freeze so I input the Matter device code directly, I think that should can work?

JoeyYang2010, 2024-08-22:
Update the latest issue, 
class RequestHandler: MatterAddDeviceExtensionRequestHandler {
        override func commissionDevice(in home: MatterAddDeviceRequest.Home?, onboardingPayload: String, commissioningID: UUID) async throws

is never called, so I can not get the onboardingPayload, I checked many bugs, but no clear response about this problem.

bzbarsky-apple, 2024-08-23:
> so I input the Matter device code directly, I think that should can work?

If it's the right code.  The right code in this case is the one generated when the new commissioning window is opened, not the one printed on the device.

JoeyYang2010, 2024-08-29:
@bzbarsky-apple Did we have sample code using MatterSupport and Matter framework to commission device?

JoeyYang2010, 2024-08-29:
@bzbarsky-apple For now, darwin.xcworkspace can not be compile on Xcode, I raise another ticket here and I saw others also raise similar issue here, but no response yet. When I try to use Apple's Matter/MatterSupport framework, I got the issue I raised here.

bzbarsky-apple, 2024-08-29:
@JoeyYang2010 If strongly suggest you follow the instructions at https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md#providing-feedback-to-apple to report issues that have to do with Apple-specific bits and not the open-source SDK.