project-chip/connectedhomeip #35145: [BUG] MatterAddDeviceExtensionRequestHandler is never invoked

JoeyYang2010, 2024-08-22:
### Reproduction steps

Added Matter Extension and add class RequestHandler: MatterAddDeviceExtensionRequestHandler , but commissionDevice is not invoked, so can not get onboarding payload

### Bug prevalence

Each time

### GitHub hash of the SDK that was being used

2024/4/19 master version

### Platform

darwin

bzbarsky-apple, 2024-08-23:
This sounds like a MatterSupport.framework issue, yes?

If so, please report it to Apple via Feedback Assistant.  If you drop the id of the resulting ticket here, I can try to get it followed up on.  But you'll probably need to provide some more information, logs, etc, for this to be actionable.

closed 2024-08-23 by bzbarsky-apple

JoeyYang2010, 2024-08-23:
@bzbarsky-apple  There is a same issue here: https://forums.developer.apple.com/forums/thread/726873. 
Add MatterExtension to receive "MatterAddDeviceExtensionRequestHandler" callback, but no response after scanned QR code.It seems many people have such problem but no clear answer about this issue, it make develop on Matter so hard.

JoeyYang2010, 2024-08-23:
@bzbarsky-apple  I raise another ticket here: https://forums.developer.apple.com/forums/thread/762695

bzbarsky-apple, 2024-08-23:
Please file a feedback assistant ticket, not a forum post.  See https://github.com/project-chip/connectedhomeip/issues/23566#issuecomment-1312351343 for instructions.