project-chip/connectedhomeip #35165: Remaining M-ACL Issues

tleacmcsa, 2024-08-22:
These issues need to be addressed in a follow-up PR once #34932 merges:

- Return AccessRestricted when ARL denies an otherwise passing ACL check

- In TestAccessRestriction.cpp Cecille said:

for these, include #include <app-common/zap-generated/ids/Attributes.h> at the top and use
app::Clusters::NetworkCommissioning::Id etc. rather than hard coding.

Missing tests:
pase connection with a fabric index set (will happen after addnoc)
case connection before commissioning complete has been called.

It would be good to have ARLs that differ only by the fabric, to ensure that the correct fabric ARL is being applied.

- Boris pointed out lacking test coverage for AccessRestrictionProvider class

- Boris pointed out

"So the way this is currently implemented, the client can get the event before it gets the command response. This is fine, and unavoidable, but the spec should probably have some informative language making this clear to spec readers."

and "I don't see floors for these constants defined in the spec. Or ceilings. There should probably be such definitions." for CHIPConfig.h

tleacmcsa, 2024-08-28:
@bzbarsky-apple I chatted with @chrisdecenzo about one of your comments in the SDK PR that was moved to be tracked here around limits on ARL counts and their Restrictions.  Given that the ARLs are not writable and are set by the device vendor, we dont feel a limit makes sense.  Controllers would be the consumers and they shouldnt be size bound either.

bzbarsky-apple, 2024-08-29:
@tleacmcsa In general, we do in fact have caps on the size of things clients might need to read, so clients know how much memory to allocate for it.

Some clients are not memory-bound, some are....

tleacmcsa, 2024-08-29:
Is this something we should address in the spec and sdk, or just the sdk?

bzbarsky-apple, 2024-08-29:
@tleacmcsa Which exact comment are you talking about?

This is why followups should link to the actual comments involved...

tcarmelveilleux, 2024-09-04:
Also need to remove AccessRestrictionEntryChanged, and make sure ReviewFabricRestrictions has response command indicated, and update fields in FabricRestrictionReviewUpdate to be optional rather than nullable.

bzbarsky-apple, 2025-03-07:
@tleacmcsa If your "should address" question was about size caps on lists: that needed to be addressed in the spec.