project-chip/connectedhomeip #35173: [Test Failed] What method should I use to pair a Matter device connected to the Internet via an Ethernet cable?Android

JasonClark2, 2024-08-23:
### Test issue(s)

1.pairDeviceWithCode. Should I use this method directly? Or should I use pairDeviceWithAddress?
2.In the master branch, the "ChipDeviceController" method has added startDnssd() and stopDnssd(). These two methods are used to start and stop? Is this a test function or has it been improved?
3.
...

### Platform

android

joonhaengHeo, 2024-08-27:
1.pairDeviceWithCode. Should I use this method directly? Or should I use pairDeviceWithAddress?
  -> Yes. You can do this method.

2.In the master branch, the "ChipDeviceController" method has added startDnssd() and stopDnssd(). These two methods are used to start and stop? Is this a test function or has it been improved?
  -> This code is related ICD checkin message. This is a workaround when dnssd does not work due to an Android Framework issue.

JasonClark2, 2024-08-27:
1164577787
***@***.***

 pairDeviceWithCode(long deviceId, String setupCode, boolean discoverOnce, boolean useOnlyOnNetworkDiscovery, @Nullable byte[] csrNonce, @Nullable NetworkCredentials networkCredentials, @Nullable ICDRegistrationInfo icdRegistrationInfo)

I use pairDeviceWithCode(deviceID,setupCode,false,true,null,null,null) 

my matter device:
OnboardingPayload(version=0, vendorId=4965, productId=28673, commissioningFlow=0, discoveryCapabilities=[ON_NETWORK], discriminator=3899, hasShortDiscriminator=false, setupPinCode=91300495, optionalQRCodeInfo={}, optionalVendorData={}, optionalExtensionData={})

I use pairDeviceWithCode("deviceid","91300495",false,true,null,null,null)

I tried to connect the device, but it always failed

Do I still need to transmit the parameter csrNonce, and what does this parameter correspond to?

byte[] csrNonce = discriminator?

 

 -&gt; Yes. You can do this method.
 
 -&gt; This code is related ICD checkin message. This is a workaround when dnssd does not work due to an Android Framework issue.