project-chip/connectedhomeip #35212: [Test Failed] I use pairDeviceWithCode to Connect，but it always failed.Android

JasonClark2, 2024-08-27:
### Test issue(s)

1. pairDeviceWithCode(long deviceId, String setupCode, boolean discoverOnce, boolean useOnlyOnNetworkDiscovery, @Nullable byte[] csrNonce, @Nullable NetworkCredentials networkCredentials, @Nullable ICDRegistrationInfo icdRegistrationInfo)

I use pairDeviceWithCode(deviceID,setupCode,false,true,null,null,null)

my matter device:
OnboardingPayload(version=0, vendorId=4965, productId=28673, commissioningFlow=0, discoveryCapabilities=[ON_NETWORK], discriminator=3899, hasShortDiscriminator=false, setupPinCode=91300495, optionalQRCodeInfo={}, optionalVendorData={}, optionalExtensionData={})

I use pairDeviceWithCode("deviceid","91300495",false,true,null,null,null)

I tried to connect the device, but it always failed

Do I still need to transmit the parameter csrNonce, and what does this parameter correspond to?

byte[] csrNonce = discriminator?
2.
3.
...

joonhaengHeo, 2024-08-29:
@JasonClark2 
Could you share android Chiptool full log?

yunhanw-google, 2024-10-08:
i cannot reproduce this issue, closing this one, feel free to reopen it with the log if the issue is still outstanding.

closed 2024-10-08 by yunhanw-google