project-chip/connectedhomeip #35318: [TC-OCC-2.1] Python Script has to be aligned with Test plan for Test step 5 and 6.

Ashwinigrl, 2024-08-30:
### Feature Area

Other

### Test Case

TC-OCC-2.1

### Reproduction steps

In **TC-OCC-2.1**, Test Step 5 has been updated in the test plan to reflect the correct range for the **HoldTimeLimits** attribute in the` HoldTimeMin` and` HoldTimeMax` fields, as per the specification. Consequently, the Python script needs to be updated accordingly.

Currently, the script verifies the HoldTimeMin field using `asserts.assert_greater_equal(hold_time_limits_dut.holdTimeMin, 0, "HoldTimeMin is not in valid range")`. This should be modified to verify that HoldTimeMin is greater than or equal to 1. Additionally, the script should verify that the HoldTimeMax field value is at least 10.

These updates should also be applied to Test Step 6.

**Python Script References:**
- https://github.com/project-chip/connectedhomeip/blob/master/src/python_testing/TC_OCC_2_1.py

[image]

Test plan PR References:
- https://github.com/CHIP-Specifications/chip-test-plans/pull/4559

Test Plan References:
- https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/occupancysensing.adoc
- 

### Bug prevalence

Whenever I do

### GitHub hash of the SDK that was being used

dd875401ae2f5c2f39b9e69f156cd6fabebe2aa8

### Platform

raspi

jaehs6sam, 2024-08-30:
PR https://github.com/project-chip/connectedhomeip/pull/35324 address this issue and the PR is merged.

Ashwinigrl, 2024-09-02:
Issue fixed in commit 708558881bf6e5b0dac3c09489124907927b6cd2. Hence closing this issue.

closed 2024-09-02 by Ashwinigrl