project-chip/connectedhomeip #35334: [Fabric Sync] Follow ups on data-model-no-unique-id example application

tehampson, 2024-08-30:
After https://github.com/project-chip/connectedhomeip/pull/35195 merged we need to address two issues:

1. When zap codegen generates `examples/lighting-app-data-mode-no-unique-id/lighting-common/lighting-app.matter` it adds the mandatory UniqueId even though it is removed from .zap file
2. Need to fix `scripts/tools/zap/update_cluster_revisions.py` to not update the ZAP file or this app, in a separate PR. [from this comment](https://github.com/project-chip/connectedhomeip/pull/35195#pullrequestreview-2270255028)

tehampson, 2024-08-30:
@jlatusek For some reason I cannot assign the issue to you

jlatusek, 2024-09-01:
@tehampson 
Unfortunately I will not be able to finish this task. I've changed project that I'm working on.

closed 2024-09-11 by tehampson