project-chip/connectedhomeip #35342: [BUG] Failed to read attribute: src/app/codegen-data-model-provider/CodegenDataModelProvider_Read.cpp:341: IM Error 0x00000501: General error: 0x01 (FAILURE)

lboue, 2024-09-01:
### Reproduction steps

Hello,

I am facing a crash running [Matter Linux Energy Management Example](https://github.com/project-chip/connectedhomeip/tree/master/examples/energy-management-app/linux).

## Build

    cd ~/connectedhomeip
    scripts/activate.sh
    cd examples/energy-management-app/linux
    gn gen out/debug
    ninja -C out/debug
    
## Run

**Emulate Water Heater**

    out/debug/chip-energy-management-app --application water-heater

## Error

```
[1725208807.823652][25093:25093] CHIP:DMG: Reading attribute: Cluster=0x0000_0099 Endpoint=0x1 AttributeId=0x0000_0000 (expanded=1)
[1725208807.823736][25093:25093] CHIP:DMG: AccessControl: checking f=1 a=c s=0x000000000001B669 t= c=0x0000_0099 e=1 p=v r=r
[1725208807.823821][25093:25093] CHIP:DMG: AccessControl: allowed
[1725208807.823899][25093:25093] CHIP:DMG: Failed to read attribute: src/app/codegen-data-model-provider/CodegenDataModelProvider_Read.cpp:341: IM Error 0x00000501: General error: 0x01 (FAILURE)
[1725208807.823971][25093:25093] CHIP:TST: Different written length: 714 (Ember) vs 693 (DataModel)
[1725208807.824051][25093:25093] CHIP:-: chipDie chipDie chipDie
Aborted (core dumped)
```

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

6c6b11f723692855446744c8bbc36ec1c5404512

### Platform

other

### Anything else?

**Log file**
[chip-energy-management-app-linux-WH.log](https://github.com/user-attachments/files/16829849/chip-energy-management-app-linux-WH.log)

bzbarsky-apple, 2024-09-03:
@lboue This should have been fixed by PR #35319.  Is this still an issue on tip?

lboue, 2024-09-05:
Thanks for the information. I'll do another test.

andy31415, 2024-10-21:
@lboue is this still happening?

closed 2024-11-10 by lboue