project-chip/connectedhomeip #35354: [BUG] CASE timeout not triggered

ldoering, 2024-09-02:
### Reproduction steps

After sending/receiving a Sigma1 (but no follow up). a CASE Session is opened but no timeout is ever triggered. This leads to an error when trying to etablish a new CASE Sesssion;
 [SC] Received error (protocol code 4) during pairing process: src/protocols/secure_channel/CASESession.cpp:2144: CHIP Error 0x000000DB: The Resource is busy and cannot process the request
It also prevents an ICD to leave active mode, draining the battery (I will open another bug for this).

I reproduced the issue with the LIT-ICD App on the Silabs EFR32 Platform, compiled with the Following Parameters: --icd --low-power enable_openthread_cli=false chip_enable_ota_requestor=true chip_persist_subscriptions=false
I also enabled the openthread log, so I can see poll atempts to check the interval.

- Flash the software, commission the device
- Trigger a CASE Session from Chiptool, but exit Chiptool directly after sending the Sigma1
- Watch the Logs of the Device. It will never timeout the case session
- Wait some time (at least the expected case timeout)
- Start Chiptool again, Trigger a CASE Session again
- Observer the "Resource is busy" Error message in Chiptool and the device log.

The Issue can also be observed in the reverse way, if the DUT triggered the CASE Session, but I did not manage to do that with an unmodified example.

### Bug prevalence

Everytime I ran the listed steps.

### GitHub hash of the SDK that was being used

0e3434a0665ab908e27bef9a84d4429ab8f9d93a

### Platform

efr32

bzbarsky-apple, 2024-09-03:
> a CASE Session is opened but no timeout is ever triggered. 

This should happen after 30+ seconds, when the device stops waiting for a response to the Sigma2 that it sent.

@ldoering when you "Wait some time (at least the expected case timeout)", what was the actual wait time and where did the "expected case timeout" come from?  Can you please attach, not paste, the logs from the device that was the CASE responder in this case?

ldoering, 2024-09-04:
I expected the session to fail after 30s (observing the retries after several seconds) and waited multiple minutes without observing a timeout. To get the actual expected timeout I added logging to SetResponseTimeout in ExchangeContext.cpp.
Result:
[00:00:06.805][info  ][IN] CASE Server received Sigma1 message . Starting handshake. EC 0x2000da00
[00:00:06.805][info  ][EM] Set Timeout: 45896846 ms

This timeout ist much longer than expected, it it based on the CHIP_DEVICE_CONFIG_ICD_SLOW_POLL_INTERVAL which is set to 3600000ms (1 hour) in the silabs LIT-ICD sample app (examples/lit-icd-app/silabs/openthread.gni), with additional backoff.

What kind of timeouts are expected in the LIT case? Is the prolonged interval a bug or by design? (I assume a bug, since it also prevents the ICD from sleeping)

bzbarsky-apple, 2024-09-04:
Hmm. `src/messaging/ReliableMessageProtocolConfig.cpp` does:

```
config.mIdleRetransTimeout += ICDConfigurationData::GetInstance().GetSlowPollingInterval();
```

in `GetLocalMRPConfig`.  And then in CASESession's `ComputeRoundTripTimeout` we have:

```
        GetRetransmissionTimeout(localMRPConfig.mActiveRetransTimeout, localMRPConfig.mIdleRetransTimeout,
                                 // Peer will assume we are active, since it's
                                 // responding to our message.
                                 System::SystemClock().GetMonotonicTimestamp(), localMRPConfig.mActiveThresholdTime);
```

What values are you seeing passed to `GetRetransmissionTimeout` there?  What does it return?

bzbarsky-apple, 2024-09-04:
@ldoering And definitely a timeout of 45896846 for a Sigma2 seems like a bug.

ldoering, 2024-09-05:
For the LIT-Mode:

```
[00:00:08.661][info  ][IN] CASE Server received Sigma1 message . Starting handshake. EC 0x2000b238
[00:00:08.661][info  ][EM] S ComputeRoundTripTimeout: processing 30000
[00:00:08.662][info  ][EM] GetRetransmissionTimeout: active 300 idle 500 tslast 2 thresh 4000
[00:00:08.662][info  ][EM] UA GetMessageReceiptTimeout: ourLastActivity 8662
[00:00:08.662][info  ][EM] GetRetransmissionTimeout: active 3000 idle 3602000 tslast 0 thresh 5000
[00:00:08.663][info  ][EM] Set Timeout: 45896846 ms
```

I also disabled the LIT Mode and received a timeout of 281907 ms for the SIT configuration. 

```
[00:03:37.693][info  ][IN] CASE Server received Sigma1 message . Starting handshake. EC 0x2000b238
[00:03:37.693][info  ][EM] S ComputeRoundTripTimeout: processing 30000
[00:03:37.693][info  ][EM] GetRetransmissionTimeout: active 300 idle 500 tslast 2 thresh 4000
[00:03:37.694][info  ][EM] UA GetMessageReceiptTimeout: ourLastActivity 217694
[00:03:37.694][info  ][EM] GetRetransmissionTimeout: active 3000 idle 17000 tslast 0 thresh 5000
[00:03:37.694][info  ][EM] Set Timeout: 281907 ms
```

I also attached the full device log for both modes:

[ICD SIT Timeout.txt](https://github.com/user-attachments/files/16893076/ICD.SIT.Timeout.txt)
[ICD LIT Timeout.txt](https://github.com/user-attachments/files/16893078/ICD.LIT.Timeout.txt)

bzbarsky-apple, 2024-09-06:
@ldoering Is `tslast 0` logging the `lastActivityTime` value inside the `GetRetransmissionTimeout` call?  If so, both 2 and 0 seem like really odd values for that; who is calling it with those values? 

I assume `UA GetMessageReceiptTimeout` means `UnathenticatedSession::GetMessageReceiptTimeout`, but that should be passing `ourLastActivity` for `lastActivityTime`, which is being logged as very much not being 0 in both of the logs above.

bzbarsky-apple, 2024-09-06:
Oh, this is interesting.  So `ourLastActivity` is a TimeStamp, but `lastActivityTime` on `GetRetransmissionTimeout` is a `Timeout`. 

```
using Timeout = Milliseconds32;
using Timestamp = Milliseconds64;
```

Does fixing the type of the `lastActivityTime` argument of `GetRetransmissionTimeout` to be `Timestamp` instead of `Timeout` change the behavior at all for you?

bzbarsky-apple, 2024-09-06:
Ah, this even came up in the review in https://github.com/project-chip/connectedhomeip/pull/33093 but it was missed that this is in fact a serious problem because the actual _timestamp_ might not fit in a 32-bit millisecond counter....  That said, I can't quite reconcile this possible issue with the values for `ourLastActivity` and `tslast` logged above, but I don't know what casting and logging format was used there....

ldoering, 2024-09-06:
Sorry to confuse you there. I converted the timestamps to 32bit timeout for printing since the lib did not support direct 64bit printing. I assumed that this would be fine, since all tests run directly after reboot, so 32bit should be sufficient to count the runtime in ms.
Tslast is timeSinceLastActivity, I thought the absolute time passed in the lastActivityTime would not be too helpful. Since this is run directly after receiving the packet it looked good to me.

To me it looks like a problem to use the LIT idle time to calculate timeouts.

bzbarsky-apple, 2024-09-06:
> To me it looks like a problem to use the LIT idle time to calculate timeouts.

Sure, but why is it being used?

I wonder whether one issue is that the threshold (5000) is so close to the active interval (4000).  What that's saying is:

1. In active mode we check for activity ever 4 seconds.
2. After 5 seconds of inactivity we go into idle mode.
3. In Idle mode we check every 3602000 seconds or something.

So when computing how MRP will work here, assuming we start in active mode.

1. First packet will get sent.
2. Second packet will get sent something like 4 seconds later.
3. If the device has not gotten the packet by now it will transition to idle mode.
4. The idle interval will be used for computing the backoff on the remaining retransmits.

What the device _actually_ does is that it never transitions to idle mode at all.  But that's not what it _claims_ to do, either to the other side or to itself....

@turon @Damian-Nordic @mkardous-silabs we need to figure out what should actually be happening here, because this is clearly broken.

Damian-Nordic, 2024-09-10:
The question is if we should treat the active threshold shorter than all the MRP backoffs as misconfiguration and we should warn a user if such is configured, or we should just use it to cancel a transmission earlier (before retransmitting a message 4 times).

In the latter case, we could make `GetRetransmissionTimeout()` take a flag indicating if the device operates as LIT and it should be modified as such:
```
for (uint8_t i = 0; i < CHIP_CONFIG_RMP_DEFAULT_MAX_RETRANS + 1; i++)
{
    System::Clock::Timeout baseInterval;
    if (timeSinceLastActivity + timeout < activityThreshold)
    {
        baseInterval = activeInterval;
    }
    else if (lit)
    {
        break;
    }
    else
    {
         baseInterval = idleInterval;
    }
    timeout += Messaging::ReliableMessageMgr::GetBackoff(baseInterval, i, /* computeMaxPossible */ true);
}
```

However, then `ReliableMessageMgr::CalculateNextRetransTime()` would also detect whenever an idle interval is about to be used for LIT and remove the message from the retrans table in such a case.

Btw, `System::Clock::Timestamp` could be replaced with `System::Clock::Timeout` as our timers don't handle 64-bit timeouts anyway.

bzbarsky-apple, 2024-09-10:
As @Damian-Nordic pointed out on Slack, `GetRetransmissionTimeout` also needs to match the logic in `ReliableMessageMgr::CalculateNextRetransTime` which always treats things as "active" for non-initial messages on the exchange, which is very much not what `GetRetransmissionTimeout()` does.

sasdf, 2024-12-24:
Is there any update on this issue? It's still affecting the recently released v1.4 SDK.

I believe there's a problem with the implementation of `GetMessageReceiptTimeout`.

According to the specification:

> When waiting for an acknowledgement, an Intermittently Connected Device will always be in Active Mode since it will have at least one active exchange.

Since an ICD remains active while an ExchangeContext exists, a simple solution would be to use the Active timeout instead of the Idle timeout when the ICD feature is enabled.

With the following change, the timeout on my setup (with a 1-hour slow poll) is reduced from 40939253ms to 84501ms.

https://github.com/project-chip/connectedhomeip/blob/22496289a1a2cd250a391f58661e1bd75db4f013/src/transport/UnauthenticatedSessionTable.h#L114-L124

patched to 
```c
        	return GetRetransmissionTimeout(localMRPConfig.mActiveRetransTimeout,
#if CHIP_CONFIG_ENABLE_ICD_SERVER
                                        	localMRPConfig.mActiveRetransTimeout,
#else
                                        	localMRPConfig.mIdleRetransTimeout,
#endif
                                        	ourLastActivity, localMRPConfig.mActiveThresholdTime);
```

There's still room for improvement. Keeping the LIT device active for 84 seconds seems like a significant power drain. Implementing an exponential backoff polling strategy might be beneficial for such long wait times. (and also for waiting ActiveThreshold)

The controller-side GetAckTimeout also needs attention, as it doesn't currently consider ExchangeContext activeness.

---

BTW, the backoff formula in the spec also seems weird to me when there’s an active / idle transition.
> baseInterval * 1.6 ^ sendCount;

If we transition to idle when sendCount is 4, the timeout would jump from `fastInterval * 4` to `slowInterval * 6.5`. This could be an unnecessarily large increase, depending on the difference between the fast and slow intervals.

bzbarsky-apple, 2025-01-03:
@sasdf This is waiting for someone to fix it.  If you think you have a fix, please create a PR and request review.

sasdf, 2025-01-09:
@bzbarsky-apple Yes, I'd like to help address this issue.

Before preparing a PR, I’d like to outline my proposed changes and get your feedback. I'd appreciate it if you could let me know if this approach sounds reasonable.

---

My plan is to add a boolean `always_active` parameter to the following timeout functions to indicate whether the device should be considered continuously active:

```c++
GetAckTimeout(bool always_active);
GetMessageReceiptTimeout(Timestamp ourLastActivity, bool always_active);

ComputeRoundTripTimeout(Timeout upperlayerProcessingTimeout, bool remote_active);
```

When `always_active` is true, the `GetAckTimeout` and `GetMessageReceiptTimeout` functions will always use the `mActiveRetransTimeout` for timeout calculations.

For `ComputeRoundTripTimeout`, the boolean parameter (`remote_active`) will indicate the active status of the remote peer, while the local device waiting for its response remains active.

Callers can set these flags to true when there's an indication that the peer will remain active, thereby reducing wait times. e.g., when `HasReceivedAtLeastOneMessage()` is true for the current exchange context.

The Python bindings will also include this boolean as a keyword arg with a default value of False for backward compatibility:

```python
def ComputeRoundTripTimeout(nodeid, upperLayerProcessingTimeoutMs: int = 0, remote_active: bool = False)
```

---

FYI, Here's the original function declarations to be changed.
https://github.com/project-chip/connectedhomeip/blob/f25f635da41a2cd0b4022664d9bd812bff5ce977/src/transport/Session.h#L254-L276

bzbarsky-apple, 2025-01-10:
@sasdf I haven't had time to think through this stuff, which is why I haven't written up the fix yet.  But fundamentally, I am not sure whether the set of functions we have is even the right set given the new setup.  Again, see the comments about `GetRetransmissionTimeout` logic being just fundamentally wrong in terms of how it treats non-initial messages...

I _suspect_, though again I have not really had time to think this through that we may want some concept of whether this is an initial message on an exchange (in which case we should examine the activity information we have) or a non-initial one (in which case remote should be assumed active).  And if so, we need to figure out where the right API surface for that demarcation is, and where below that we just need booleans for "active or not" or something....

I'll see if I can spend some time reading through this more closely tomorrow.

turon, 2025-01-14:
I'll add discussion of this to the agenda for the 1/27 ICD call.

bzbarsky-apple, 2025-01-27:
@turon I don't think item 3 in the above list is true for a SIT ICD.

> Definitely sender should drop attempt state across reboots (restarted chip tool should not report "Resource busy")

The _server_ is reporting "resource busy".  There is no state being stored across "reboots" on the "sender" here.

bzbarsky-apple, 2025-01-27:
More generally: we should fix our activity-tracking logic to match reality, then see whether there's anything left to do for ICDs.  The only thing I expect to neeed some thought here is the "what to do for an initial message to a LIT ICD?" question.

yunhanw-google, 2025-03-07:
@sasdf could you have a retest with https://github.com/project-chip/connectedhomeip/pull/37696? thanks

closed 2025-03-15 by mergify[bot]