project-chip/connectedhomeip #35364: [BUG] Chip-Cert-Bins build failing to build in the bloaty section

raju-apple, 2024-09-03:
### Reproduction steps

Tried to create a chip-cert-bins image using the following command and its failing in the bloat comparison tools section

`docker buildx build --build-arg COMMITHASH=1d7477934bbfaaaba3556d60a617c071830acf42 --platform linux/arm64 --tag connectedhomeip/chip-cert-bins:1d7477934bbfaaaba3556d60a617c071830acf42 --push .
`

Failure 
`ERROR: failed to solve: process "/bin/sh -c set -x     && git clone https://github.com/google/bloaty.git     && mkdir -p bloaty/build     && cd bloaty/build     && cmake -G Ninja ../     && ninja     && ninja install     && cd ../..     && rm -rf bloaty     && : # last line" did not complete successfully: exit code: 1
`

### Bug prevalence

Every time

### GitHub hash of the SDK that was being used

1d7477934bbfaaaba3556d60a617c071830acf42

### Platform

other

### Platform Version(s)

arm64

### Anything else?

Attached full failure logs
[Bloat_failure.rtf.zip](https://github.com/user-attachments/files/16843734/Bloat_failure.rtf.zip)

maher-nakesh, 2024-09-03:
try this
https://cdn.discordapp.com/attachments/1277836331501883449/1280291965879193731/fix.zip?ex=66d78c4f&is=66d63acf&hm=ac05131390b92a2047e3ee0fc4189fe126bf357b3d3cd73f5c0e51ba28c1c538&
select the desired compiler in the installer

raju-apple, 2024-09-03:
@andy31415 do you know if anything changed recently in the bloaty section ?

joyoungjun, 2024-09-03:
I had the same problem and I solved it with below.

1. Copy the codes below (referenced [here](https://github.com/project-chip/connectedhomeip/blob/c552dda703424acbc1485cdf9c03769f67bcd99b/integrations/docker/images/base/chip-build/Dockerfile#L126))

> RUN git clone https://github.com/abseil/abseil-cpp.git /tmp/abseil-cpp \
>     && cd /tmp/abseil-cpp \
>     && cmake -B build -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr/local \
>     && cmake --build build --target install \
>     && rm -rf /tmp/abseil-cpp
 

2. Past to the Dockerfile at 'connectedhomeip/integerations/docker/images/chip-cert-bins/Dockerfile' (referenced [here](https://github.com/project-chip/connectedhomeip/blob/c552dda703424acbc1485cdf9c03769f67bcd99b/integrations/docker/images/chip-cert-bins/Dockerfile#L143) )

tehampson, 2024-09-03:
Please look at this thread for more information, but pretty much what Amine. I am just summarizing discovered was:

This is my understanding of the issue:
* Issue with bloaty started [in this PR](https://github.com/google/bloaty/pull/347/files)
* Bloaty has it's own copy of some version of abseil
* Prior to the PR from Bloaty that caused the issue they selected the abseil files that they wanted to compile
* Whoever made that change maybe they weren't affected because they had a version of abseil in their system path
* With that CMake change they have a option for someone to use the abseil version they want, but will use the default on in their repo that will fail on that fail if compiled (which it seems to be for us but not for them)

The work around suggested by @joyoungjun is what we need right now, but I think bloaty is working on a fix, maybe https://github.com/google/bloaty/pull/384 is an alternative

raju-apple, 2024-09-03:
Let me try the workaround , maybe we should put the PR up also on the docker file in cert-bins if it works.

Alami-Amine, 2024-09-03:
FYI an issue has been opened at Bloaty: https://github.com/google/bloaty/issues/386
and as @tehampson said, their PR: https://github.com/google/bloaty/pull/384 should fix the issue for us, and then we can remove the workaround

raju-apple, 2024-09-04:
Confirmed the workaround is good , put up a [PR](https://github.com/project-chip/connectedhomeip/pull/35391)  , Please review when you get a chance @andy31415 @Alami-Amine @tehampson  . Thanks ! FYI @naveenkommareddi , we need this before cutoff tomorrow.

Alami-Amine, 2024-09-04:
by the way @raju-apple, this fix is not needed anymore.
Bloaty integrated their PR https://github.com/google/bloaty/pull/384 which indeed fixed the issue.
I have created a PR to remove the workaround from the base image.

closed 2024-10-16 by Alami-Amine