project-chip/connectedhomeip #35378: [BUG] Clusters::ScenesManagement::ScenesServer error / SceneTableImpl.h:33:79: error: expected primary-expression before ‘;’ token

lboue, 2024-09-03:
### Reproduction steps

**Reproduction steps**

Build Chef waterleakdetector for Linux with ScenesManagement enabled by defaut:
```
cd ~/connectedhomeip
scripts/activate.sh
./examples/chef/chef.py -t linux -d rootnode_waterleakdetector_0b067acfa3 -zb
```

Error

```
In file included from ../third_party/connectedhomeip/src/app/clusters/scenes-server/scenes-server.h:24,
                 from ../third_party/connectedhomeip/src/app/clusters/groups-server/groups-server.cpp:33:
../third_party/connectedhomeip/src/app/clusters/scenes-server/SceneTableImpl.h:33:79: error: expected primary-expression before ‘;’ token
   33 | static constexpr uint16_t kMaxScenesPerEndpoint = SCENES_MANAGEMENT_TABLE_SIZE;
      |                                                                               ^
../third_party/connectedhomeip/src/app/clusters/scenes-server/SceneTableImpl.h:38:37: error: non-constant condition for static assertion
   38 | static_assert(kMaxScenesPerEndpoint <= CHIP_CONFIG_MAX_SCENES_TABLE_SIZE,
../third_party/connectedhomeip/src/app/clusters/scenes-server/SceneTableImpl.h:41:37: error: non-constant condition for static assertion
   41 | static_assert(kMaxScenesPerEndpoint >= 16, "Per spec, kMaxScenesPerEndpoint must be at least 16");
      |               ~~~~~~~~~~~~~~~~~~~~~~^~~~~
../third_party/connectedhomeip/src/app/clusters/scenes-server/SceneTableImpl.h:42:77: error: conversion from ‘int’ to ‘uint16_t’ {aka ‘short unsigned int’} may change value [-Werror=conversion]
   42 | static constexpr uint16_t kMaxScenesPerFabric = (kMaxScenesPerEndpoint - 1) / 2;
```

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

5dd517c05d0b3c1b0a662c794bc483d5d232f7e1

### Platform

core

### Platform Version(s)

N/A

### Anything else?

**Build log** 
[chef_ScenesManagement_enabled_error.log](https://github.com/user-attachments/files/16853642/chef_ScenesManagement_enabled_error.log)

lpbeliveau-silabs, 2024-09-04:
There are two problems here causing this, one is that, in its .zap file, the app defines `"name": "SceneTableSize",` value as
`"defaultValue": "",` (which should not be accepted);

And SceneTableImp.h uses: `#ifdef SCENES_MANAGEMENT_TABLE_SIZE` instead of `#if defined(SCENES_MANAGEMENT_TABLE_SIZE) && SCENES_MANAGEMENT_TABLE_SIZE`.

Note that the later would still fail on an empty define but the error message would at least be more clear.

Will open a PR shortly to fix both problems.

lpbeliveau-silabs, 2024-09-04:
As mentioned [here](https://github.com/project-chip/connectedhomeip/pull/35406/files#r1744126477) in the PR comment, turns out that waterleakdetector does not need scene. 

Thus I will remove the cluster from the template as well to avoid this scenario in the future for other apps that might add Scenes by mistake.

closed 2024-09-04 by mergify[bot]