project-chip/connectedhomeip #35431: [BUG] TH was successfully paired to the DUT with wrong Discriminator

yufengwangca, 2024-09-05:
### Reproduction steps

When running test MCORE_FS_1_4, we start two Matter devices (Lighting Example and Fabric-Bridge) using the same passcode, and attempt to distinguish them using the discriminator.

According to the logs, it appears that the TH has successfully paired with the Fabric-Bridge despite the discriminator not matching.

It looks something wrong with mDNSResolver, the Lighting server is broadcasting with correct Discriminator
```
[SERVER] [1725495299.702053][53195:53195] CHIP:SPT: LinuxCommissionableDataProvider didn't get a PASE salt, generating one.
[SERVER] [1725495299.703226][53195:53195] CHIP:DL: Device Configuration:
[SERVER] [1725495299.703233][53195:53195] CHIP:DL:   Serial Number: TEST_SN
[SERVER] [1725495299.703237][53195:53195] CHIP:DL:   Vendor Id: 65521 (0xFFF1)
[SERVER] [1725495299.703242][53195:53195] CHIP:DL:   Product Id: 32769 (0x8001)
[SERVER] [1725495299.703243][53195:53195] CHIP:DL:   Product Name: TEST_PRODUCT
[SERVER] [1725495299.703247][53195:53195] CHIP:DL:   Hardware Version: 0
[SERVER] [1725495299.703249][53195:53195] CHIP:DL:   Setup Pin Code (0 for UNKNOWN/ERROR): 20202021
[SERVER] [1725495299.703251][53195:53195] CHIP:DL:   Setup Discriminator (0xFFFF for UNKNOWN/ERROR): 1596 (0x63C)
[SERVER] [1725495299.703255][53195:53195] CHIP:DL:   Manufacturing Date: (not set)
[SERVER] [1725495299.703256][53195:53195] CHIP:DL:   Device Type: 257 (0x101)
```

And TH has correctly configured the filter to mDNSResolver

        await self.default_controller.CommissionOnNetwork(
            nodeId=th_server_th_node_id,
            setupPinCode=self.th_server_passcode,
            filterType=ChipDeviceCtrl.DiscoveryFilterType.LONG_DISCRIMINATOR,
            filter=self.th_server_discriminator,
        )
        
```
[MatterTest] 09-04 17:14:59.695 INFO: test_TC_MCORE_FS_1_4: 1596
[FS-BRIDGE][35m[1mINF[0m  [1725495299.697907][53141:53141] CHIP:EM: >>> [E:1898r S:45918 M:78020313 (Ack:215262786)] (S) Msg RX from 1:000000000001B669 [7875
[FS-BRIDGE][96m[1mDBG[0m  [1725495299.697950][53141:53141] CHIP:EM: Found matching exchange: 1898r, Delegate: (nil)
[FS-BRIDGE][96m[1mDBG[0m  [1725495299.697977][53141:53141] CHIP:EM: Rxd Ack; Removing MessageCounter:215262786 from Retrans Table on exchange 1898r
[SERVER] [1725495299.698424][53195:53195] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/TC_MCORE_FS_1_4v3rw03o2/kvs-app-b9v080wp
[MatterTest] 09-04 17:14:59.698 INFO :pychip_DeviceController_OnNetworkCommission
[MatterTest] 09-04 17:14:59.698 INFO :filter.code:1596
[SERVER] [1725495299.698653][53195:53195] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[SERVER] [1725495299.698658][53195:53195] CHIP:DL: ChipLinuxStorage::Init: Attempt to re-initialize with KVS config file: /tmp/chip_kvs
[MatterTest] 09-04 17:14:59.698 INFO :DiscoverCommissionableNodes:filter.code:1596
```

### Bug prevalence

Whenever

### GitHub hash of the SDK that was being used

7fdfc88383676eeecd022e047ae1c5a8ce1b9b47

### Platform

raspi

### Platform Version(s)

1.3

yufengwangca, 2024-09-05:
[Uploading pair_to_dut_wrong_discriminator.txt…]()

yufengwangca, 2024-09-05:
[pair_to_dut_wrong_discriminator.txt](https://github.com/user-attachments/files/16897120/pair_to_dut_wrong_discriminator.txt)

bzbarsky-apple, 2024-09-05:
@yufengwangca Which dns-sd implementation is TH using here?  Minimal or platform?