project-chip/connectedhomeip #35437: [BUG] ZAP tool --> Primary Device is not correctly saved to zap file

lboue, 2024-09-05:
### Reproduction steps

I am facing a issue with ZAP tool. Primary Device is not correctly saved to zap file.

**Reproduce steps**

- cd ~/connectedhomeip
- ./scripts/tools/zap/run_zaptool.sh examples/chef/devices/rootnode_waterleakdetector_0b067acfa3.zap
- Edit EP1 and add `Matter Power Source` in the device list
- Check that Primary Device is still `Matter Water Leak Detector`
- Save save with CTRL+S
- Close the ZCL windows
- ./scripts/tools/zap/run_zaptool.sh examples/chef/devices/rootnode_waterleakdetector_0b067acfa3.zap
- Edit EP1 and notice that Primary Device is not `Matter Water Leak Detector` but `Matter Power Source`

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

c6f66811c00f6c13e09938147a7c37310ee7bdc9

### Platform

core

### Platform Version(s)

N/A

QuentinCaldeira-eaton, 2024-09-06:
I don't know if it's linked, but you need to activate Power Source Configuration in the endpoint 0 in order to use PowerSource Cluster in other endpoints. Tell me if it helps !

lboue, 2024-09-06:
> I don't know if it's linked, but you need to activate Power Source Configuration in the endpoint 0 in order to use PowerSource Cluster in other endpoints. Tell me if it helps !

@QuentinCaldeira-eaton I used the contact sensor as an example and EP 0 does not contain this cluster. 
This is not related to my issue.

[image]

bzbarsky-apple, 2024-09-06:
@lboue ZAP tool issues go in https://github.com/project-chip/zap/issues

lboue, 2024-09-06:
> @lboue ZAP tool issues go in https://github.com/project-chip/zap/issues

https://github.com/project-chip/zap/issues/1420

closed 2024-09-06 by lboue