project-chip/connectedhomeip #35477: [1.2] The thermostat under the bridge device cannot adjust the temperature.

wolen666, 2024-09-09:
### Reproduction steps

1. Add a thermostat under the bridge device of the esp32 platform, where the thermostat is initialized as shown below
[image]
3.Thermostat writing function
[image]
4.Thermostat reading function
[image]
5.There is always an error when adding the thermostat to adjust the temperature in the home, but there is no problem in the bar mode.
https://github.com/user-attachments/assets/da100354-6769-482c-aac3-c83657848f39

### Bug prevalence

1-2 times week

### GitHub hash of the SDK that was being used

13028cf6bf21f211215ed513e5e03d78b46418eb

### Platform

esp32, freeRTOS

### Platform Version(s)

1.2

### Type

Core SDK Performance Improvement

wolen666, 2024-09-09:
Even if I modify some initialized properties, I still can't change the temperature on the disc.

DECLARE_DYNAMIC_ATTRIBUTE_LIST_BEGIN(gThermostatAttrs)
DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::LocalTemperature::Id, INT16S,2,ATTRIBUTE_MASK_WRITABLE), 
DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::AbsMinHeatSetpointLimit::Id, INT16S,2,0), 
DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::AbsMaxHeatSetpointLimit::Id, INT16S,2,0), 
DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::AbsMinCoolSetpointLimit::Id, INT16S,2,0), 
DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::AbsMaxCoolSetpointLimit::Id, INT16S,2,0), 
DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::HVACSystemTypeConfiguration::Id, BITMAP8,1,ATTRIBUTE_MASK_NULLABLE), 
DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::OccupiedCoolingSetpoint::Id, INT16S,2,ATTRIBUTE_MASK_WRITABLE), 
DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::OccupiedHeatingSetpoint::Id, INT16S,2,ATTRIBUTE_MASK_WRITABLE), 
DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::MinHeatSetpointLimit::Id, INT16S,2,ATTRIBUTE_MASK_MIN_MAX), 
DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::MaxHeatSetpointLimit::Id, INT16S,2,ATTRIBUTE_MASK_MIN_MAX), 
DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::MaxCoolSetpointLimit::Id, INT16S,2,ATTRIBUTE_MASK_MIN_MAX), 
DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::MaxCoolSetpointLimit::Id, INT16S,2,ATTRIBUTE_MASK_MIN_MAX), 
DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::MinSetpointDeadBand::Id, INT8S,1,ATTRIBUTE_MASK_MIN_MAX), 
DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::ControlSequenceOfOperation::Id, ENUM8,4,ATTRIBUTE_MASK_WRITABLE), 
DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::SystemMode::Id, ENUM8,1,ATTRIBUTE_MASK_WRITABLE), 
DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::FeatureMap::Id, BITMAP32,4,ATTRIBUTE_MASK_WRITABLE), 
    DECLARE_DYNAMIC_ATTRIBUTE_LIST_END();

bzbarsky-apple, 2024-09-09:
HandleWriteThermostatAttribute never does an actual write and more importantly never returns success, yes?

closed 2024-09-09 by bzbarsky-apple

wolen666, 2024-09-10:
Yes, if I set it to 25 degrees, I can see in the device log that the write operation of the OccupiedCoolingSetpoint attribute is indeed triggered, and then a write operation of a fixed 30 degrees is triggered. This 30 degrees seems to be triggered every time.

wolen666, 2024-09-10:
I think it's caused by some initialization errors, but I couldn't find any reference material or process on how to add a thermostat under the bridge device, so there are always a lot of errors during development.

wolen666, 2024-09-10:
This is the main code for the thermostat under my bridge. I have modified the code for a long time and read the official documents. The above problem still occurs. I am very confused about where the problem occurs.

#define ZCL_THERMOSTAT_CLUSTER_VERSION  (6u)
#define DEVICE_TYPE_THERMOSTAT          0x0301       
#define DEVICE_TYPE_ROOT_NODE 0x0016
#define DEVICE_TYPE_BRIDGE    0x000e
#define DEVICE_VERSION_DEFAULT 1
const EmberAfDeviceType gBridgedthermostatTypes[]   = { { DEVICE_TYPE_THERMOSTAT, DEVICE_VERSION_DEFAULT },
                                                        { DEVICE_TYPE_BRIDGED_NODE, DEVICE_VERSION_DEFAULT } };                                        

uint8_t Thermostat_mode;  // current mode 
uint16_t value_16t;       // recv wirte data

uint16_t value_16t_heat;  // current heat value
uint16_t value_16t_cool;  // current cool value
Device gThermostat("Thermostat", "Office");

DECLARE_DYNAMIC_ATTRIBUTE_LIST_BEGIN(gThermostatAttrs)
DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::LocalTemperature::Id, INT16S,2,ATTRIBUTE_MASK_NULLABLE), 
DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::AbsMinHeatSetpointLimit::Id, INT16S,2,0), 
DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::AbsMaxHeatSetpointLimit::Id, INT16S,2,0), 
DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::AbsMinCoolSetpointLimit::Id, INT16S,2,0), 
DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::AbsMaxCoolSetpointLimit::Id, INT16S,2,0), 
DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::HVACSystemTypeConfiguration::Id, BITMAP8,1,ATTRIBUTE_MASK_WRITABLE), 
DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::OccupiedCoolingSetpoint::Id, INT16S,2,ATTRIBUTE_MASK_WRITABLE), 
DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::OccupiedHeatingSetpoint::Id, INT16S,2,ATTRIBUTE_MASK_WRITABLE),
DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::SystemMode::Id, ENUM8,1,ATTRIBUTE_MASK_WRITABLE), 
DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::MinHeatSetpointLimit::Id, INT16S,2,ATTRIBUTE_MASK_MIN_MAX), 
DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::MaxHeatSetpointLimit::Id, INT16S,2,ATTRIBUTE_MASK_MIN_MAX), 
DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::MaxCoolSetpointLimit::Id, INT16S,2,ATTRIBUTE_MASK_MIN_MAX), 
DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::MaxCoolSetpointLimit::Id, INT16S,2,ATTRIBUTE_MASK_MIN_MAX), 
DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::MinSetpointDeadBand::Id, INT8S,1,ATTRIBUTE_MASK_MIN_MAX), 
DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::ControlSequenceOfOperation::Id, ENUM8,1,ATTRIBUTE_MASK_WRITABLE), 
DECLARE_DYNAMIC_ATTRIBUTE(Thermostat::Attributes::FeatureMap::Id, BITMAP32,4,ATTRIBUTE_MASK_WRITABLE), 
    DECLARE_DYNAMIC_ATTRIBUTE_LIST_END();

// Declare Descriptor cluster attributes
DECLARE_DYNAMIC_ATTRIBUTE_LIST_BEGIN(descriptorAttrs)
DECLARE_DYNAMIC_ATTRIBUTE(Descriptor::Attributes::DeviceTypeList::Id, ARRAY, 254, 0), /* device list */
    DECLARE_DYNAMIC_ATTRIBUTE(Descriptor::Attributes::ServerList::Id, ARRAY, 254, 0), /* server list */
    DECLARE_DYNAMIC_ATTRIBUTE(Descriptor::Attributes::ClientList::Id, ARRAY, 254, 0), /* client list */
    DECLARE_DYNAMIC_ATTRIBUTE(Descriptor::Attributes::PartsList::Id, ARRAY, 254, 0),  /* parts list */
    DECLARE_DYNAMIC_ATTRIBUTE_LIST_END();

// Declare Bridged Device Basic Information cluster attributes
DECLARE_DYNAMIC_ATTRIBUTE_LIST_BEGIN(bridgedDeviceBasicAttrs)
DECLARE_DYNAMIC_ATTRIBUTE(BridgedDeviceBasicInformation::Attributes::NodeLabel::Id, CHAR_STRING, 32, 0), /* NodeLabel */
    DECLARE_DYNAMIC_ATTRIBUTE(BridgedDeviceBasicInformation::Attributes::Reachable::Id, BOOLEAN, 1, 0),              /* Reachable */
    DECLARE_DYNAMIC_ATTRIBUTE_LIST_END();

constexpr CommandId thermostatCommands[] = {
    app::Clusters::Thermostat::Commands::SetpointRaiseLower::Id,
    kInvalidCommandId,
};

DECLARE_DYNAMIC_CLUSTER_LIST_BEGIN(bridgedThermostatClusters)
DECLARE_DYNAMIC_CLUSTER(Thermostat::Id, gThermostatAttrs, thermostatCommands, nullptr),
    DECLARE_DYNAMIC_CLUSTER(Descriptor::Id, descriptorAttrs, nullptr, nullptr),
    DECLARE_DYNAMIC_CLUSTER(BridgedDeviceBasicInformation::Id, bridgedDeviceBasicAttrs, nullptr,
                            nullptr) DECLARE_DYNAMIC_CLUSTER_LIST_END;
                    
DECLARE_DYNAMIC_ENDPOINT(bridgedThermostatEndpoint, bridgedThermostatClusters);

DataVersion gThermostatVersions[ArraySize(bridgedThermostatClusters)];

/*------------------------------------------------------------------------------------*/
EmberAfStatus HandleReadThermostatAttribute(Device * dev, chip::AttributeId attributeId, uint8_t * buffer, uint16_t maxReadLength)
{
    ChipLogProgress(DeviceLayer, "HandleReadThermostatAttribute: attrId=%" PRIu32 ", maxReadLength=%u", attributeId, maxReadLength);
    uint16_t recv;
[... 148 lines cut ...]
    }
    else 
    {
        Thermostat_mode=*buffer;
    }
    
    ChipLogProgress(DeviceLayer, "THermostat write: attrId=%" PRIu32, attributeId);
    ReturnErrorCodeIf((attributeId != Thermostat::Attributes::OccupiedCoolingSetpoint::Id) || (!dev->IsReachable()), EMBER_ZCL_STATUS_FAILURE);
    ReturnErrorCodeIf((attributeId != Thermostat::Attributes::OccupiedHeatingSetpoint::Id) || (!dev->IsReachable()), EMBER_ZCL_STATUS_FAILURE);
    ReturnErrorCodeIf((attributeId != Thermostat::Attributes::SystemMode::Id) || (!dev->IsReachable()), EMBER_ZCL_STATUS_FAILURE);
[... cut ...]
    ReturnErrorCodeIf((attributeId != Thermostat::Attributes::MinHeatSetpointLimit::Id) || (!dev->IsReachable()), EMBER_ZCL_STATUS_FAILURE);
    ReturnErrorCodeIf((attributeId != Thermostat::Attributes::MaxHeatSetpointLimit::Id) || (!dev->IsReachable()), EMBER_ZCL_STATUS_FAILURE);
    ReturnErrorCodeIf((attributeId != Thermostat::Attributes::MaxCoolSetpointLimit::Id) || (!dev->IsReachable()), EMBER_ZCL_STATUS_FAILURE);
    ReturnErrorCodeIf((attributeId != Thermostat::Attributes::MinSetpointDeadBand::Id) || (!dev->IsReachable()), EMBER_ZCL_STATUS_FAILURE);

    return EMBER_ZCL_STATUS_SUCCESS;
}

wolen666, 2024-09-10:
--The log of the device when I set it to off,I am confused as to why the sysmode attribute is triggered so many times.

I (10658) ROUTE_HOOK: Received RIO
I (10658) ROUTE_HOOK: prefix FD08:1598:1875:: lifetime 1800
I (15358) ROUTE_HOOK: Received RIO
I (15358) ROUTE_HOOK: prefix FD08:1598:1875:: lifetime 1800
I (19268) chip[EM]: >>> [E:19554r S:59204 M:35852238] (S) Msg RX from 1:000000004CAC9D54 [3DED] --- Type 0001:06 (IM:WriteRequest)
I (19268) chip[DL]: External: attribute=65532, maxReadLength=4
I (19278) chip[DL]: HandleReadThermostatAttribute: attrId=65532, maxReadLength=4
recv buffer0
recv value_16t 0
I (19288) chip[DL]: THermostat write: attrId=28
I (19298) chip[EM]: <<< [E:19554r S:59204 M:255315805 (Ack:35852238)] (S) Msg TX to 1:000000004CAC9D54 [3DED] [UDP:[FE80::4E6:B993:29B9:1620%st1]:62399] --- Type 0001:07 (IM:WriteResponse)
I (19318) chip[EM]: >>> [E:19555r S:59204 M:35852239] (S) Msg RX from 1:000000004CAC9D54 [3DED] --- Type 0001:02 (IM:ReadRequest)
I (19328) chip[EM]: >>> [E:19556r S:59204 M:35852240] (S) Msg RX from 1:000000004CAC9D54 [3DED] --- Type 0001:02 (IM:ReadRequest)
I (19338) chip[EM]: >>> [E:19557r S:59204 M:35852241] (S) Msg RX from 1:000000004CAC9D54 [3DED] --- Type 0001:02 (IM:ReadRequest)
I (19348) chip[DL]: External: attribute=65532, maxReadLength=4
I (19358) chip[DL]: HandleReadThermostatAttribute: attrId=65532, maxReadLength=4
I (19368) chip[DL]: External: attribute=28, maxReadLength=1
I (19368) chip[DL]: HandleReadThermostatAttribute: attrId=28, maxReadLength=1
mode:0
I (19378) chip[EM]: <<< [E:19555r S:59204 M:255315806 (Ack:35852239)] (S) Msg TX to 1:000000004CAC9D54 [3DED] [UDP:[FE80::4E6:B993:29B9:1620%st1]:62399] --- Type 0001:05 (IM:ReportData)
I (19398) chip[DL]: External: attribute=65532, maxReadLength=4
I (19398) chip[DL]: HandleReadThermostatAttribute: attrId=65532, maxReadLength=4
I (19408) chip[DL]: External: attribute=28, maxReadLength=1
I (19418) chip[DL]: HandleReadThermostatAttribute: attrId=28, maxReadLength=1
mode:0
I (19428) chip[EM]: <<< [E:19556r S:59204 M:255315807 (Ack:35852240)] (S) Msg TX to 1:000000004CAC9D54 [3DED] [UDP:[FE80::4E6:B993:29B9:1620%st1]:62399] --- Type 0001:05 (IM:ReportData)
I (19448) chip[DL]: External: attribute=65532, maxReadLength=4
I (19448) chip[DL]: HandleReadThermostatAttribute: attrId=65532, maxReadLength=4
I (19458) chip[DL]: External: attribute=28, maxReadLength=1
I (19458) chip[DL]: HandleReadThermostatAttribute: attrId=28, maxReadLength=1
mode:0
I (19468) chip[EM]: <<< [E:19557r S:59204 M:255315808 (Ack:35852241)] (S) Msg TX to 1:000000004CAC9D54 [3DED] [UDP:[FE80::4E6:B993:29B9:1620%st1]:62399] --- Type 0001:05 (IM:ReportData)
I (19498) ROUTE_HOOK: Received RIO
I (19498) ROUTE_HOOK: prefix FD08:1598:1875:: lifetime 1800
I (19498) chip[EM]: >>> [E:19554r S:59204 M:35852242 (Ack:255315805)] (S) Msg RX from 1:000000004CAC9D54 [3DED] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (19518) chip[EM]: >>> [E:19558r S:59204 M:35852243] (S) Msg RX from 1:000000004CAC9D54 [3DED] --- Type 0001:02 (IM:ReadRequest)
I (19528) chip[EM]: >>> [E:19559r S:59204 M:35852244] (S) Msg RX from 1:000000004CAC9D54 [3DED] --- Type 0001:02 (IM:ReadRequest)
I (19538) chip[EM]: >>> [E:19560r S:59204 M:35852245] (S) Msg RX from 1:000000004CAC9D54 [3DED] --- Type 0001:02 (IM:ReadRequest)
I (19558) chip[EM]: >>> [E:19555r S:59204 M:35852246 (Ack:255315806)] (S) Msg RX from 1:000000004CAC9D54 [3DED] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (19568) chip[EM]: >>> [E:19556r S:59204 M:35852247 (Ack:255315807)] (S) Msg RX from 1:000000004CAC9D54 [3DED] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (19578) chip[DL]: External: attribute=65532, maxReadLength=4
I (19588) chip[DL]: HandleReadThermostatAttribute: attrId=65532, maxReadLength=4
I (19598) chip[DL]: External: attribute=28, maxReadLength=1
I (19598) chip[DL]: HandleReadThermostatAttribute: attrId=28, maxReadLength=1
mode:0
I (19608) chip[EM]: <<< [E:19558r S:59204 M:255315809 (Ack:35852243)] (S) Msg TX to 1:000000004CAC9D54 [3DED] [UDP:[FE80::4E6:B993:29B9:1620%st1]:62399] --- Type 0001:05 (IM:ReportData)
I (19628) chip[DL]: External: attribute=65532, maxReadLength=4
I (19638) chip[DL]: HandleReadThermostatAttribute: attrId=65532, maxReadLength=4
I (19638) chip[DL]: External: attribute=28, maxReadLength=1
I (19648) chip[DL]: HandleReadThermostatAttribute: attrId=28, maxReadLength=1
mode:0
I (19658) chip[EM]: <<< [E:19559r S:59204 M:255315810 (Ack:35852244)] (S) Msg TX to 1:000000004CAC9D54 [3DED] [UDP:[FE80::4E6:B993:29B9:1620%st1]:62399] --- Type 0001:05 (IM:ReportData)
I (19678) chip[DL]: External: attribute=65532, maxReadLength=4
I (19678) chip[DL]: HandleReadThermostatAttribute: attrId=65532, maxReadLength=4
I (19688) chip[DL]: External: attribute=28, maxReadLength=1
I (19698) chip[DL]: HandleReadThermostatAttribute: attrId=28, maxReadLength=1
mode:0
I (19708) chip[EM]: <<< [E:19560r S:59204 M:255315811 (Ack:35852245)] (S) Msg TX to 1:000000004CAC9D54 [3DED] [UDP:[FE80::4E6:B993:29B9:1620%st1]:62399] --- Type 0001:05 (IM:ReportData)
I (19728) chip[EM]: >>> [E:19557r S:59204 M:35852248 (Ack:255315808)] (S) Msg RX from 1:000000004CAC9D54 [3DED] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (19738) chip[EM]: >>> [E:19561r S:59204 M:35852249] (S) Msg RX from 1:000000004CAC9D54 [3DED] --- Type 0001:02 (IM:ReadRequest)
[... cut ...]
I (19918) chip[EM]: <<< [E:19562r S:59204 M:255315814 (Ack:35852253)] (S) Msg TX to 1:000000004CAC9D54 [3DED] [UDP:[FE80::4E6:B993:29B9:1620%st1]:62399] --- Type 0001:05 (IM:ReportData)
I (19938) chip[EM]: >>> [E:19561r S:59204 M:35852254 (Ack:255315813)] (S) Msg RX from 1:000000004CAC9D54 [3DED] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (19998) chip[EM]: >>> [E:19562r S:59204 M:35852255 (Ack:255315814)] (S) Msg RX from 1:000000004CAC9D54 [3DED] --- Type 0000:10 (SecureChannel:StandaloneAck)

wolen666, 2024-09-10:
2 .When I change the temperature to sysmode cool 27 degrees, the device’s log,device should be set 2700,but log is 2500,this value is localtemperature 

recv buffer140
recv value_16t 2700
I (111228) chip[DL]: THermostat write: attrId=17
I (111238) chip[EM]: <<< [E:19596r S:59204 M:255315848 (Ack:35852322)] (S) Msg TX to 1:000000004CAC9D54 [3DED] [UDP:[FE80::4E6:B993:29B9:1620%st1]:62399] --- Type 0001:07 (IM:WriteResponse)
I (111258) chip[EM]: >>> [E:19597r S:59204 M:35852323] (S) Msg RX from 1:000000004CAC9D54 [3DED] --- Type 0001:02 (IM:ReadRequest)
I (111268) chip[DL]: External: attribute=65532, maxReadLength=4
I (111278) chip[DL]: HandleReadThermostatAttribute: attrId=65532, maxReadLength=4
I (111278) chip[DL]: External: attribute=28, maxReadLength=1
I (111288) chip[DL]: HandleReadThermostatAttribute: attrId=28, maxReadLength=1
mode:140
I (111298) chip[EM]: <<< [E:19597r S:59204 M:255315849 (Ack:35852323)] (S) Msg TX to 1:000000004CAC9D54 [3DED] [UDP:[FE80::4E6:B993:29B9:1620%st1]:62399] --- Type 0001:05 (IM:ReportData)
I (111318) chip[EM]: >>> [E:19596r S:59204 M:35852324 (Ack:255315848)] (S) Msg RX from 1:000000004CAC9D54 [3DED] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (111328) chip[EM]: >>> [E:19598r S:59204 M:35852325] (S) Msg RX from 1:000000004CAC9D54 [3DED] --- Type 0001:02 (IM:ReadRequest)
I (111348) chip[DL]: External: attribute=65532, maxReadLength=4
I (111348) chip[DL]: HandleReadThermostatAttribute: attrId=65532, maxReadLength=4
I (111358) chip[DL]: External: attribute=28, maxReadLength=1
I (111358) chip[DL]: HandleReadThermostatAttribute: attrId=28, maxReadLength=1
mode:140
I (111378) chip[EM]: <<< [E:19598r S:59204 M:255315850 (Ack:35852325)] (S) Msg TX to 1:000000004CAC9D54 [3DED] [UDP:[FE80::4E6:B993:29B9:1620%st1]:62399] --- Type 0001:05 (IM:ReportData)
I (111458) chip[EM]: >>> [E:19597r S:59204 M:35852326 (Ack:255315849)] (S) Msg RX from 1:000000004CAC9D54 [3DED] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (111458) chip[EM]: >>> [E:19599r S:59204 M:35852327] (S) Msg RX from 1:000000004CAC9D54 [3DED] --- Type 0001:02 (IM:ReadRequest)
I (111478) chip[DL]: External: attribute=65532, maxReadLength=4
I (111478) chip[DL]: HandleReadThermostatAttribute: attrId=65532, maxReadLength=4
I (111488) chip[DL]: External: attribute=0, maxReadLength=2
I (111498) chip[DL]: HandleReadThermostatAttribute: attrId=0, maxReadLength=2
I (111508) chip[EM]: <<< [E:19599r S:59204 M:255315851 (Ack:35852327)] (S) Msg TX to 1:000000004CAC9D54 [3DED] [UDP:[FE80::4E6:B993:29B9:1620%st1]:62399] --- Type 0001:05 (IM:ReportData)
I (111528) chip[EM]: >>> [E:19598r S:59204 M:35852328 (Ack:255315850)] (S) Msg RX from 1:000000004CAC9D54 [3DED] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (111538) chip[EM]: >>> [E:19600r S:59204 M:35852329] (S) Msg RX from 1:000000004CAC9D54 [3DED] --- Type 0001:02 (IM:ReadRequest)
I (111548) chip[DL]: External: attribute=65532, maxReadLength=4
I (111548) chip[DL]: HandleReadThermostatAttribute: attrId=65532, maxReadLength=4
I (111558) chip[DL]: External: attribute=0, maxReadLength=2
I (111568) chip[DL]: HandleReadThermostatAttribute: attrId=0, maxReadLength=2
I (111578) chip[EM]: <<< [E:19600r S:59204 M:255315852 (Ack:35852329)] (S) Msg TX to 1:000000004CAC9D54 [3DED] [UDP:[FE80::4E6:B993:29B9:1620%st1]:62399] --- Type 0001:05 (IM:ReportData)
I (111598) chip[EM]: >>> [E:19599r S:59204 M:35852330 (Ack:255315851)] (S) Msg RX from 1:000000004CAC9D54 [3DED] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (111648) chip[EM]: >>> [E:19600r S:59204 M:35852331 (Ack:255315852)] (S) Msg RX from 1:000000004CAC9D54 [3DED] --- Type 0000:10 (SecureChannel:StandaloneAck)

wolen666, 2024-09-10:
yes , I am confused about the write operation. I have seen the callback in the device log, but it cannot control the specific temperature.