project-chip/connectedhomeip #35505: [BUG] Commissioning window timeout can be set to the value incompatible with the spec

kkasperczyk-no, 2024-09-10:
### Reproduction steps

The `CommissioningWindorManager::MaxCommissioningTimeout` is used to determine the maximum allowed commissioning timeout used in the  `CommissioningWindowManager::OpenCommissioningWindow` method. This method was modified btw of adding BLE extended announcement feature: https://github.com/project-chip/connectedhomeip/commit/ca17912497a7eb8db22d42ddbe55f36370214d8c#diff-d6f2798a18ed8d25dbb88df7816ca0584e4d6869dc2e4618dc74ee494db7f314R64 and if `CHIP_DEVICE_CONFIG_BLE_EXT_ADVERTISING` is enabled, the returned value is equal to 48 hours.

This is correct for the commissioning window opened for the Bluetooth LE, however it's not compatible with the spec for the commissioning window opened for IP. For the IP transport, the spec does not allow the window to exceed 900 s.

It makes the CADMIN 1.21 i 1.22 certification tests fail.

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

83d345e18cbff6943e4925a5936a3edc72ab076b (master)

### Platform

core

### Platform Version(s)

v1.3.0

bzbarsky-apple, 2024-09-10:
> For the IP transport, the spec does not allow the window to exceed 900 s.

Yes, it does, for _uncommissioned_ devices already on the IP network.  The "Extended Announcement" feature in the spec is transport-independent.

kkasperczyk-no, 2024-09-11:
@bzbarsky-apple thank you for the clarification. We must have had some misunderstanding of this feature in the verification area. Maybe because the spec in the past mentioned about "unnecessary pollution of the crowded 2.4 GHz wireless spectrum, especially with BLE" and in our heads it turned out into BLE related :)

kkasperczyk-no, 2024-09-11:
Or it's the `CHIP_DEVICE_CONFIG_BLE_EXT_ADVERTISING` config name in the SDK that suggests it's related to BLE only.

PR #35507 (2024-09-16): Fix max timeout for open commissioning window
Commissioning window can be opened using timeout exceeding
the maximum value of 900 s defined by the spec. This can happen
if selected transport is IP, but the device uses BLE extended
announcement feature.

Added checking if device is commissioned to be able to determine
what max timeout should be used for the particular scenario.

Renamed CHIP_DEVICE_CONFIG_BLE_EXT_ADVERTISING to the
CHIP_DEVICE_CONFIG_EXT_ADVERTISING, as config name sounds
misleading and it seems it relates only to BLE.

Fixes: #35505
commit ba9faf2b (2024-09-16): Fix max timeout for open commissioning window (#35507)

Commissioning window can be opened using timeout exceeding
the maximum value of 900 s defined by the spec. This can happen
if selected transport is IP, but the device uses BLE extended
announcement feature.

Added checking if device is commissioned to be able to determine
what max timeout should be used for the particular scenario.

Renamed CHIP_DEVICE_CONFIG_BLE_EXT_ADVERTISING to the
CHIP_DEVICE_CONFIG_EXT_ADVERTISING, as config name sounds
misleading and it seems it relates only to BLE.

Fixes: #35505
closed 2024-09-16 by mergify[bot]