project-chip/connectedhomeip #35529: [chip-tool] Tracking paired device's data model state

w8floosh, 2024-09-10:
### Reproduction steps / Feature

Hello, 
I would like to know if it is possible to leverage the `chip::app::DataModel::Provider` interface to create a provider that tracks the data model of the device the chip-tool is paired to. I need to track the data model metadata tree of the remote device (logical nodes, endpoints, clusters and attributes) to analyze it in my application. 

I see the `CodegenDataModelProvider` is used in the `InteractionModelEngine` and, as of my knowledge, it uses the generated ZAP data to get information about the available clusters, so I assume it provides the data model of the chip-tool client, which I don't need. 
Can this provider be adapted to behave as I want? Should an implementation of a remote provider follow the `Provider` interface or use a different approach, for example just sending Descriptor commands and storing the results from the callbacks?

May someone introduce me to how these providers actually work?

Thank you in advance.

### Platform

core (please add to version below)

### Type

Manually tested with SDK

bzbarsky-apple, 2024-09-11:
@andy31415

bzbarsky-apple, 2024-09-11:
The general intent of the provider interface is to be used server-side as a backend for the interaction model....  You could try using it for something on the client side, but it's not really designed around that.

andy31415, 2025-02-04:
As an inteface, it is designed to "provide all data that is available" for interaction-model purposes specifically. You can use it as an interface anywhere, however some of the wrappers are not as easily accessible for "simple data types" (e.g. reads and writes use a AttributeValueEncoder/Decoder which are reasonably complex to create and use).