project-chip/connectedhomeip #35565: [Feature] Commission Mode off and device reconnect

mideayanghui, 2024-09-13:
### Feature description

...
【Background】
If the device is currently commissioned into one fabric, the device doesn't support BLE for discovery according the Matter main Spec 5.1.3.  So If user want to setup device using BLE again, she need to  place a `User-Intent Commissioning Flow` device into advertising mode, for example, the user long press device's button to factory reset the device, and then scan the QR code to commission the device. The device can be discover during commissioning window time, such as 15mins.
[image]
【Feature】
In the situation talk above, if the device has commissioned into one fabric, and then user open the commission mode of device according  `User-Intent Commissioning Flow` .  Can the device be restored to factory default and just open the commissioning window during 15mins? If timeout, the device close the commission mode and then reconnect to the previous fabric? This is separate from factory reset.

Just like Alexa echo dot, when user press the button, the device enter paring mode,  3mins timeout, the device close the setup mode and reconnect to Wi-Fi.

### Platform

all

bzbarsky-apple, 2024-09-13:
@mideayanghui If the device is already commissioned onto a fabric, it's already on the IP network.  Why would you need to do anything in particular over BLE?

tcarmelveilleux, 2024-09-13:
Commissioning again over BLE not being possible if already on a fabric was a security choice to avoid proximity protocol attacks related to access to on-device QR code. Removal of last fabric or factory reset are the only times it is allowed to start advertising over BLE again. This makes it harder to snap the device setup code and take over the device after initial setup

mideayanghui, 2024-09-14:
@tcarmelveilleux Yes, Leo Rozendaal  also mention as below,
If a device is on a Matter fabric,  the spec (intentionally) does not allow it be commissionable via BLE. We need to use the multi-admin flow to join another (additional) fabric.

Only for BCM (which is optional).
For ECM (mandatory), the existing admin displays a NEW numeric/QR code to be used by the new commissioner - the QR code on the product cannot be used for ECM.

mideayanghui, 2024-09-14:
@bzbarsky-apple  Let me talk about the source of feature demand. We have a Matter product support both Matter protocol(Using BLE for device discovery) and Manufacturer custom protocol with custom app(using Wi-Fi Soft-AP Config). 

Because we maintain product functional compatibility, it's not friendly to put Soft-Ap config and BLE config together in the same product.
Previous user story is as below,
1.If the user use Matter protocol to commission at first, then the device is already on the IP network. 
2. And then if the user want to use custom app , they should long-press the deivce's button to factory reset the device(Matter fabric will be clear) and device start AP mode, and then use the custom app to commission the device. 
3. In this time, If we want to use Matter App, they should commission the device again. (The information of custom protocol will be restored)
4. So the user can both use custom app and Matter app at the same time right now.

If we can improve the step 2 and step 3 to reduce Matter commission , it will be great. Because Step 1 we have already finished Matter Commission.