project-chip/connectedhomeip #35588: [BUG] HeatPump device type refers to Ballast Configuration (0x0301) cluster

lboue, 2024-09-15:
### Reproduction steps

HeatPump device type refers to `Ballast Configuration (0x0301)` cluster:

https://github.com/project-chip/connectedhomeip/blob/a068855d27b42b2c17d38b3a8a9d30d672163aee/data_model/master/device_types/HeatPump.xml#L68C18-L68C24

### Bug prevalence

always

### GitHub hash of the SDK that was being used

a068855

### Platform

core

### Platform Version(s)

N/A

hasty, 2024-09-22:
PR to fix on spec side: https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/10457

lboue, 2024-10-11:
@hasty Is it fixed now?

hasty, 2024-12-15:
Appears to be.

lboue, 2024-12-15:
So someone with access to CHIP-Specifications repo could trigger an update?

lboue, 2025-05-01:
Fixed

closed 2025-05-01 by lboue