project-chip/connectedhomeip #35591: [Build] [chip-tool] Building chip-tool always fails on Raspberry pi

renu285, 2024-09-15:
### Build issue(s)

After following the description at 
https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/chip_tool_guide.md

Building of chip-tool always fails at the following step:

[402/2515] ACTION //examples/energy-management-app/energy-management-common:energy-management-common_zapgen_generate(//build/toolchain/linux:linux_arm64_gcc)
FAILED: gen/examples/energy-management-app/energy-management-common/zapgen/zap-generated/access.h gen/examples/energy-management-app/energy-management-common/zapgen/zap-generated/gen_config.h gen/examples/energy-management-app/energy-management-common/zapgen/zap-generated/endpoint_config.h gen/examples/energy-management-app/energy-management-common/zapgen/zap-generated/IMClusterCommandHandler.cpp
python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../../examples/energy-management-app/energy-management-common --default-toolchain=//build/toolchain/linux:linux_arm64_gcc --current-toolchain=//build/toolchain/linux:linux_arm64_gcc --capture-output --python-dep-list-files gen/examples/energy-management-app/energy-management-common/energy-management-common_zapgen_generate_metadata_path_list.txt -- ../../scripts/tools/zap/generate.py --no-prettify-output --templates /home/ubuntu/chip_git/connectedhomeip/src/app/zap-templates/app-templates.json --output-dir /home/ubuntu/chip_git/connectedhomeip/out/host/gen/examples/energy-management-app/energy-management-common/zapgen/zap-generated --lock-file /home/ubuntu/chip_git/connectedhomeip/out/host/zap_gen.lock --parallel /home/ubuntu/chip_git/connectedhomeip/examples/energy-management-app/energy-management-common/energy-management-app.zap
🚀 Executing: git log -1 --format={"hash": "%H","timestamp": %ct}
🔍 Git commit: 309238bb4dea36dfa87baed4eefc34e9dd4d189a from Fri Sep 13 2024 15:31:16 GMT+0000 (Coordinated Universal Time)
🔍 Current src hash: UXLQRBDOwI9GXtZcvtJlFZe8nP0=
🔍 Current src-shared hash: Fwbu7CwefZXOH5LqawYL+8aAyqY=
👎 Error reading old hash file: /home/ubuntu/zap_src/zap/spa/hash.json
🐝 Front-end code changed, so we need to rebuild SPA.
🚀 Executing: npx quasar build

 .d88888b.
d88P" "Y88b
888     888
888     888 888  888  8888b.  .d8888b   8888b.  888d888
888     888 888  888     "88b 88K          "88b 888P"
888 Y8b 888 888  888 .d888888 "Y8888b. .d888888 888
Y88b.Y8b88P Y88b 888 888  888      X88 888  888 888
 "Y888888"   "Y88888 "Y888888  88888P' "Y888888 888
       Y8b

 Build mode................ spa
 Pkg quasar................ v2.16.8
 Pkg @quasar/app-webpack... v3.13.5
 Pkg webpack............... v5
 Debugging................. no
 Publishing................ no

 Configured browser support (>= 64.69% of global marketshare):
 · Chrome for Android >= 127
 · Firefox for Android >= 127
 · QQ for Android >= 14.9
 · UC for Android >= 15.5
 · Android >= 127
 · Chrome >= 127
 · Edge >= 127
 · Firefox >= 128
 · IE >= 11
 · iOS >= 17.5
 · Kaios >= 3.0-3.1
 · Opera Mini >= all
 · Opera Mobile >= 80
 · Opera >= 111
 · Safari >= 17.5
 · Samsung >= 25

 App • Running "@quasar/testing" Quasar App Extension...
 App • Running "@quasar/testing-unit-jest" Quasar App Extension...
 App • Running "@quasar/testing-e2e-cypress" Quasar App Extension...
 App • Extending "UI" Webpack config
 App • Cleaned build artifact: "/home/ubuntu/zap_src/zap/spa"
 App •  WAIT  • Compiling of "UI" in progress...
 App •  DONE  • "UI" compiled with errors • 185275ms
⇝
⇝  App •  ERROR  •  UI  in ./src/App.vue?vue&type=script&lang=js
⇝
⇝ Module not found: Can't resolve imported dependency "querystring"
⇝ Did you forget to install it? You can run: npm install --save querystring
⇝
⇝  App •  ERROR  •  UI  in ./src/boot/ws.js
⇝
⇝ Module not found: Can't resolve imported dependency "events"
⇝ Did you forget to install it? You can run: npm install --save events
⇝

 App •  COMPILATION FAILED  • Please check the log above for details.

👎 Program npx exited with error code: 1

bzbarsky-apple, 2024-09-16:
@renu285 This seems like an issue with the ZAP setup you have, so you might want to file it in https://github.com/project-chip/zap/actions/

That said:

1. Have you tried the `npm install --save events` it suggests?
2. What version of node are you using?

renu285, 2024-09-16:
Yes I have tried doing both npm install --save events and npm install --save querystring and still complains the same. The node version i am using is v20.14.0.

bzbarsky-apple, 2024-09-16:
OK.  Then please report in this in the ZAP repo, since it's the ZAP node setup that's having problems.

andy31415, 2024-10-21:
This is a ZAP issue, please open in zap repository.

closed 2024-10-21 by andy31415