project-chip/connectedhomeip #35601: PacketBufferHandle::New API documentation doesn't match reality

tehampson, 2024-09-16:
[PacketBufferHandle::New API documentation states](https://github.com/project-chip/connectedhomeip/blob/54127b91de428642262339e04f2a895db6acd842/src/system/SystemPacketBuffer.h#L619-L639):

```
     * Allocates a packet buffer.
     *
     *  A packet buffer is conceptually divided into two parts:
     *  @li  Space reserved for network protocol headers. The size of this space normally defaults to a value determined
     *       by the network layer configuration, but can be given explicity by \c aReservedSize for special cases.
     *  @li  Space for application data. The minimum size of this space is given by \c aAvailableSize, and then \c Start()
     *       provides a pointer to the start of this space.
     *
     *  Fails and returns \c nullptr if no memory is available, or if the size requested is too large.
     *  When the sum of \a aAvailableSize and \a aReservedSize is no greater than \c PacketBuffer::kMaxSizeWithoutReserve,
     *  that is guaranteed not to be too large.
     *
     *  On success, it is guaranteed that \c AvailableDataSize() is no less than \a aAvailableSize.
     *
     *  @param[in]  aAvailableSize  Minimum number of octets to for application data (at `Start()`).
     *  @param[in]  aReservedSize   Number of octets to reserve for protocol headers (before `Start()`).
     *
     *  @return     On success, a PacketBufferHandle to the allocated buffer. On fail, \c nullptr.
     */
```

The key argument of interest for this issue is `aAvailableSize`. The reality is the application has 16 bytes less than what is allocated for the MIC. If anything uses between `aAvailableSize - 16` and `aAvailableSize` the message will fail to send as we are not able to Encode the MIC.

We can even see that `WriteClient::StartNewMessage()` likely hit this issue and worked around the issue [by reserving the 16 themselves in their IMBuilder ](https://github.com/project-chip/connectedhomeip/blob/54127b91de428642262339e04f2a895db6acd842/src/app/WriteClient.cpp#L186-L200)

This is a workaround to a problem where `PacketBufferHandle` does not reserve any space for the footer MIC

bzbarsky-apple, 2024-09-16:
Fundamentally: PacketBuffer treats the MIC as part of "application data".  This might not match what other things think of as "application data", of course....

The only reason Write needs to reserve the space is because it implements chunking, and hence has some sort of "graceful" handling of "does not fit in packet".  It also has to reserve space for trailing TLV "end container" bytes, and so on and so on.

Things that don't support chunking don't need to do any reserving.

pidarped, 2024-09-16:
The MIC is a fixed size chunk added at the end of the data during encoding(that is done by the message layer). So, ideally, that length should already be subtracted out from what is offered to the application to use. The kMaxAppMessageLen does that and, I think should be used as a limit for application data(including TLV container bytes, etc). 
The problem seems to be the definition of kMaxSecureSduLengthBytes that adds the TagLen to the kMaxAppMessageLen.
Curious as to what an application(IM) layer concept of chunking has anything to do with the data and MIC length. Each chunk should be sent out as an independent message that contains its own MIC at the end.

tehampson, 2024-09-16:
>Fundamentally: PacketBuffer treats the MIC as part of "application data". This might not match what other things think of as "application data", of course....

If that is the case I think we should add that comment into the documentation since that is not very clear at the moment.

> The only reason Write needs to reserve the space is because it implements chunking, and hence has some sort of "graceful" handling of "does not fit in packet". It also has to reserve space for [...]

I don't think I agree with this chunking rationale at all. I am messaging you on slack to discuss a little more in real time.

Essentially my argument is that anything that allocate a packet, whether it is for write/read/invoke (request or response) will need to account for this. If they do not and something allocate aAvailableSize and uses the builder they have to build a request/response (without reserving space for the MIC like WriteClient does and read response seemingly does) and they build between `aAvailableSize - 16` and `aAvailableSize`, the Builder will seems like everything is okay. But when you got to send the message, because there is nothing left at the end of the packet buffer it fails to send it out because it cannot Encode a MIC

bzbarsky-apple, 2024-09-17:
> Essentially my argument is that anything that allocate a packet, whether it is for write/read/invoke (request or response) will need to account for this.

As discussed on Slack: anything that allocates a packet _and reserves space at the end of it_, with the goal of being able to tell API consumers that things don't fit, needs to account for this.

CommandSender did not use to need to, when it could only do one command, for example, but now it might make sense to.