project-chip/connectedhomeip #35620: [Tooling] Apple Matter Developer Profile is no longer valid

andrew-elsayeh, 2024-09-17:
### Reproduction steps / Feature

The Matter Developer Profile which is required for developing matter devices using a Mac/iOS/iPadOS device is no longer valid. 

Link to guide for matter development on Apple devices: https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md#profile-installation

 Link to profile: https://developer.apple.com/services-account/download?path=/iOS/iOS_Logs/EnableBluetoothCentralMatterClientDeveloperMode.mobileconfig
 
 

### Platform

other

### Type

Manually tested with SDK

bzbarsky-apple, 2024-09-17:
@andrew-elsayeh Thank you for the report.  This is being worked on.

bzbarsky-apple, 2024-09-17:
This should be fixed now.

closed 2024-09-17 by bzbarsky-apple