project-chip/connectedhomeip #35626: Profile at https://developer.apple.com/services-account/download?path=/iOS/iOS_Logs/EnableBluetoothCentralMatterClientDeveloperMode.mobileconfig has been updated.

mokap13, 2024-09-17:
Profile at https://developer.apple.com/services-account/download?path=/iOS/iOS_Logs/EnableBluetoothCentralMatterClientDeveloperMode.mobileconfig has been updated.

_Originally posted by @bzbarsky-apple in https://github.com/project-chip/connectedhomeip/issues/29178#issuecomment-1718389067_
  
I think it happened again, can check it?

[image]

ReneJosefsen, 2024-09-18:
From what I understand the profile linked in the docs has been updated? I did also get this error yesterday, but I have been able to install it today.

ReneJosefsen, 2024-09-18:
@bzbarsky-apple I assume this is fixed, so I guess we can close it?

mokap13, 2024-09-18:
Thanks! Yes, you can close.

closed 2024-09-19 by bzbarsky-apple

mantosh83, 2025-09-14:
Hi Team,

The profile that was recently updated seems to have expired. Could you please help by updating it with the latest version, or kindly let me know where I can download the most recent one?

Expired:  Profile at https://developer.apple.com/services-account/download?path=/iOS/iOS_Logs/EnableBluetoothCentralMatterClientDeveloperMode.mobileconfig

bzbarsky-apple, 2025-10-06:
@mantosh83 The profile has been updated.  See https://github.com/project-chip/connectedhomeip/issues/40972