project-chip/connectedhomeip #35637: [CHIPTool] Failed Discovering and Pairing A Matter Compliant Light Bulb

skyFzz, 2024-09-17:
### Test issue(s)

Description: 
I am trying to use the CHIPTool controller to control an actual Matter light bulb. But I have been stuck on the commissioning phase for a couple weeks. I tried to use Home Assistant controller, which worked smoothly. So the light bulb itself was proved to be Matter supported and functional correctly.

Because the only information I have from the light bulb is a QR code and a pairing code (04090312526), and I don't have either the pin code or discriminator code, I used these two commands for paring: `./chip-tool pairing code 1 04090312526`, and `./chip-tool pairing code-wifi 1 [SSID] [pass] 04090312526`.

For `pairing code`, the discovery timed out over DNS-SD:
```
[1726604643.918] [23908:23908] [DL] ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_tool_kvs
[1726604643.918] [23908:23908] [DL] ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1726604643.918] [23908:23908] [DL] ChipLinuxStorage::Init: Attempt to re-initialize with KVS config file: /tmp/chip_kvs
[1726604643.921] [23908:23908] [DL] ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1726604643.921] [23908:23908] [DL] ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1726604643.921] [23908:23908] [DL] ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1726604643.921] [23908:23908] [DL] wrote settings to /tmp/chip_counters.ini
[1726604643.921] [23908:23908] [DL] NVS set: chip-counters/reboot-count = 17 (0x11)
[1726604643.922] [23908:23908] [DL] Got Ethernet interface: eth0
[1726604643.922] [23908:23908] [DL] Found the primary Ethernet interface:eth0
[1726604643.922] [23908:23908] [DL] Got WiFi interface: wlan0
[1726604643.923] [23908:23908] [DL] Found the primary WiFi interface:wlan0
[1726604643.923] [23908:23908] [IN] UDP::Init bind&listen port=0
[1726604643.923] [23908:23908] [IN] UDP::Init bound to port=37164
[1726604643.923] [23908:23908] [IN] UDP::Init bind&listen port=0
[1726604643.923] [23908:23908] [IN] UDP::Init bound to port=45330
[1726604643.923] [23908:23908] [IN] BLEBase::Init - setting/overriding transport
[1726604643.923] [23908:23908] [IN] TransportMgr initialized
[1726604643.923] [23908:23908] [FP] Initializing FabricTable from persistent storage
[1726604643.923] [23908:23908] [TS] Last Known Good Time: 2023-10-14T01:16:48
[1726604643.924] [23908:23908] [FP] Fabric index 0x1 was retrieved from storage. Compressed FabricId 0x1E00CF972DCB386B, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
[1726604643.926] [23908:23908] [ZCL] Using ZAP configuration...
[1726604643.926] [23908:23908] [IM] WARNING ┌────────────────────────────────────────────────────
[1726604643.926] [23908:23908] [IM] WARNING │ Interaction Model Engine running in 'Checked' mode.
[1726604643.926] [23908:23908] [IM] WARNING │ This executes BOTH ember and data-model code paths.
[1726604643.926] [23908:23908] [IM] WARNING │ which is inefficient and consumes more flash space.
[1726604643.926] [23908:23908] [IM] WARNING │ This should be done for testing only.
[1726604643.926] [23908:23908] [IM] WARNING └────────────────────────────────────────────────────
[1726604643.926] [23908:23908] [CTL] System State Initialized...
[1726604643.926] [23908:23908] [CTL] Setting attestation nonce to random value
[1726604643.926] [23908:23908] [CTL] Setting CSR nonce to random value
[1726604643.926] [23908:23910] [DL] CHIP task running
[1726604643.926] [23908:23910] [DL] HandlePlatformSpecificBLEEvent 32786
[1726604643.926] [23908:23910] [CTL] Setting attestation nonce to random value
[1726604643.926] [23908:23910] [CTL] Setting CSR nonce to random value
[1726604643.927] [23908:23910] [CTL] Generating NOC
[1726604643.927] [23908:23910] [FP] Validating NOC chain
[1726604643.928] [23908:23910] [FP] NOC chain validation successful
[1726604643.929] [23908:23910] [FP] Updated fabric at index: 0x1, Node ID: 0x000000000001B669
[1726604643.929] [23908:23910] [TS] Last Known Good Time: 2023-10-14T01:16:48
[1726604643.929] [23908:23910] [TS] New proposed Last Known Good Time: 2021-01-01T00:00:00
[1726604643.929] [23908:23910] [TS] Retaining current Last Known Good Time
[1726604643.930] [23908:23910] [FP] Metadata for Fabric 0x1 persisted to storage.
[1726604643.932] [23908:23910] [TS] Committing Last Known Good Time to storage: 2023-10-14T01:16:48
[1726604643.935] [23908:23910] [CTL] Joined the fabric at index 1. Fabric ID is 0x0000000000000001 (Compressed Fabric ID: 1E00CF972DCB386B)
[1726604643.950] [23908:23910] [CTL] Setting attestation nonce to random value
[1726604643.950] [23908:23910] [CTL] Setting CSR nonce to random value
[1726604643.950] [23908:23910] [CTL] Stopping commissioning discovery over DNS-SD
[1726604643.950] [23908:23910] [CTL] Starting commissioning discovery over DNS-SD
[1726604673.955] [23908:23910] [CTL] Discovery timed out
[1726604673.955] [23908:23910] [CTL] Stopping commissioning discovery over DNS-SD
[1726604673.955] [23908:23910] [TOO] Secure Pairing Failed
[1726604673.955] [23908:23908] [CTL] Shutting down the commissioner
[1726604673.955] [23908:23908] [CTL] Stopping commissioning discovery over DNS-SD
[1726604673.955] [23908:23908] [CTL] Shutting down the controller
[1726604673.955] [23908:23908] [IN] Expiring all sessions for fabric 0x1!!
[1726604673.955] [23908:23908] [FP] Forgetting fabric 0x1
[1726604673.955] [23908:23908] [TS] Pending Last Known Good Time: 2023-10-14T01:16:48
[1726604673.955] [23908:23908] [TS] Previous Last Known Good Time: 2023-10-14T01:16:48
[1726604673.955] [23908:23908] [TS] Reverted Last Known Good Time to previous value
[1726604673.955] [23908:23908] [CTL] Shutting down the commissioner
[... 40 lines cut ...]
[1726604853.675] [23917:23917] [TS] Pending Last Known Good Time: 2023-10-14T01:16:48
[1726604853.675] [23917:23917] [TS] Previous Last Known Good Time: 2023-10-14T01:16:48
[1726604853.675] [23917:23917] [TS] Reverted Last Known Good Time to previous value
[1726604853.676] [23917:23917] [DL] wrote settings to /tmp/chip_counters.ini
[1726604853.676] [23917:23917] [DL] NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1726604853.676] [23917:23917] [DL] Inet Layer shutdown
[1726604853.676] [23917:23917] [DL] BLE Layer shutdown
[1726604853.677] [23917:23917] [DL] System Layer shutdown
[1726604853.678] [23917:23917] [TOO] Run command failure: examples/chip-tool/commands/pairing/PairingCommand.cpp:372: CHIP Error 0x00000003: Incorrect state
```

FYI, I also tried to build the Linux thermostat example and use CHIPTool to pair. It was successful.

So my question is:
- Does CHIPTool really support controlling external Matter device that's not CHIPTool device?
- If CHIPTool theoretically allows to add a light bulb onto the network, what other things can I try here (only have a pairing code)?

I would love to provide any additional setting or log you want to know to help me solve this issue. Please don't kill the issue too soon. Anything will be helpful.

Configuration:
- CHIPTool Server: Ubuntu 24.04.1 LTS (GNU/Linux 6.8.0-1010-raspi aarch64)
- Bluetooth: bluetoothctl: 5.72
- Java: openjdk 21.0.4 2024-07-16
            OpenJDK Runtime Environment (build 21.0.4+7-Ubuntu-1ubuntu224.04) 
            OpenJDK 64-Bit Server VM (build 21.0.4+7-Ubuntu-1ubuntu224.04, mixed mode, sharing)

bzbarsky-apple, 2024-09-18:
@skyFzz You're using the chip-tool utility on Linux?  And I assume this is actually a WiFi light bulb, not Thread?

`pairing code` will only work for devices that are already on the IP network, so if you are trying to commission a device that is not there yet you can't use that.

`pairing code-wifi 1` should work for commissioning a device over BLE, assuming the device is in BLE range of the hardware chip-tool is running on and assuming the BLE bits on the relevant OS work correctly.  Some questions:

1. Do you have access to logs from the light bulb?  If so, could you attach that log from the pairing attempt to this issue?  Please don't paste the log; attach it as a file.
2. Similarly, please attach the full log from chip-tool from the `pairing code-wifi` attempt.
3. Just to make sure: after you paired the light bulb with Home Assistant, you either factory reset it before trying again or are using a different light bulb, so that it should be back to advertising over BLE, right?

Also, if you have the a pairing code (QR code in URL form or the numeric code) can get the "pin code" and discriminator (short only from the numeric code) from it by running:
```
chip-tool payload parse-setup-payload ${PAIRING_CODE}
```

That said, that's exactly what `code-wifi` ends up doing anyway, so I would not expect using those to change anything meaningful here.

skyFzz, 2024-09-19:
Thanks for your valued comment. Sorry I missed the important step. After factory reset the bulb can be scanned using `pairing code-wifi` and it was found!

closed 2024-09-19 by skyFzz